sifiveinc/freedom-u-sdk #3: Priv-1.10 branch doesn't work with QEMU / Master's branch user/pass?

Walkerks, 2017-04-12:
Everything built successfully on the priv-1.10 branch. The following is used to run qemu, but nothing happens.

 `qemu-system-riscv64 -kernel riscv-pk/bbl -initrd bbl.bin -nographic`

The master branch boots with the same command, but what is the user/password? Users default nor root without a password work.

Walkerks, 2017-04-12:
Just solved the user/password issue.

sorear, 2017-04-12:
QEMU has not been updated for 1.10 yet.  You will need to use Spike or rocket-chip on FPGA to boot the 1.10 branch.

Walkerks, 2017-04-12:
I see, thanks for the information. Are there any good resources to look at to understand the current state of everything?

sorear, 2017-04-12:
How do you mean by that?  Changes so far in 1.10?  State of repositories?  Can you ask a narrower question?

Walkerks, 2017-04-12:
The state of the repositories. Where can you see which versions of tools, Linux, and simulators/emulators work together? It looks like a soup of branches without a clear picture of what it's all for after researching and building various projects to see what works and what doesn't.

sorear, 2017-04-12:
@Walkerks If you look at QEMU, the readme lists a specific git revision of (1.9.1) freedom-u-sdk which is known to be able to generate images that work on qemu, and everything else is submoduled from there.  When I update qemu to add priv-1.10 support, I'll add instructions with sufficient detail to produce a matching boot image for that as well.

Walkerks, 2017-04-12:
One last thing, I'm attempting to build a kernel module using Linux in freedom-u-sdk, but it doesn't seem like gcc-6.1.0 is smart enough. It gives the following error. I know the module can build against my host.

`make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- -C ../linux SUBDIRS=/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest KBUILD_EXTMOD=/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest  modules
make[1]: Entering directory '/home/walker/Desktop/freedom-u-sdk/work/linux'
  CC [M]  /home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: "printk" [/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.ko] undefined!
  CC      /home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.o
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:8:1: error: variable '__this_module' has initializer but incomplete type
 __attribute__((section(".gnu.linkonce.this_module"))) = {
 ^~~~~~~~~~~~~
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:9:2: error: unknown field 'name' specified in initializer
  .name = KBUILD_MODNAME,
  ^
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:9:10: warning: excess elements in struct initializer
  .name = KBUILD_MODNAME,
          ^
<command-line>:0:16: note: in definition of macro 'KBUILD_STR'
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:9:10: note: in expansion of macro 'KBUILD_MODNAME'
  .name = KBUILD_MODNAME,
          ^~~~~~~~~~~~~~
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:9:10: note: (near initialization for '__this_module')
  .name = KBUILD_MODNAME,
          ^
<command-line>:0:16: note: in definition of macro 'KBUILD_STR'
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:9:10: note: in expansion of macro 'KBUILD_MODNAME'
  .name = KBUILD_MODNAME,
          ^~~~~~~~~~~~~~
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:10:2: error: unknown field 'init' specified in initializer
  .init = init_module,
  ^
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:10:10: warning: excess elements in struct initializer
  .init = init_module,
          ^~~~~~~~~~~
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:10:10: note: (near initialization for '__this_module')
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:14:2: error: unknown field 'arch' specified in initializer
  .arch = MODULE_ARCH_INIT,
  ^
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:14:10: error: 'MODULE_ARCH_INIT' undeclared here (not in a function)
  .arch = MODULE_ARCH_INIT,
          ^~~~~~~~~~~~~~~~
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:14:10: warning: excess elements in struct initializer
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:14:10: note: (near initialization for '__this_module')
/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.c:7:25: error: storage size of '__this_module' isn't known
 __visible struct module __this_module
                         ^~~~~~~~~~~~~
/home/walker/Desktop/freedom-u-sdk/linux/scripts/Makefile.modpost:114: recipe for target '/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.o' failed
make[4]: *** [/home/walker/Desktop/freedom-u-sdk/work/contextswitchtest/contextswitchtest.mod.o] Error 1
/home/walker/Desktop/freedom-u-sdk/linux/Makefile:1432: recipe for target 'modules' failed
make[3]: *** [modules] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:24: recipe for target '__sub-make' failed
make[1]: *** [__sub-make] Error 2
make[1]: Leaving directory '/home/walker/Desktop/freedom-u-sdk/work/linux'
Makefile:15: recipe for target 'all' failed
make: *** [all] Error 2
`

If I replace the toolchain in freedom-u-sdk to gcc-7 and change a few flags in the makefiles, the build will still fail due to an undetermined reasons. If I build freedom-u-sdk with gcc-6.1.0 and attempt to build the kernel module with gcc-7, in hopes it can understand what the kernel is doing, it complains about endianness. Is there any hope?

sorear, 2017-04-12:
```
    First, block code syntax in Markdown (used here) is
    three ` on a line before and after the code block.
    ```

Second I don't think the 1.9.1 kernel supports loading modules, so even if you managed to build this you couldn't use it?

Walkerks, 2017-04-12:
I wasn't really sure if it did or now? It seems to have insmod, modprob, and rmmod utilities, so I assumed it might work.

palmer-dabbelt, 2017-11-09:
QEMU still isn't up to date with the priv-1.10 spec, but we're working on it.

terpstra, 2018-03-28:
This branch is gone now and qemu works with master, so closing this issue.

closed 2018-03-28 by terpstra