sifiveinc/freedom-u-sdk #7: Kernel fault when running bash/coreutils

zhemao, 2017-10-10:
We're trying to upgrade the rocket-chip version for firesim, but are having trouble running buildroot+linux. Everything seems to work fine if we use busybox for the shell and core utilities, but if we use bash as the default shell or use some of the GNU coreutils, we get kernel faults or segfaults (respectively). This happens both on the FPGA and in Spike. 

I used the "new" branch of freedom-u-sdk. You can replicate the issue if you set bash to be the default shell.

zhemao, 2017-10-10:
[linux-boot-extract.log.gz](https://github.com/sifive/freedom-u-sdk/files/1370309/linux-boot-extract.log.gz)

Here's an extract from the spike log. The error is at line 9914.

palmer-dabbelt, 2017-11-09:
I'm pretty sure this was a kernel/user ABI mismatch, we'd seen something similar.  If this still reproduces on the latest master branch, then feel free to open another issue.  I'm not 100% sure I've got the ABI all set yet, I'm going to do another pass to sanitize everything as part of the Linux+glibc upstreaming process.

closed 2017-11-09 by palmer-dabbelt