sifiveinc/freedom-u-sdk #10: PK in the 'new' branch is failing compilation

Vasanvs, 2017-11-08:
Trying to rebuild riscv-pk 'new' branch for running the vmlinux image in spike. Since the last few updates of the repository had been giving an error as indicated in the commit note, checked out the last working commit reported in git with the checksum as f08c82a.

This checkout version is also giving a compilation error as below:

$ git status
HEAD detached at f08c82a
$ build_project riscv-pk --prefix=$RISCV --host=riscv64-unknown-linux-gnu
Building project riscv-pk
./libmachine.a(mtrap.o): In function `mcall_console_putchar.part.0':
mtrap.c:(.text+0x8): undefined reference to `platform__use_htif'
./libmachine.a(mtrap.o): In function `.L0 ':
mtrap.c:(.text+0x10a): undefined reference to `platform__use_htif'
./libmachine.a(mtrap.o): In function `.L36':
mtrap.c:(.text+0x2f0): undefined reference to `platform__use_htif'
collect2: error: ld returned 1 exit status

Is this a known issue? Kindly advice resolution.

palmer-dabbelt, 2017-11-09:
We just updated master, everything should be consistent now.  Feel free to re-open this bug (or another one) if you have more issues.

Sorry this was a mess for a bit.

closed 2017-11-09 by palmer-dabbelt

terpstra, 2017-11-09:
Try using master,  which should match the freedom/master hardware. We've just bumped it to the newest software versions and closed the new+priv-1.10 branches.