sifiveinc/freedom-u-sdk #13: make linux kernel is failing

Vasanvs, 2017-11-09:
If we do make CROSS_COMPILE=riscv64-unknown-linux-gnu- ARCH=riscv vmlinux 

it is failing with the error message

 CC      drivers/tty/serial/earlycon.o
drivers/tty/serial/earlycon.c:30:10: fatal error: asm/serial.h: No such file or directory
 #include <asm/serial.h>
          ^~~~~~~~~~~~~~
compilation terminated.

This is with EARLY_PRINTK macro enabled in the kernel .config file

Vasanvs, 2017-11-09:
If we take in the serial.h file from arch/m68k/include/asm/serial.h into arch/riscv/include/asm directory, the compilation is going through fine. 

Not sure of whether the #defined constants are correct values in this file for our RISCV simulation.

terpstra, 2017-11-09:
Please try these commands:
```
cd freedom-u-sdk
git checkout master
git pull
git submodule update --recursive
git clean -xfd
unset RISCV
make
```

If you still have problems, blow away your freedom-u-sdk directory and run:
```
git checkout --recursive git@github.com:sifive/freedom-u-sdk.git
cd freedom-u-sdk
unset RISCV
make
```

If you still have problems, please describe the host machine you are using. I just built master successfully with no hacks like you describe on an Intel machine with Ubuntu 16.04.1.

Vasanvs, 2017-11-10:
While updating my existing repository with the given git update instructions and then making, for the linux kernel compilation, I am getting the following error:

admin1@admin:~/Desktop/riscv/freedom-u-sdk$ make
mkdir -p /home/admin1/Desktop/riscv/freedom-u-sdk/work/linux/
cp -p /home/admin1/Desktop/riscv/freedom-u-sdk/conf/linux_defconfig /home/admin1/Desktop/riscv/freedom-u-sdk/work/linux/.config
make -C /home/admin1/Desktop/riscv/freedom-u-sdk/linux O=/home/admin1/Desktop/riscv/freedom-u-sdk/work/linux ARCH=riscv olddefconfig
make[1]: Entering directory '/home/admin1/Desktop/riscv/freedom-u-sdk/linux'
make[2]: Entering directory '/home/admin1/Desktop/riscv/freedom-u-sdk/work/linux'
  GEN     ./Makefile
  HOSTCC  scripts/kconfig/zconf.tab.o
scripts/kconfig/zconf.tab.c:2466:23: fatal error: zconf.lex.c: No such file or directory
compilation terminated.
scripts/Makefile.host:118: recipe for target 'scripts/kconfig/zconf.tab.o' failed
make[3]: *** [scripts/kconfig/zconf.tab.o] Error 1
/home/admin1/Desktop/riscv/freedom-u-sdk/linux/Makefile:548: recipe for target 'olddefconfig' failed
make[2]: *** [olddefconfig] Error 2
make[2]: Leaving directory '/home/admin1/Desktop/riscv/freedom-u-sdk/work/linux'
Makefile:145: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
make[1]: Leaving directory '/home/admin1/Desktop/riscv/freedom-u-sdk/linux'
Makefile:70: recipe for target '/home/admin1/Desktop/riscv/freedom-u-sdk/work/linux/.config' failed
make: *** [/home/admin1/Desktop/riscv/freedom-u-sdk/work/linux/.config] Error 2

Currently trying to get a fresh checkout from scratch of this repository and then running a make.

closed 2017-11-11 by Vasanvs