sifiveinc/freedom-u-sdk #14: spike bbl command is going into debug mode

Vasanvs, 2017-11-09:
With the latest version from master branch of this repository, the spike bbl command is running into a debug mode and refusing to load up the entire kernel

admin1@admin:~/Desktop/riscv/freedom-u-sdk/riscv-pk/build$ spike bbl /home/admin1/Desktop/riscv/freedom-u-sdk/linux/vmlinux
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.14.0-rc7+ (admin1@admin) (gcc version 7.2.0 (GCC)) #4 Thu Nov 9 16:33:19 IST 2017
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0xffffffe00002a9f0 (406030 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 515592
[    0.000000] Kernel command line: 
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 2049964K/2095104K available (4709K kernel code, 444K rwdata, 2424K rodata, 576K init, 1047K bss, 45140K reserved, 0K cma-reserved)
[    0.000000] random: get_random_u64 called from cache_random_seq_create+0x52/0xf4 with crng_init=0
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000000] Console: colour dummy device 80x25
[    0.000000] console [tty0] enabled
[    0.000000] bootconsole [early0] disabled
^C: 
core   0: 0xffffffe00025855e (0x00f586b3) add     a3, a1, a5
: 
core   0: 0xffffffe000258562 (0x00006294) ld      a3, 0(a3)
: 
core   0: 0xffffffe000258564 (0x0000983a) add     a6, a6, a4
: 
core   0: 0xffffffe000258566 (0x000097aa) add     a5, a5, a0
: 
core   0: 0xffffffe000258568 (0x41068333) sub     t1, a3, a6
: ^Cadmin1@admin:~/Desktop/riscv/freedom-u-sdk/riscv-pk/build$

Vasanvs, 2017-11-09:
Have used initramfs.txt and busybox for building the vmlinux as per the documented procedure for performing this activity.

Vasanvs, 2017-11-09:
My current .config file that is being used for building vmlinux is enclosed. 
[config.txt](https://github.com/sifive/freedom-u-sdk/files/1457494/config.txt)

terpstra, 2017-11-09:
Your spike version may be out-of-date. Please try running:
```
cd freedom-u-sdk
unset RISCV
make sim
```

I just used your config.txt successfully to both compile and boot with no other modifications to freedom-u-sdk.

Vasanvs, 2017-11-09:
If we do unset RISCV and try to build riscv-isa-sim in freedom-u-sdk, then we get an error stating 

admin1@admin:~/Desktop/riscv/freedom-u-sdk/riscv-isa-sim$ ./configure
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for ar... ar
checking for ranlib... ranlib
checking for dtc... /usr/bin/dtc
checking how to run the C++ preprocessor... g++ -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking whether byte ordering is bigendian... no
checking for a BSD-compatible install... /usr/bin/install -c
checking for ANSI C header files... (cached) yes
configure: configuring default subproject : riscv
checking for library containing dlopen... -ldl
checking for libfesvr_is_present in -lfesvr... no
configure: error: libfesvr is required

terpstra, 2017-11-10:
Yeah, based on your other issues I think you have somehow got a bad checkout. Please try a fresh 'git clone --recursive'.

Vasanvs, 2017-11-10:
On a fresh repository checkout and build after throwing out the older repository, all commands seems to be getting build EXCEPT spike. Where would spike be found?

admin1@admin:~/Desktop/riscv/new-freedom-u-sdk/freedom-u-sdk$ find . -name spike -print
admin1@admin:~/Desktop/riscv/new-freedom-u-sdk/freedom-u-sdk$

terpstra, 2017-11-10:
When you run 'make sim' it will build it.

terpstra, 2017-11-10:
FYI, with the freedom-u-sdk, make sure you never have RISCV set, or it will use whatever toolchain you have in your environment. The SDK brings in an up-to-date matching toolchain of it's own when RISCV is unset.

Vasanvs, 2017-11-10:
Thanks for all your help. Yeah. I saw it and it worked like a breeze. The login prompt has come up with spike now. 

Appreciate all your and teams immediate responses to resolve issues. 

Thanks

closed 2017-11-10 by Vasanvs