sifiveinc/freedom-u-sdk #15: Incorrect selection of kernel headers: expected 4.6.x, got 4.1.x

hylee17, 2017-11-12:
@palmer-dabbelt 

The source code has been updated, but it still has the same problem.

According to the guide document, proceed as follows.

1. git clone --recursive https://github.com/sifive/freedom-u-sdk.git

2. cd freedom-u-sdk

3. make
  -> Incorrect selection of kernel headers: expected 4.6.x, got 4.1.x

--------------------------------------------------------------------------------------------
root@ubuntu:/risc-v/DIR/freedom-u-sdk# git submodule
 672f627ce0fe9720b4a193fb9bd583518a990882 buildroot (2016.05-515-g672f627)
 160fa8470d0f78b1571c04da8b9af50ac05be378 linux (v4.14-rc7-28-g160fa84)
 f683e01542acf60e50774d061bcb396b628e3e67 riscv-fesvr (heads/master)
 b4dae89f85bf882852c6186b1284df11065bfcd9 riscv-gnu-toolchain (v20171107)
 092b464c06d2eba580b95a3ac7013449cb2687e4 riscv-isa-sim (remotes/origin/dtm_reset_error-33-g092b464)
 57740d97cdec928c372d8e752c04a691418cbfa1 riscv-pk (remotes/origin/enable-32bit)
-----------------------------------------------------------------------------------------------
root@ubuntu:/risc-v/DIR/freedom-u-sdk# riscv64-unknown-linux-gnu-gcc --version
riscv64-unknown-linux-gnu-gcc (GCC) 6.1.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@ubuntu:/risc-v/DIR/freedom-u-sdk# riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 6.1.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
------------------------------------------------------------------------------------------------

terpstra, 2017-11-12:
Before running make, try running 'unset RISCV'.

hylee17, 2017-11-12:
@terpstra 

Thanks. I'll try it.
And I have already installed tool-chain 6.1 on riscv,
Do I need to reinstall riscv-gnu-tools in the freedom-u-sdk directory?

hylee17, 2017-11-12:
@terpstra 
OK. Thanks.
I am working on make now.

For bitstream files, the guide documents provide pre-built files.
Is there a guide that I can make and test myself?

terpstra, 2017-11-12:
It does not install it's toolchain; it's just in the toolchain/ subdirectory.

palmer-dabbelt, 2017-11-13:
I'm going to close this as a feature, not a bug :).  It's best to just use the toolchain in freedom-u-sdk, as that's guaranteed to match up.

closed 2017-11-13 by palmer-dabbelt

hylee17, 2017-11-15:
@palmer-dabbelt 
Thanks. I made image