sifiveinc/freedom-u-sdk #16: Build without support of Compress ISA

Vasanvs, 2017-11-13:
We are trying to build the tool chain without the support of compression ISA set. 

We did the following changes in the different files for this purpose:
a) In the Makefile @ freedom-u-sdk directory, we changed the rule for building bbl as follows:
$(bbl): $(pk_srcdir) $(vmlinux_stripped)
        rm -rf $(pk_wrkdir)
        mkdir -p $(pk_wrkdir)
        cd $(pk_wrkdir) && $</configure \
                --host=$(target) \
                --with-payload=$(vmlinux_stripped) \
                --enable-logo \
                --with-logo=$(abspath conf/sifive_logo.txt)
        CFLAGS="-mabi=lp64d -march=rv64imafd" $(MAKE) -C $(pk_wrkdir)

- Basically, replaced the architecture support of rv64imafdc to rv64imafd in the last line of the rule. 

b) In the riscv-pk/configure script, made the change in the CFLAGS and LDFLAGS for a 64-bit architecture as follows:

case "${BUILD_32BIT}" in
  yes|default)
        echo "Building 32-bit pk"
        CFLAGS="$default_CFLAGS -march=rv32imafdc -mabi=ilp32d"
        LDFLAGS="-march=rv32imafdc -mabi=ilp32d"
        install_subdir="riscv32-unknown-elf"
        ;;
  *)
        CFLAGS="$default_CFLAGS -march=rv64imafd -mabi=lp64d"
        LDFLAGS="-march=rv64imafd -mabi=lp64d"
        install_subdir="riscv64-unknown-elf"
        ;;
esac

- Basically, replacing the architecture for the * case above as rv64imafd instead of what was set as rv64imafdc for the architecture in the repository.

After making the above changes in the files, did a 'make clean' and then issued a fresh make and then subsequently make sim. Verified that there was no reference of architecture as rv64imafdc and reference of architecture was only rv64imafd - in the entire make build output - thereby proving that the architecture was never getting set as rv64imafdc through out the entire repository build.

When the bbl has got built and appended with the vmlinux as payload as part of this make, and then spike started this bbl loading, there is an error message that is coming in as part of the load process as below:

           SiFive RISC-V Coreplex
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.14.0-rc7-00028-g160fa84-dirty (admin1@admin) (gcc version 7.2.0 (GCC)) #1 SMP Mon Nov 13 14:19:43 IST 2017
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0xffffffe00001e200 (6598656 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 14 pages/cpu @ffffffe07e1ee000 s28376 r0 d28968 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516615
[    0.000000] Kernel command line: earlyprintk 
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 2052044K/2095104K available (2826K kernel code, 239K rwdata, 758K rodata, 6599K init, 774K bss, 43060K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU event tracing is enabled.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
/home/admin1/Desktop/riscv/new-freedom-u-sdk/freedom-u-sdk/riscv-pk/machine/mtrap.c:18: machine mode: unhandlable trap 4 @ 0x0000000080003434

On the disassembled bbl executable, 80003434 is having:
80003434:   00066703                lwu     a4,0(a2) # 1000 <_ftext-0x7ffff000>

-------------------

For having the support of RV64IMAFD and NOT having the support of RV64IMAFDC ISA, should there be anything else that is done for a successful load up in spike till the login prompt? 

Thanks in advance for your inputs and help,

Vasan

jim-wilson, 2017-11-14:
I tried looking at this.  Running spike with the -l option, I see
core   0: 0x0000000080003434 (0x00066703) lwu     a4, 0(a2)
core   0: exception trap_load_address_misaligned, epc 0x0000000080003434
core   0:           badaddr 0xffffffe0007a807e
Looking at the linux kernel System.map file I see
ffffffe0007a8056 t next_event
ffffffe0007a8098 T timer_riscv_init
so the badaddr is inside the next_event function.  Disassembling this function I see
ffffffe0007a8056 <next_event>:
ffffffe0007a8056:       1141                    addi    sp,sp,-16
...
ffffffe0007a807e:       c01027f3                rdtime  a5

It looks like most of the SDK was built with the default rv64imafdc, and only riscv-pk got the -march=rv64imafd option.  I don't know why this didn't fail earlier.  I can fix the kernel by adding
CONFIG_RISCV_ISA_C=n
to the top of the conf/linux_defconfig file.  Forcing the rebuild of the kernel I can now boot to the login prompt.

However, I see that buildroot busybox still has compressed instructions in it, though I'm not seeing failures inside spike.  The -march option should be passed down to buildroot somehow, perhaps via CFLAGS.  And you also need to pass the -march option down to spike somehow also.  I see that spike has a --isa option, so if I change the toplevel sim rule to
	$(spike) --isa=rv64imafd -p4 $(bbl)
that seems to do the right thing, but now I get a linux kernel panic, because init has compressed instructions in it.  And those compressed instructions apparently come from ld.so from glibc.  I think the reason things mostly worked even with compressed instructions is because spike was silently executing the compressed instructions by default.  When the spike compressed support is turned off, things break badly, but it shows where the build problems are.

I think you need to pass the -march option down into the toolchain build also to get ld.so built correctly without compressed instructions.  This needs to be passed to the toolchain configure command.

Clearly, this is not something supported out of the fox, but it looks doable with a bit more work.

Vasanvs, 2017-11-14:
Thanks for your detailed comments Jim. Appreciate it much. 

If I add up the line CONFIG_RISCV_ISA_C=n at the top of freedom-u-sdk/conf/linux_defconfig file and then force a build of linux kernel by removing up the work/linux directory, I still find that this config variable does not come in the generated .config file in the work/linux directory.

jim-wilson, 2017-11-14:
In my work/linux/.config file I see
# CONFIG_RISCV_ISA_C is not set                                                 

All I did to get that is add CONFIG_RISCV_ISA_C=n to the linux_defconfig file.  This setting is on by default, so just having the comment line in the .config file is all you need to get the right result.

In an unmodified build tree, the work/linux/.config file has
CONFIG_RISCV_ISA_C=y
which is not what you want.

jim-wilson, 2017-11-14:
Foiled by markdown.  That was supposed to be a line starting with a hash/octothorpe, indicating a comment line in the .config file.

Vasanvs, 2017-11-14:
Thanks much. I observed as you indicated in the comment on CONFIG_RISCV_ISA_C being marked as not set. 

Now when I run 'make sim' after modifying the rule for sim in the Makefile as follows:
sim: $(spike) $(bbl)
        $(spike) --isa=RV64IMAFD $(bbl)
I get the buildroot login prompt successfully till the end
           SiFive RISC-V Coreplex
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.14.0-rc7-00028-g160fa84 (admin1@admin) (gcc version 7.2.0 (GCC)) #2 SMP Tue Nov 14 07:01:31 IST 2017
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0xffffffe000025200 (6598656 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] elf_hwcap is 0x1129
[    0.000000] percpu: Embedded 14 pages/cpu @ffffffe07e1ee000 s28376 r0 d28968 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516615
[    0.000000] Kernel command line: earlyprintk 
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 2050884K/2095104K available (3956K kernel code, 239K rwdata, 756K rodata, 6631K init, 774K bss, 44220K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU event tracing is enabled.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000000] console [hvc0] enabled
[    0.000000] console [hvc0] enabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.000000] pid_max: default: 32768 minimum: 301
[    0.000000] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Hierarchical SRCU implementation.
[    0.000000] smp: Bringing up secondary CPUs ...
[    0.000000] smp: Brought up 1 node, 1 CPU
[    0.000000] devtmpfs: initialized
[    0.000000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.000000] futex hash table entries: 256 (order: 2, 16384 bytes)
[    0.010000] random: get_random_u32 called from bucket_table_alloc+0x134/0x3c0 with crng_init=0
[    0.010000] NET: Registered protocol family 16
[    0.010000] vgaarb: loaded
[    0.010000] SCSI subsystem initialized
[    0.010000] usbcore: registered new interface driver usbfs
[    0.010000] usbcore: registered new interface driver hub
[    0.010000] usbcore: registered new device driver usb
[    0.010000] pps_core: LinuxPPS API ver. 1 registered
[    0.010000] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.010000] PTP clock support registered
[    0.010000] clocksource: Switched to clocksource riscv_clocksource
[    0.010000] NET: Registered protocol family 2
[    0.020000] TCP established hash table entries: 16384 (order: 5, 131072 bytes)
[    0.020000] TCP bind hash table entries: 16384 (order: 6, 262144 bytes)
[    0.020000] TCP: Hash tables configured (established 16384 bind 16384)
[    0.020000] UDP hash table entries: 1024 (order: 3, 32768 bytes)
[    0.020000] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)
[    0.020000] NET: Registered protocol family 1
[    0.030000] Unpacking initramfs...
[    0.040000] workingset: timestamp_bits=62 max_order=19 bucket_order=0
[    0.050000] random: fast init done
[... 10 lines cut ...]
[    0.080000] IR NEC protocol handler initialized
[    0.080000] IR RC5(x/sz) protocol handler initialized
[    0.080000] IR RC6 protocol handler initialized
[    0.080000] IR JVC protocol handler initialized
[    0.080000] IR Sony protocol handler initialized
[    0.080000] IR SANYO protocol handler initialized
[    0.080000] IR Sharp protocol handler initialized
[    0.090000] IR MCE Keyboard/mouse protocol handler initialized
[    0.090000] IR XMP protocol handler initialized
[    0.090000] usbcore: registered new interface driver usbhid
[    0.090000] usbhid: USB HID core driver
[    0.090000] NET: Registered protocol family 17
[    0.090000] Freeing unused kernel memory: 6628K
[    0.090000] This architecture does not have kernel memory protection.
Starting logging: OK
Starting mdev...
modprobe: can't change directory to '/lib/modules': No such file or directory
Initializing random number generator... done.
Starting network...
Waiting for interface eth0 to appear............... timeout!
run-parts: /etc/network/if-pre-up.d/wait_iface: exit status 1
Starting dropbear sshd: OK

Welcome to Buildroot
buildroot login:

Vasanvs, 2017-11-14:
Here are the changes made by me till now as compared to the checked in repository:

admin1@admin:~/Desktop/riscv/FUSDK-13Nov17/freedom-u-sdk$ git diff Makefile
diff --git a/Makefile b/Makefile
index c1c4c00..8f97f79 100644
--- a/Makefile
+++ b/Makefile
@@ -49,7 +49,7 @@ all: $(hex)
 
 $(toolchain_dest)/bin/$(target)-gcc: $(toolchain_srcdir)
        mkdir -p $(toolchain_wrkdir)
-       cd $(toolchain_wrkdir); $(toolchain_srcdir)/configure --prefix=$(toolchain_dest)
+       cd $(toolchain_wrkdir); $(toolchain_srcdir)/configure --prefix=$(toolchain_dest) --with-arch=rv64imafd --with-abi=lp64d
        $(MAKE) -C $(toolchain_wrkdir) linux
        sed 's/^#define LINUX_VERSION_CODE.*/#define LINUX_VERSION_CODE 263682/' -i $(toolchain_dest)/sysroot/usr/include/linux/version.h
 
@@ -94,7 +94,7 @@ $(bbl): $(pk_srcdir) $(vmlinux_stripped)
                --with-payload=$(vmlinux_stripped) \
                --enable-logo \
                --with-logo=$(abspath conf/sifive_logo.txt)
-       CFLAGS="-mabi=lp64d -march=rv64imafdc" $(MAKE) -C $(pk_wrkdir)
+       CFLAGS="-mabi=lp64d -march=rv64imafd" $(MAKE) -C $(pk_wrkdir)
 $(bin): $(bbl)
        $(target)-objcopy -S -O binary --change-addresses -0x80000000 $< $@
@@ -134,4 +134,4 @@ clean:
 
 .PHONY: sim
 sim: $(spike) $(bbl)
-       $(spike) -p4 $(bbl)
+       $(spike) --isa=RV64IMAFD $(bbl)

admin1@admin:~/Desktop/riscv/FUSDK-13Nov17/freedom-u-sdk/riscv-pk$ git diff
diff --git a/configure b/configure
index 61c0292..e22a5b5 100755
--- a/configure
+++ b/configure
@@ -4089,8 +4089,8 @@ case "${BUILD_32BIT}" in
        install_subdir="riscv32-unknown-elf"
        ;;
   *)
-       CFLAGS="$default_CFLAGS -march=rv64imafdc -mabi=lp64d"
-       LDFLAGS="-march=rv64imafdc -mabi=lp64d"
+       CFLAGS="$default_CFLAGS -march=rv64imafd -mabi=lp64d"
+       LDFLAGS="-march=rv64imafd -mabi=lp64d"
        install_subdir="riscv64-unknown-elf"
        ;;
 esac

Vasanvs, 2017-11-14:
The code diffs in the above comment has been foiled by Markdown as bullet points...

richardxia, 2017-11-14:
If you surround you code/console output with triple backticks (```), then it'll be rendered as a code block and Markdown won't try to style it.

Vasanvs, 2017-11-14:
Done. Hope it is clear now.

Vasanvs, 2017-11-14:
Since the spike here has been specifically started with --isa=RV64IMAFD and we are able to get to the buildroot login stage in the boot process, can we assume that with these changes in the toolchain build process as well as in kernel, we have generated a successful build without Compressed ISA support?

palmer-dabbelt, 2017-11-14:
@Vasanvs If you can login in Spike without the C extension in the `--isa` list, then you've probably gotten far enough that everything is C-free.  Do you mind adding some sort of Makefile option that enables or disables the C extension?

Vasanvs, 2017-11-14:
Thanks for the confirmation Palmer. 

Regarding the Makefiles, is it an option to have two Makefiles separately to make it explicit of whether the user wants a build with or without C extension?

palmer-dabbelt, 2017-11-14:
I think it'd be better to have something like

````
make ARCH=rv64imafd
````

so then we don't have to fork the makefiles.

Vasanvs, 2017-11-15:
@palmer-dabbelt 
Have made the diffs in Makefile to conditionally build it for rv64imafd or rv64imafdc. 
The usage is as follows:

$ make ISA=rv64imafd

If the ISA variable is not passed as argument, it would default to build it for CISA.

The diff in the Makefile is as follows:
```
diff --git a/Makefile b/Makefile
index c1c4c00..40d7eff 100644
--- a/Makefile
+++ b/Makefile
@@ -1,3 +1,7 @@
+isa = $(ISA)
+ifeq ($(strip $(isa)),)
+isa = rv64imafdc
+endif
 RISCV ?= $(CURDIR)/toolchain
 PATH := $(RISCV)/bin:$(PATH)

@@ -42,6 +46,7 @@ target := riscv64-unknown-linux-gnu

 .PHONY: all
 all: $(hex)
+       @echo This image has been generated for ARCH of $(isa)
        @echo
        @echo Find the SD-card image in work/bbl.bin
        @echo Program it with: dd if=work/bbl.bin of=/dev/sd-your-card bs=1M
@@ -49,7 +54,7 @@ all: $(hex)

 $(toolchain_dest)/bin/$(target)-gcc: $(toolchain_srcdir)
        mkdir -p $(toolchain_wrkdir)
-       cd $(toolchain_wrkdir); $(toolchain_srcdir)/configure --prefix=$(toolchain_dest)
+       cd $(toolchain_wrkdir); $(toolchain_srcdir)/configure --prefix=$(toolchain_dest) --with-arch=$(isa) --with-abi=lp64d
        $(MAKE) -C $(toolchain_wrkdir) linux
        sed 's/^#define LINUX_VERSION_CODE.*/#define LINUX_VERSION_CODE 263682/' -i $(toolchain_dest)/sysroot/usr/include/linux/version.h

@@ -69,6 +74,13 @@ $(sysroot_stamp): $(buildroot_tar)
 $(linux_wrkdir)/.config: $(linux_defconfig) $(linux_srcdir)
        mkdir -p $(dir $@)
        cp -p $< $@
+ifeq ($(isa), rv64imafd)
+       @echo "CONFIG_RISCV_ISA_C=n" > /tmp/linux_defconfig
+       cat /tmp/linux_defconfig $(linux_defconfig) > /tmp/linux_defconfig.tmp
+       mv /tmp/linux_defconfig.tmp $(linux_defconfig)
+else
+       @echo "ISA is not set: " $(isa)
+endif
        $(MAKE) -C $(linux_srcdir) O=$(linux_wrkdir) ARCH=riscv olddefconfig

 $(vmlinux): $(linux_srcdir) $(linux_wrkdir)/.config $(sysroot_stamp)
@@ -94,7 +106,7 @@ $(bbl): $(pk_srcdir) $(vmlinux_stripped)
                --with-payload=$(vmlinux_stripped) \
                --enable-logo \
                --with-logo=$(abspath conf/sifive_logo.txt)
-       CFLAGS="-mabi=lp64d -march=rv64imafdc" $(MAKE) -C $(pk_wrkdir)
+       CFLAGS="-mabi=lp64d -march=$(isa)" $(MAKE) -C $(pk_wrkdir)

 $(bin): $(bbl)
        $(target)-objcopy -S -O binary --change-addresses -0x80000000 $< $@
@@ -134,4 +146,4 @@ clean:

 .PHONY: sim
 sim: $(spike) $(bbl)
-       $(spike) -p4 $(bbl)
+       $(spike) --isa=$(isa) -p4 $(bbl)

```

Vasanvs, 2017-11-15:
Kindly review and provide comment

palmer-dabbelt, 2017-11-15:
Your patch didn't apply, so I manually applied it and opened #17.

palmer-dabbelt, 2017-11-15:
I'm closing this, as I think my patch should fix it.

closed 2017-11-15 by palmer-dabbelt

Vasanvs, 2017-11-18:
Line #57 of the patch is mentioning the argument to configure script as --with-isa. This argument should be --with-arch for it to work as expected.

Since --with-isa is not a supported configure option, it is getting ignored and even after specifying it as rv64imafd, the toolchain is getting built as rv64imafdc ISA.