sifiveinc/freedom-u-sdk #19: RISCV-PK Non-CISA build

Vasanvs, 2017-11-22:
The architecture field (-march) in the configure script of riscv-pk is hardcoded to build only with rv64imafdc support. The global Makefile has already been modified to take care of both ISA and CISA support but this configure script also should be parametrized to build as per global Makefile -march parameter.

The relevant portion of the code in freedom-u-sdk/riscv-pk/configure script (Line #4084) is as follows:

case "${BUILD_32BIT}" in
  yes|default)
        echo "Building 32-bit pk"
        CFLAGS="$default_CFLAGS -march=rv32imafdc -mabi=ilp32d"
        LDFLAGS="-march=rv32imafdc -mabi=ilp32d"
        install_subdir="riscv32-unknown-elf"
        ;;
  *)
        CFLAGS="$default_CFLAGS -march=rv64imafdc -mabi=lp64d"
        LDFLAGS="-march=rv64imafdc -mabi=lp64d"
        install_subdir="riscv64-unknown-elf"
        ;;
esac

palmer-dabbelt, 2017-11-22:
Can you submit a patch to fix it?  There should probably be something like a "--with-target-isa" and "--with-target-abi" argument to configure in riscv-pk.

georgejsh, 2017-11-23:
This is a patch I made.

*** Makefile	2017-11-23 16:18:16.686808161 +0530
*** 110,119 ****
```  		--host=$(target) \
  		--with-payload=$(vmlinux_stripped) \
  		--enable-logo \
! 		--with-logo=$(abspath conf/sifive_logo.txt) \
! 		--with-target-isa=$(ISA) \
! 		--with-target-abi=$(ABI) 
! 	 $(MAKE) -C $(pk_wrkdir)
  
  $(bin): $(bbl)
  	$(target)-objcopy -S -O binary --change-addresses -0x80000000 $< $@
--- 110,117 ----
  		--host=$(target) \
  		--with-payload=$(vmlinux_stripped) \
  		--enable-logo \
! 		--with-logo=$(abspath conf/sifive_logo.txt)
! 	CFLAGS="-mabi=$(ABI) -march=$(ISA)" $(MAKE) -C $(pk_wrkdir)
  
  $(bin): $(bbl)
  	$(target)-objcopy -S -O binary --change-addresses -0x80000000 $< $@ ```
*** riscv-pk/configure.ac	2017-11-23 16:11:42.224300391 +0530
***************
*** 80,96 ****
  ```
  default_CFLAGS="-Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks"
  
- 
- AC_ARG_WITH([target-isa],
- 	AS_HELP_STRING([--with-target-isa], [Specify the target architecture]),
- 	[AC_SUBST([BUILD_ARCH], $with_target_isa, [Arch for build])],
- 	[AC_SUBST([BUILD_ARCH], [rv64imafdc], [Arch for build])])
- 
- AC_ARG_WITH([target-abi],
- 	AS_HELP_STRING([--with-target-abi], [Specify the target abi]),
- 	[AC_SUBST([BUILD_ABI], $with_target_abi, [abi for build])],
- 	[AC_SUBST([BUILD_ABI], [lp64d], [abi for build])])
- 
  AC_ARG_ENABLE([32bit],
  	AS_HELP_STRING([--enable-32bit], [Build a 32-bit pk]),
  	BUILD_32BIT=$enableval,
--- 80,85 ----
***************
*** 104,111 ****
  	install_subdir="riscv32-unknown-elf"
  	;;
    *)
! 	CFLAGS="$default_CFLAGS -march=${BUILD_ARCH} -mabi=${BUILD_ABI}"
! 	LDFLAGS="-march=${BUILD_ARCH} -mabi=${BUILD_ABI}"
  	install_subdir="riscv64-unknown-elf"
  	;;
  esac
--- 93,100 ----
  	install_subdir="riscv32-unknown-elf"
  	;;
    *)
! 	CFLAGS="$default_CFLAGS -march=rv64imafdc -mabi=lp64d"
! 	LDFLAGS="-march=rv64imafdc -mabi=lp64d"
  	install_subdir="riscv64-unknown-elf"
  	;;
  esac ```

Vasanvs, 2017-12-12:
@palmer-dabbelt

palmer-dabbelt, 2017-12-12:
I'm not really sure what's going on with the patch above.  Can you either use `git diff` to produce the patch, or just submit a pull request?

tprrt, 2022-11-15:
This issue no longer occurs.

closed 2022-11-15 by tprrt