sifiveinc/freedom-u-sdk #20: the bootloader runs, but fails to boot linux

hylee17, 2017-12-06:
@palmer-dabbelt 

Hello.

I can not boot because of the following reasons.
https://github.com/sifive/freedom/issues/38

The source code was not booted for the same reason before it was updated.
Is the installation and setup wrong?

Below are the logs that created the bitstream and image file using freedom and freedom-u-sdk.

freedom(/master)
# make -f Makefile.u500vc707devkit verilog
# make -f Makefile.u500vc707devkit mcs

Bitstream generation succeeded (Vivado : 2016_01)

freedom-u-sdk(/master)
$ /freedom-u-sdk# cd riscv-gnu-toolchain
$ make & linux make
 -> Toolchain installation (7.2.0)
$ /freedom-u-sdk# make
 -> linux image generation

palmer-dabbelt, 2017-12-07:
Well, first let's try to figure out if it's hardware or software.  Can you try the bitstream from SiFive's website?  That's at least a known-good image on the hardware end.

hylee17, 2017-12-07:
@palmer-dabbelt 

When I try with a pre-built bitstream provided by SiFive, the boot loader will repeat the loading indefinitely as shown below. 
[https://www.sifive.com/documentation/freedom-soc/freedom-u500-vc707-fpga-dev-kit-getting-started-guide/]
[image]

No response from BOOT when creating bitstream and image with updated freedom / freedom-u-sdk.

In freedom, I tested both master and no-pcie, and the results are the same. [have FMC Card]

I think the bitstream I created is generated correctly. I suspect that there is something missing in the process of creating the image file.

Can you confirm that the process of creating the image file I have listed above is correct?

Below are my ubuntu preferences.
[image]
and Vivado 2016.01 version

terpstra, 2017-12-07:
Can you run these commands:
cd freedom-u-sdk_master
git checkout master
git submodule update --recursive
git clean -xfd
unset RISCV # this one matters; i suspect you have a bad toolchain installed and this disables it
make -j4

hylee17, 2017-12-07:
@terpstra 
I am proceeding with the commands you have written.
I also think that the process of creating a Linux image is wrong.
I am doing unset RISCV and doing make, but I did unset RISCV in previous work.
Is there anything else I need to set up for other environment variables?

I am currently running make -j4 and I will write again after the test.

hylee17, 2017-12-07:
@terpstra 

The make is done, but the result is still the same.
There is no response after BOOT.
[image]

palmer-dabbelt, 2017-12-07:
Are you writing to the raw device?  It should look like "/dev/sdb", not "/dev/sdb1".

hylee17, 2017-12-09:
@palmer-dabbelt 
Thanks. 
I succeeded in booting Linux.
[image]

I was writing the "dd" command incorrectly.
As you said, I put it into "/dev/sdb" correctly and it booted.

closed 2018-03-28 by terpstra

yanyh15, 2019-02-15:
I hope somebody can still help even this issue was closed. 

What is the reason that it has to be written to the raw device /dev/sda? I had the same issue on vc707. Linux only boot when bbl.bin is written to raw device. The latest freedom-u-sdk has a ```make DISK=/dev/sda format-boot-loader``` target to partition the ssd and write bbl.bin to one partition. see below. But the ssd created won't boot. So does the default kernel build has the ssd driver? Is there instructions or doc for creating the ssd that has kernel and root file system to boot on vc707?

Thank you. 

```
yanyh@lynx:~/chips/freedom-u-sdk$ sudo make DISK=/dev/sda format-boot-loader
sgdisk --clear                                                               \
	--new=1:2048:67583  --change-name=1:bootloader --typecode=1:2E54B353-1271-4842-806F-E436D6AF6985   \
	--new=2:264192:     --change-name=2:root       --typecode=2:0FC63DAF-8483-4772-8E79-3D69D8477DE4 \
	/dev/sda
Setting name!
partNum is 0
REALLY setting name!
Setting name!
partNum is 1
REALLY setting name!
The operation has completed successfully.
dd if=/home/yanyh/chips/freedom-u-sdk/work/bbl.bin of=/dev/sda1 bs=4096
4086+1 records in
4086+1 records out
16738804 bytes (17 MB, 16 MiB) copied, 0.271233 s, 61.7 MB/s
mke2fs -t ext3 /dev/sda2
mke2fs 1.42.13 (17-May-2015)
Creating filesystem with 31184123 4k blocks and 7798784 inodes
Filesystem UUID: 9082c809-8f4c-49fa-b726-65a02a3d18da
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
	4096000, 7962624, 11239424, 20480000, 23887872

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information:        

done
```

jim-wilson, 2019-02-15:
I believe that the vc707 FPGA image has a slightly different boot loader than the HiFive Unleashed.  freedom-u-sdk is only correct for the HiFive Unleashed.  I don't have a vc707 to play with, so i don't know the details.  But yes, I believe that the HiFive Unleashed boot loader supports partitions, and the vc707 does not, so the bbl+kernel must be written to a raw device for the vc707.

jim-wilson, 2019-02-15:
The vc707 getting started guide here
https://www.sifive.com/documentation
in chapter 4, does mention that you have to write the bbl+kernel to the raw sdcard.  If we want to mount a disk, then we attach a pcie extender and use that to connect a disk, as also mentioned in the docs.

yanyh15, 2019-02-15:
Thank you Jim. I want to give a shot of u-boot (https://github.com/Microsemi-SoC-IP/HiFive_U-Boot) or [freedom-u540-c000s-bootloader](https://www.sifive.com/blog/an-open-source-release-of-the-freedom-u540-c000s-bootloader) on VC707 to see whether I can have it working. What do you or others think of this idea?

jim-wilson, 2019-02-15:
I've never played with a vc707 or the bootloader code, so I have no idea if this can work or be useful.