sifiveinc/freedom-u-sdk #23: Build 32-bit target failed

kting93, 2018-01-22:
Hi, 

I am trying to build 32-bit target from freedom-u-sdk. The build of 64-bit target works well but the 32-bit failed. What I am doing is modifying the Makefile and changed the variables 

ISA ?= rv32ima
ABI ?= ilp32

Then type make linux and I got no rule to make target error:

[image]

Please help to point out what I am doing wrong. 

Thanks in advance, 
Kai-Yuan

jim-wilson, 2018-01-22:
32-bit builds aren't supported yet.  The ISA and ABI variables are relatively new and untested.  Only very minor changes to them are likely to work.

kting93, 2018-01-22:
Hi Jim,

Thanks for your reply. Looks like I have to manually configure and build the gcc, linux, buildroot, spike, bbl and qemu to run in 32-bit mode. 

Thanks,
Kai-Yuan

closed 2018-01-23 by kting93