sifiveinc/freedom-u-sdk #26: Linux headers need to be updated in build process?

sunnystormy, 2018-01-25:
When I run a "make all" after updating the submodules, the script crashes with an

"Incorrect selection of kernel headers: expected 4.6.x, got 4.13.x
package/pkg-generic.mk:185: recipe for target '../freedom-u-sdk/work/buildroot_initramfs/build/toolchain-external-undefined/.stamp_configured' failed
make[1]: *** [../freedom-u-sdk/work/buildroot_initramfs/build/toolchain-external-undefined/.stamp_configured] Error 1
make[1]: Leaving directory '/home/michael/Code/risc_v/freedom-u-sdk/buildroot'
Makefile:80: recipe for target '../freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar' failed
make: *** [../freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar] Error 2"

Please fix! :)

Thank you!

palmer-dabbelt, 2018-01-25:
This line should be tricking buildroot into thinking we have the headers for a 4.6 kernel

https://github.com/sifive/freedom-u-sdk/blob/master/Makefile#L71

and it works for me.

sunnystormy, 2018-01-25:
@palmer-dabbelt Thank you for looking into it. I'm still getting the same error, but I also get a

">>> toolchain-external undefined Configuring"

before it.

I have the "riscv-tools" installed in another directory, with a "build" subdirectory set to my "$RISCV" variable. Could that be responsible?

palmer-dabbelt, 2018-01-25:
Yes.  You should use the tools built by this repository.

sunnystormy, 2018-01-25:
@palmer-dabbelt Awesome. I just unset the variable, and it's currently building! Thank you very much for your help!

closed 2018-01-25 by sunnystormy