sifiveinc/freedom-u-sdk #28: Can not SSH to qemu?

hoangt, 2018-01-30:
Hello All,

I run "make qemu" the linux is boot with U/P=root/sifive. 

The IP of qemu is configured by default as below
inet addr:**10.0.2.15**  Bcast:10.0.2.255  Mask:255.255.255.0

In host, I configure its IP as
inet addr:**10.0.2.30**  Bcast:10.0.2.255  Mask:255.255.255.0

From the host, I ping the qemu (and vice versa) and all packets are lost. When I do "ssh root@10.0.2.15", an error message showed below
ssh: connect to host **10.0.2.15** port 22: No route to host

Do you have any idea to resolve this issue? Many thanks !

/T

sorear, 2018-01-30:
`make qemu` does not support inbound connections by default.  You will need to modify the command line to ssh in (the `hostfwd` option from `man qemu` looks promising)

hoangt, 2018-01-31:
Thank sorear !

Here is what I changed in Makefile according to your suggestion. Current status:
* Ping qemu (10.11.46.50) from host   >>> Passed
* Ping host   (10.11.46.33) from qemu >>> Failed
* SSH qemu (10.11.46.50) from  host   >>> Failed (ssh: connect to host ... port 22: Connection refused)

All suggestions are welcome!

/T

```
#**** Changes in Makefile ****
host_name      = hoangt
host_ip        = 10.11.46.33
qemu_ip        = 10.11.46.60
qemu_dhcpstart = 10.11.46.50
host_fwd       = tcp::5556-:22,hostname=$(host_name),addr=$(host_ip)

.PHONY: qemu
qemu: $(qemu) $(bbl) $(rootfs)
  $(qemu) -nographic \
    -machine virt \
    -kernel  $(bbl) \
    -drive   file=$(rootfs),format=raw,id=hd0 \
    -device  virtio-blk-device,drive=hd0 \
    -netdev  user,id=net0,net=$(qemu_ip)/22,dhcpstart=$(qemu_dhcpstart),hostfwd=$(host_fwd) \
    -device  virtio-net-device,netdev=net0

```

sorear, 2018-01-31:
My understanding of the manual is that you will want `ssh -p 5556 localhost` in order to use the tunnel.  I don't have this working myself, and bridge/tap may work better for what you're doing.  This is also not a RISC-V specific question.

hoangt, 2018-01-31:
Thanks rorear,

I am trying to handle. Your comment is valuable and I will post the update!

hoangt, 2018-01-31:
Resolved by command
ssh root@localhost -p5556 (pass: sifive)

closed 2018-01-31 by hoangt