sifiveinc/freedom-u-sdk #32: An error occurred in the new version using make all

l919898756, 2018-02-15:
I downloaded the latest version, did not change Makefile, and then I used make all to build it with the following errors. 
It looks like the library file could not be found while compiling the linux kernel. Where is the path configuration error? 
How do I fix it?
[image]
[image]
@jim-wilson Hi,jim.I need your help again,thank you.
And It's the new year in China now,happy new year to you.

jim-wilson, 2018-02-15:
This is a host tool using host libraries.  So you need to install another package on your host system.  Depending on your OS, I think it should be something like libssl-dev (ubuntu) or openssl-devel (fedora). 
Maybe the new linux kernel version is enabling a feature that wasn't enabled before.

tprrt, 2022-11-15:
This issue no longer occurs.

closed 2022-11-15 by tprrt