sifiveinc/freedom-u-sdk #34: Incorrect selection of kernel headers: expected 4.6.x, got 4.13.x

michaeljclark, 2018-02-24:
Trying to build a fresh checkout of `freedom-u-sdk` using `make qemu`.

```
$ git branch -v
* master 783c89a linux_defconfig: support software restart
```

QEMU builds okay, but then I get an error with buildroot. I've done `git submodule update --init --recursive` and everything is up-to-date.

```
$ make qemu
rm -rf /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/
mkdir -p /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/
cp /home/mclark/src/sifive/freedom-u-sdk/conf/buildroot_initramfs_config /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/.config
make -C /home/mclark/src/sifive/freedom-u-sdk/buildroot RISCV=/opt/riscv/toolchain PATH=/opt/riscv/toolchain/bin:/home/mclark/bin:/home/mclark/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/opt/riscv/toolchain/bin:/opt/riscv/qemu/bin:/opt/riscv/musl-riscv-toolchain-7.2.0-7/bin O=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs olddefconfig
make[1]: Entering directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
mkdir -p /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/lxdialog
PKG_CONFIG_PATH="" make CC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc" \
    obj=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config -C support/kconfig -f Makefile.br conf
make[2]: Entering directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot/support/kconfig'
/usr/bin/gcc -D_GNU_SOURCE -D_DEFAULT_SOURCE -DCURSES_LOC="<ncurses.h>" -DLOCALE  -I/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config -DCONFIG_=\"\"  -MM *.c > /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/.depend 2>/dev/null || :
/usr/bin/gcc -D_GNU_SOURCE -D_DEFAULT_SOURCE -DCURSES_LOC="<ncurses.h>" -DLOCALE  -I/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config -DCONFIG_=\"\"   -c conf.c -o /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/conf.o
/usr/bin/gcc -D_GNU_SOURCE -D_DEFAULT_SOURCE -DCURSES_LOC="<ncurses.h>" -DLOCALE  -I/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config -DCONFIG_=\"\"  -I. -c /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/zconf.tab.c -o /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/zconf.tab.o
/usr/bin/gcc -D_GNU_SOURCE -D_DEFAULT_SOURCE -DCURSES_LOC="<ncurses.h>" -DLOCALE  -I/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config -DCONFIG_=\"\"   /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/conf.o /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/zconf.tab.o  -o /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/conf
rm /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/zconf.tab.c
make[2]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot/support/kconfig'
  GEN     /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/Makefile
BR2_DEFCONFIG='' KCONFIG_AUTOCONFIG=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/auto.conf KCONFIG_AUTOHEADER=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/autoconf.h KCONFIG_TRISTATE=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/tristate.config BR2_CONFIG=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/.config BR2_EXTERNAL=support/dummy-external HOST_GCC_VERSION="6" SKIP_LEGACY= /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/conf --olddefconfig Config.in
#
# configuration written to /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/.config
#
make[1]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
make -C /home/mclark/src/sifive/freedom-u-sdk/buildroot RISCV=/opt/riscv/toolchain PATH=/opt/riscv/toolchain/bin:/home/mclark/bin:/home/mclark/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/opt/riscv/toolchain/bin:/opt/riscv/qemu/bin:/opt/riscv/musl-riscv-toolchain-7.2.0-7/bin O=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs
make[1]: Entering directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
/usr/bin/make -j1 O=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" silentoldconfig
make[2]: Entering directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
  GEN     /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/Makefile
BR2_DEFCONFIG='' KCONFIG_AUTOCONFIG=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/auto.conf KCONFIG_AUTOHEADER=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/autoconf.h KCONFIG_TRISTATE=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/tristate.config BR2_CONFIG=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/.config BR2_EXTERNAL=support/dummy-external HOST_GCC_VERSION="6" SKIP_LEGACY= /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/buildroot-config/conf --silentoldconfig Config.in
make[2]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
>>> host-mkpasswd buildroot-2016.08-git Extracting
>>> host-mkpasswd buildroot-2016.08-git Patching
>>> host-mkpasswd buildroot-2016.08-git Configuring
>>> host-mkpasswd buildroot-2016.08-git Building
/usr/bin/gcc -O2 -I/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -L/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/lib -L/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/lib -Wl,-rpath,/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/lib package/mkpasswd/mkpasswd.c package/mkpasswd/utils.c -o /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/host-mkpasswd-buildroot-2016.08-git/mkpasswd -lcrypt
In file included from /usr/include/stdio.h:27:0,
                 from package/mkpasswd/mkpasswd.c:24:
/usr/include/features.h:148:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
>>> host-mkpasswd buildroot-2016.08-git Installing to host directory
/usr/bin/install -D -m 755 /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/host-mkpasswd-buildroot-2016.08-git/mkpasswd /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/mkpasswd
>>> skeleton undefined Extracting
>>> skeleton undefined Patching
>>> skeleton undefined Configuring
>>> skeleton undefined Building
>>> skeleton undefined Installing to staging directory
/usr/bin/install -d -m 0755 /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/riscv64-buildroot-linux-gnu/sysroot/usr/lib
/usr/bin/install -d -m 0755 /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/riscv64-buildroot-linux-gnu/sysroot/usr/bin
/usr/bin/install -d -m 0755 /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/riscv64-buildroot-linux-gnu/sysroot/usr/sbin
/usr/bin/install -d -m 0755 /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/riscv64-buildroot-linux-gnu/sysroot/usr/include
/usr/bin/install -d -m 0755 /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/riscv64-buildroot-linux-gnu/sysroot/bin
/usr/bin/install -d -m 0755 /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/riscv64-buildroot-linux-gnu/sysroot/sbin
/usr/bin/install -d -m 0755 /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/riscv64-buildroot-linux-gnu/sysroot/lib
ln -snf lib /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/riscv64-buildroot-linux-gnu/sysroot/lib64
ln -snf lib /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/riscv64-buildroot-linux-gnu/sysroot/usr/lib64
>>> skeleton undefined Fixing libtool files
>>> skeleton undefined Installing to target
[... cut ...]
package/pkg-generic.mk:185: recipe for target '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/toolchain-external-undefined/.stamp_configured' failed
make[1]: *** [/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/toolchain-external-undefined/.stamp_configured] Error 1
make[1]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
Makefile:87: recipe for target '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar' failed
make: *** [/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar] Error 2
```

michaeljclark, 2018-02-24:
I spent a lot of time mucking with buildroot in early December and coudn't get it to work due to similar sorts of errors with gcc and linux header version dependencies. I tried fixing it but I gave up hacking on buildroot and went ahead and made a tiny busybox+dropbear linux root image builder. Caveat: it requires sudo as it loop mounts an ext4 filesystem

- https://github.com/michaeljclark/busybear-linux

There is obviously some buildroot patches I must be missing or I'm on the wrong branch. When I run it a third time, the build log is shorter, but I get the same error.

```
$ make qemu
make -C /home/mclark/src/sifive/freedom-u-sdk/buildroot RISCV=/opt/riscv/toolchain PATH=/opt/riscv/toolchain/bin:/home/mclark/bin:/home/mclark/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/opt/riscv/toolchain/bin:/opt/riscv/qemu/bin:/opt/riscv/musl-riscv-toolchain-7.2.0-7/bin O=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs
make[1]: Entering directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
>>> toolchain-external undefined Configuring
Incorrect selection of kernel headers: expected 4.6.x, got 4.13.x
package/pkg-generic.mk:185: recipe for target '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/toolchain-external-undefined/.stamp_configured' failed
make[1]: *** [/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/toolchain-external-undefined/.stamp_configured] Error 1
make[1]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
Makefile:87: recipe for target '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar' failed
make: *** [/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar] Error 2
```

It's possibly my linux toolchain version. My linux toolchain is top-of-tree default master branch of `riscv-gnu-toolchain` as of 01-Feb-2018

```
$ /opt/riscv/toolchain-20180201/bin/riscv64-unknown-linux-gnu-gcc --version
riscv64-unknown-linux-gnu-gcc (GCC) 7.1.1 20170509
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

terpstra, 2018-02-24:
Try unset RISCV before running make in a clean checkout of freedom-u-sdk.
It will use it's own known-to-work toolchain then.

michaeljclark, 2018-02-25:
Oh that's good to know. I'll give it a try.

We probably should have master (default branch) in riscv-gnu-toolchain pointing at a newer toolchain. I'll try and figure out which branch freedom-u-sdk is using...

michaeljclark, 2018-02-25:
We have quite a few repos that build private copies of the toolchain (freedom, freedom-u-sdk, freedom-e-sdk, riscv-tools (if built separately) and riscv-gnu-toolchain (if built separately). Hopefully, we'll be able to fix some of these issues by using packaged dependencies instead of submodules. At some point...

@palmer-dabbelt @jim-wilson what branch of riscv-gnu-toolchain does freedom-u-sdk depend on? linux-headers-4.15-rc3? master in riscv-gnu-toolchain seems to be a little behind.

michaeljclark, 2018-02-25:
Appears to be the issue. The build is getting a lot further... it is still building...

I'll leave the issue open as its likely many folk will have `RISCV` set.

We should probably add a test for the headers if an external toolchain is being used.

We could perhaps add it to the README.md when we give it a README.md. I can make a PR...

AndreRH, 2018-03-07:
Hacky but works:
in buildroot/support/scripts/check-kernel-headers.sh change the line return 1; -> return 0;

terpstra, 2018-03-28:
the documentation now includes a comment about this gotcha

closed 2018-03-28 by terpstra