sifiveinc/freedom-u-sdk #36: How to add files or directories into bbl.bin

xubaqian, 2018-02-28:
Linux has run on my zc707 successfully, and I want to add files or directories in some subdirectories, such as /root.

When I modified the initramfs.txt  in conf subdirectory, the bbl.bin got bigger. But I couldn't see my file in linux.

When I modified the initramfs.txt  in linux/arch/riscv subdirectory and added its path through make linux-menuconfig, the bbl.bin remains invariant.

I don't know why.

I tried to modified buildroot, but don't konw how to do it.

Thanks for your help.

jim-wilson, 2018-02-28:
Buildroot is used to build the target filesystem, so you will have to spend some time looking at that.  There are multiple ways to modify it.  Using BR2_ROOTFS_OVERLAY="<filenames>" looks like one of the easier approaches.  Add it to config/buildroot-initramfs_config or conf/buildroot_roorfs_config depending on which configuration you are using, or add it to both if you aren't sure.

See for instance
    https://buildroot.uclibc.org/downloads/manual/manual.html#rootfs-custom
You should probably read other parts of the manual also.

I'm not a buildroot expert, and I've never done this myself, so I can't add any specific info.

xubaqian, 2018-03-01:
I didn't see  config/buildroot-initramfs_config or conf/buildroot_roorfs_config
but modified the line  BR2_ROOTFS_OVERLAY="/home/xqz/mybin"  in  work/buildroot/.config 

```
cd work/buildroot
make
cd ../..
make
```

then the bbl.bin got bigger and in the work/sysroot I can see the files I added

However, on zc707, it stopped after the logo appeared.

```
                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           555555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Coreplex

```

terpstra, 2018-03-01:
If you are using sdboot, beware that it only downloads 16MiB. If your image exceeds that limit, it won't boot.

xubaqian, 2018-03-01:
Success ! But can we raise its download limit ？

jim-wilson, 2018-03-01:
sdboot seems to be a reference to a Xilinx board feature, so I don't know anything about that.

However, if you are booting from sdcard, then you probably have two partitions on your sdcard, and you should be able to put a linux filesystem on the second partition and mount it.  You can then easily add as many files as you want to this linux filesystem.  You could also put a linux disk image on it and chroot into it.

terpstra, 2018-03-01:
@jim-wilson is mistaken. He's thinking of the system we are using internally for the HiFive/Unleashed board and have not yet ported to the freedom repository.

closed 2018-03-02 by xubaqian