sifiveinc/freedom-u-sdk #44: no console output after bootconsole disabled.

xfguo, 2018-03-22:
Just found that there is no output after bootconsole disabled by kernel.

```
$ make qemu
/data/riscv/freedom-u-sdk/work/riscv-qemu/prefix/bin/qemu-system-riscv64 -nographic -machine virt -kernel /data/riscv/freedom-u-sdk/work/riscv-pk/bbl \
	-drive file=/data/riscv/freedom-u-sdk/work/rootfs.bin,format=raw,id=hd0 -device virtio-blk-device,drive=hd0 \
	-netdev user,id=net0 -device virtio-net-device,netdev=net0
bbl loader

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Coreplex
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00044-g2b0aa1d (alex@alex-x230) (gcc version 7.2.0 (GCC)) #7 SMP Thu Mar 22 11:16:07 CST 2018
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (5672960 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000]   Normal   [mem 0x0000000088000000-0x0000087fffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] software IO TLB [mem 0x83e3d000-0x87e3d000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 14 pages/cpu @        (ptrval) s28632 r0 d28712 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 31815
[    0.000000] Kernel command line: earlyprintk
[    0.000000] Dentry cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 4, 65536 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 50372K/129024K available (3055K kernel code, 214K rwdata, 827K rodata, 5695K init, 775K bss, 78652K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU event tracing is enabled.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,plic0,c000000: mapped 10 interrupts to 1/2 handlers
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000193] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.004177] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.006901] pid_max: default: 32768 minimum: 301
[    0.009112] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.010731] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.036960] Hierarchical SRCU implementation.
[    0.046558] smp: Bringing up secondary CPUs ...
[    0.047811] smp: Brought up 1 node, 1 CPU
[    0.084107] devtmpfs: initialized
[    0.093955] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.096091] futex hash table entries: 256 (order: 2, 16384 bytes)
[    0.102939] random: get_random_u32 called from bucket_table_alloc+0xe6/0x276 with crng_init=0
[    0.105357] NET: Registered protocol family 16
[    0.139454] vgaarb: loaded
[    0.141203] SCSI subsystem initialized
[    0.144195] usbcore: registered new interface driver usbfs
[    0.145932] usbcore: registered new interface driver hub
[    0.147567] usbcore: registered new device driver usb
[    0.149665] pps_core: LinuxPPS API ver. 1 registered
[    0.150879] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.153130] PTP clock support registered
[    0.161982] clocksource: Switched to clocksource riscv_clocksource
[    0.170369] NET: Registered protocol family 2
[    0.175418] TCP established hash table entries: 1024 (order: 1, 8192 bytes)
[    0.177069] TCP bind hash table entries: 1024 (order: 2, 16384 bytes)
[    0.178490] TCP: Hash tables configured (established 1024 bind 1024)
[    0.181621] UDP hash table entries: 256 (order: 1, 8192 bytes)
[    0.182795] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes)
[    0.185023] NET: Registered protocol family 1
[    0.448495] Unpacking initramfs...
[    0.745703] Initialise system trusted keyrings
[    0.748555] workingset: timestamp_bits=62 max_order=14 bucket_order=0
[    0.774849] random: fast init done
[    0.805829] Key type asymmetric registered
[    0.806810] Asymmetric key parser 'x509' registered
[    0.807919] io scheduler noop registered
[    0.809979] io scheduler cfq registered (default)
[    0.811298] io scheduler mq-deadline registered
[    0.813097] io scheduler kyber registered
[    1.027782] libphy: Fixed MDIO Bus: probed
[    1.034595] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    1.036306] ehci-pci: EHCI PCI platform driver
[    1.038008] usbcore: registered new interface driver usb-storage
[    1.041234] usbcore: registered new interface driver usbhid
[    1.043040] usbhid: USB HID core driver
[    1.044963] NET: Registered protocol family 17
[    1.048917] bootconsole [early0] uses init memory and must be disabled even before the real one is ready
[    1.051654] bootconsole [early0] disabled

```

michaeljclark, 2018-03-22:
I can't see the UART initialization in the console log. You should see something like this:

```
Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
10000000.uart: ttyS0 at MMIO 0x10000000 (irq = 13, base_baud = 230400) is a 16550A
```

Perhaps the kernel config has changed an no longer has the 16550A UART. You need these config options in the kernel `.config`

```
CONFIG_SERIAL_8250=y
CONFIG_SERIAL_8250_CONSOLE=y
CONFIG_SERIAL_OF_PLATFORM=y
```

mirzak, 2018-03-22:
Recently experienced something similar and updating the getty baudrate from 38400 to 115200 solved it.

terpstra, 2018-03-22:
Ahh. This is probably my fault. I removed 8250 UART as it's not on the Unleashed board and we had passed the 3-byte SPI 16MiB threshold.

PR #46 (2018-03-23): linux_defconfig: restore support for qemu's uart
Now that we've fixed the 16MiB limit on HiFive Unleashed, there is
no need to drop this drive any more.

Fixes #44.
commit 858adf9e (2018-03-23): linux_defconfig: restore support for qemu's uart (#46)

Now that we've fixed the 16MiB limit on HiFive Unleashed, there is
no need to drop this drive any more.

Fixes #44.
closed 2018-03-23 by terpstra