sifiveinc/freedom-u-sdk #47: Running bbl image from freedom-u-sdk with fesvr

lucashmorais, 2018-03-28:
The Linux image generated from freedom-u-sdk tag hifiveu-0.0 runs nicely on spike with 4 cores, but I wasn't able to use the corresponding bbl image, which is also automatically generated, for booting Linux on top of a single-core Rocket Chip running on a ZC706.

In light of this, should I make any modification to the Kernel config or to any other tool to boot that same spike-working image on an FPGA?

I've already made sure that riscv-pk and riscv-fesvr are updated and that the /usr/local/lib path on the board holds the new version of libfesvr.so. Running `$./riscv-fesvr pk hello` on the board also works as intended.

terpstra, 2018-03-28:
How are you loading the output to the FPGA? The instructions for programming an SD card rely on a boot loader that is not yet part of the freedom FPGA boot ROM.

terpstra, 2018-03-28:
You should take work/bbl.bin and write it directly to the SD card (destroying the partition table) for the current FPGA boot flow.

closed 2022-11-15 by tprrt