sifiveinc/freedom-u-sdk #48: linx doesn't boot on Spike

noureddine-as, 2018-03-28:
Hello,

I'm trying to boot Linux on Spike with -p4 using **make sim**
It has booted but doesn't log in to busybox.
Any idea where the problem did come from? I waited a long time but no reaction. 

Best regards,
Noureddine
```
                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Coreplex

[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00044-g2b0aa1d (noureddine-as@as) (gcc version 7.2.0 (GCC)) #1 SMP Wed Mar 28 11:21:01 CEST 2018
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (11025408 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000fffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] software IO TLB [mem 0xfa3fd000-0xfe3fd000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x1129
[    0.000000] percpu: Embedded 14 pages/cpu @        (ptrval) s28632 r0 d28712 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516615
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 1980032K/2095104K available (4309K kernel code, 219K rwdata, 844K rodata, 10963K init, 780K bss, 115072K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU event tracing is enabled.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,1: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,2: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,3: 64 local interrupts mapped
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000010] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.000995] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.002200] pid_max: default: 32768 minimum: 301
[    0.002805] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.003590] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.005355] Hierarchical SRCU implementation.
[    0.007430] smp: Bringing up secondary CPUs ...
[    0.008570] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.009880] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.011640] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.012950] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.014690] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.016000] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.017145] smp: Brought up 1 node, 4 CPUs
[    0.018075] devtmpfs: initialized
[    0.019020] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.020160] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.021110] random: get_random_u32 called from bucket_table_alloc+0x134/0x3c0 with crng_init=0
[    0.022170] NET: Registered protocol family 16
[    0.025470] vgaarb: loaded
[    0.025910] SCSI subsystem initialized
[    0.026645] usbcore: registered new interface driver usbfs
[    0.027295] usbcore: registered new interface driver hub
[    0.028060] usbcore: registered new device driver usb
[    0.028675] pps_core: LinuxPPS API ver. 1 registered
[    0.029255] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.030320] PTP clock support registered
[    0.031210] clocksource: Switched to clocksource riscv_clocksource
[    0.033130] NET: Registered protocol family 2
[    0.033920] TCP established hash table entries: 16384 (order: 5, 131072 bytes)
[    0.034960] TCP bind hash table entries: 16384 (order: 6, 262144 bytes)
[    0.035925] TCP: Hash tables configured (established 16384 bind 16384)
[    0.036720] UDP hash table entries: 1024 (order: 3, 32768 bytes)
[    0.037455] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)
[    0.038290] NET: Registered protocol family 1
[    0.062985] Unpacking initramfs...
[    0.091610] Initialise system trusted keyrings
[    0.092325] workingset: timestamp_bits=62 max_order=19 bucket_order=0
[    0.097310] random: fast init done
[    0.111115] jitterentropy: Initialization failed with host not compliant with requirements: 2
[    0.112220] Key type asymmetric registered
[    0.112700] Asymmetric key parser 'x509' registered
[    0.113270] io scheduler noop registered
[    0.113830] io scheduler cfq registered (default)
[    0.114380] io scheduler mq-deadline registered
[    0.114910] io scheduler kyber registered
[    0.173790] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    0.175490] libphy: Fixed MDIO Bus: probed
[    0.176045] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.176805] ehci-pci: EHCI PCI platform driver
[    0.177360] usbcore: registered new interface driver usb-storage
[    0.178195] usbcore: registered new interface driver usbhid
[    0.178845] usbhid: USB HID core driver
[    0.179455] NET: Registered protocol family 17
[    0.180435] bootconsole [early0] uses init memory and must be disabled even before the real one is ready
[    0.181645] bootconsole [early0] disabled

```

terpstra, 2018-03-28:
So the problem is that you need to remove this line from conf/linux_defconfig:
# CONFIG_HVC_RISCV_SBI is not set

However, if that line is removed, on our forth-coming hardware board the SBI console driver takes over the console instead of the SiFive serial console driver. The later knows how to handle frequency changes and interrupts, so is a lot better in real use.

@palmer-dabbelt do you know how to make linux prefer the SiFive console over the SBI console, so we can have both enabled at the same time?

lucashmorais, 2018-03-29:
What commit are you working with?

If you don't mind using an older version of the tools, `make sim` works fine with tag hifiveu-0.0, from March 6th.

noureddine-as, 2018-03-30:
@terpstra @lucashmorais 
Thank you for answering, actually when I got the problem I was in the master branch. I've moved to hifive-unleashed branch, rebuild everything and it logs in to ash now.

Thanks for your help.
Best regards.

closed 2018-03-30 by noureddine-as