sifiveinc/freedom-u-sdk #49: Issue with creating Sifive risc-v port

billasurendra, 2018-04-03:
I am executing a **$ make Sim** command.but i got  error like this **bootconsole [early0] disabled
** .
Can any one please give me solution for this

/home/billa/Sifive/freedom-u-sdk/work/riscv-isa-sim/prefix/bin/spike --isa=rv64imafdc -p4 /home/billa/Sifive/freedom-u-sdk/work/riscv-pk/bbl
bbl loader

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Coreplex
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00044-g2b0aa1d (billa@billa-PC) (gcc version 7.2.0 (GCC)) #2 SMP Mon Apr 2 15:26:12 IST 2018
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (9436160 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000fffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] software IO TLB [mem 0xfa3fd000-0xfe3fd000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 14 pages/cpu @        (ptrval) s28632 r0 d28712 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516615
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 1982864K/2095104K available (3073K kernel code, 212K rwdata, 840K rodata, 9379K init, 780K bss, 112240K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU event tracing is enabled.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,1: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,2: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,3: 64 local interrupts mapped
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000010] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.000995] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.002200] pid_max: default: 32768 minimum: 301
[    0.002805] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.003590] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.005355] Hierarchical SRCU implementation.
[    0.007430] smp: Bringing up secondary CPUs ...
[    0.008570] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.009880] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.011640] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.012950] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.014690] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.016000] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.017145] smp: Brought up 1 node, 4 CPUs
[    0.018075] devtmpfs: initialized
[    0.019020] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.020160] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.021110] random: get_random_u32 called from bucket_table_alloc+0xe6/0x276 with crng_init=0
[    0.022160] NET: Registered protocol family 16
[    0.025460] vgaarb: loaded
[    0.025900] SCSI subsystem initialized
[    0.026635] usbcore: registered new interface driver usbfs
[    0.027285] usbcore: registered new interface driver hub
[    0.028050] usbcore: registered new device driver usb
[    0.028665] pps_core: LinuxPPS API ver. 1 registered
[    0.029245] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.030310] PTP clock support registered
[    0.031200] clocksource: Switched to clocksource riscv_clocksource
[    0.033115] NET: Registered protocol family 2
[    0.033905] TCP established hash table entries: 16384 (order: 5, 131072 bytes)
[    0.034945] TCP bind hash table entries: 16384 (order: 6, 262144 bytes)
[    0.035910] TCP: Hash tables configured (established 16384 bind 16384)
[    0.036705] UDP hash table entries: 1024 (order: 3, 32768 bytes)
[    0.037440] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)
[    0.038275] NET: Registered protocol family 1
[    0.060930] Unpacking initramfs...
[    0.087145] Initialise system trusted keyrings
[    0.087860] workingset: timestamp_bits=62 max_order=19 bucket_order=0
[    0.093905] random: fast init done
[    0.108730] jitterentropy: Initialization failed with host not compliant with requirements: 2
[    0.109720] Key type asymmetric registered
[    0.110200] Asymmetric key parser 'x509' registered
[    0.110780] io scheduler noop registered
[    0.111430] io scheduler cfq registered (default)
[    0.111980] io scheduler mq-deadline registered
[    0.112510] io scheduler kyber registered
[    0.170965] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    0.172820] libphy: Fixed MDIO Bus: probed
[    0.173370] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.174130] ehci-pci: EHCI PCI platform driver
[    0.174690] usbcore: registered new interface driver usb-storage
[    0.175525] usbcore: registered new interface driver usbhid
[    0.176175] usbhid: USB HID core driver
[    0.176795] NET: Registered protocol family 17
[    0.177780] bootconsole [early0] uses init memory and must be disabled even before the real one is ready
[    0.178880] bootconsole [early0] disabled

billasurendra, 2018-04-03:
when i am using **$ make qemu** i got another error like this

rm -rf /home/billa/Sifive/freedom-u-sdk/work/riscv-qemu
mkdir -p /home/billa/Sifive/freedom-u-sdk/work/riscv-qemu
mkdir -p /home/billa/Sifive/freedom-u-sdk/work/riscv-qemu/prefix/bin/
cd /home/billa/Sifive/freedom-u-sdk/work/riscv-qemu && /home/billa/Sifive/freedom-u-sdk/riscv-qemu/configure \
	--prefix=/home/billa/Sifive/freedom-u-sdk/work/riscv-qemu/prefix/ \
	--target-list=riscv64-softmmu

ERROR: glib-2.22 gthread-2.0 is required to compile QEMU

Makefile:189: recipe for target '/home/billa/Sifive/freedom-u-sdk/work/riscv-qemu/prefix/bin/qemu-system-riscv64' failed
make: *** [/home/billa/Sifive/freedom-u-sdk/work/riscv-qemu/prefix/bin/qemu-system-riscv64] Error 1

jim-wilson, 2018-04-03:
Did you try installing the missing packages?  Or maybe you are using a linux system that is too old to support building current qemu?  If so, then maybe the qemu in riscv-gnu-toolchain/riscv-qemu will build OK.

billasurendra, 2018-04-04:
yes your correct Jim, Now it's working..
Thank you so much

closed 2018-04-04 by jim-wilson