sifiveinc/freedom-u-sdk #52: buildroot compile failure on Debian Buster (testing)

michaeljclark, 2018-04-21:
I am having a problem building buildroot when running `make qemu` with top-of-tree on Debian Linux x86_64 (Note I have done `unset RISCV` so that I use the toolchain bundled with `freedom-u-sdk`):

```
$ make qemu
[snip]
make[4]: Entering directory '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/host-e2fsprogs-1.43.1/debugfs'
	CC debug_cmds.c
	CC extent_cmds.c
	CC create_inode.o
	CC quota.c
	CC xattrs.c
	CC journal.c
./../misc/create_inode.c: In function ‘do_mknod_internal’:
./../misc/create_inode.c:290:13: warning: In the GNU C Library, "major" is defined
 by <sys/sysmacros.h>. For historical compatibility, it is
 currently defined by <sys/types.h> as well, but we plan to
 remove this soon. To use "major", include <sys/sysmacros.h>
 directly. If you did not intend to use a system-defined macro
 "major", you should undefine it after including <sys/types.h>.
   devmajor = major(st->st_rdev);
             ^~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                                                                                                                                    
./../misc/create_inode.c:291:13: warning: In the GNU C Library, "minor" is defined
 by <sys/sysmacros.h>. For historical compatibility, it is
 currently defined by <sys/types.h> as well, but we plan to
 remove this soon. To use "minor", include <sys/sysmacros.h>
 directly. If you did not intend to use a system-defined macro
 "minor", you should undefine it after including <sys/types.h>.
   devminor = minor(st->st_rdev);
             ^~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                                                                                                                                    
./../misc/create_inode.c: At top level:
./../misc/create_inode.c:395:18: error: conflicting types for ‘copy_file_range’
 static errcode_t copy_file_range(ext2_filsys fs, int fd, ext2_file_t e2_file,
                  ^~~~~~~~~~~~~~~
In file included from ./../misc/create_inode.c:19:0:
/usr/include/unistd.h:1110:9: note: previous declaration of ‘copy_file_range’ was here
 ssize_t copy_file_range (int __infd, __off64_t *__pinoff,
         ^~~~~~~~~~~~~~~
Makefile:417: recipe for target 'create_inode.o' failed
make[4]: *** [create_inode.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/host-e2fsprogs-1.43.1/debugfs'
Makefile:404: recipe for target 'all-progs-recursive' failed
make[3]: *** [all-progs-recursive] Error 1
make[3]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/host-e2fsprogs-1.43.1'
Makefile:339: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/host-e2fsprogs-1.43.1'
package/pkg-generic.mk:195: recipe for target '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/host-e2fsprogs-1.43.1/.stamp_built' failed
make[1]: *** [/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/host-e2fsprogs-1.43.1/.stamp_built] Error 2
make[1]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
Makefile:90: recipe for target '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar' failed
make: *** [/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar] Error 2
```

I have done a `make clean` and `git fetch`, `git reset origin/master` and recursively reset submodules. Below is the output excluding riscv-qemu. 

```
$ git branch -v
* master                62ae153 Merge pull request #50 from sifive/add_u500_openocd_cfg
$ git submodule foreach --recursive git reset --hard
Entering 'buildroot'
HEAD is now at dfc1531953 Add tyrquake, an open source quake 1 implementation
Entering 'linux'
HEAD is now at 2b0aa1de45f6 serial/sifive: initial driver from Paul Walmsley
Entering 'riscv-fesvr'
HEAD is now at f683e01 dtm: Avoid back-to-back writes to dmcontrol (#35)
Entering 'riscv-gnu-toolchain'
HEAD is now at b4dae89 Bump GCC
Entering 'riscv-gnu-toolchain/riscv-binutils-gdb'
HEAD is now at d0176cb165 RISC-V: Fix riscv g++ testsuite EH failures.
[... 168 lines cut ...]
(cd /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/libtirpc-1.0.1/ && rm -rf config.cache && PATH="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/bin:/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/sbin:/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin:/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/sbin:/home/mclark/src/sifive/freedom-u-sdk/toolchain/bin:/home/mclark/bin:/home/mclark/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/opt/riscv/toolchain/bin:/opt/riscv/qemu/bin:/opt/riscv/musl-riscv-toolchain-7.2.0-7/bin" AR="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-ar" AS="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-as" LD="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-ld" NM="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-nm" CC="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-gcc" GCC="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-gcc" CPP="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-cpp" CXX="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-g++" FC="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-gfortran" RANLIB="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-ranlib" READELF="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-readelf" STRIP="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-strip" OBJCOPY="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-objcopy" OBJDUMP="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-objdump" AR_FOR_BUILD="/usr/bin/ar" AS_FOR_BUILD="/usr/bin/as" CC_FOR_BUILD="/usr/bin/gcc" GCC_FOR_BUILD="/usr/bin/gcc" CXX_FOR_BUILD="/usr/bin/g++" LD_FOR_BUILD="/usr/bin/ld" CPPFLAGS_FOR_BUILD="-I/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/include" CFLAGS_FOR_BUILD="-O2 -I/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/include" CXXFLAGS_FOR_BUILD="-O2 -I/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/include" LDFLAGS_FOR_BUILD="-L/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/lib -L/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/lib -Wl,-rpath,/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/lib" FCFLAGS_FOR_BUILD="" DEFAULT_ASSEMBLER="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-as" DEFAULT_LINKER="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/riscv64-unknown-linux-gnu-ld" CPPFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" CFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os " CXXFLAGS="-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -Os " LDFLAGS="" FCFLAGS="" PKG_CONFIG="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/pkg-config" STAGING_DIR="/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/riscv64-buildroot-linux-gnu/sysroot" INTLTOOL_PERL=/usr/bin/perl ac_cv_lbl_unaligned_fail=yes ac_cv_func_mmap_fixed_mapped=yes ac_cv_func_memcmp_working=yes [... cut ...]
make[1]: *** [/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/libtirpc-1.0.1/.stamp_configured] Error 77
make[1]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
Makefile:90: recipe for target '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar' failed
make: *** [/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar] Error 2
```

Hmmm. Frustrating. Let's get back onto the new gcc-7.3 compiler and glibc-2.27 in Debian Buster.

To be continued...

jim-wilson, 2018-04-21:
Looks like copy_file_range is a relatively new system call.  In the glibc release notes I see
Version 2.27
...
* The copy_file_range function was added.

Ubuntu 17.10 is still using glibc-2.26, so we haven't seen this conflict yet.  Looks like a qemu internal function will have to be renamed to avoid the conflict.

jim-wilson, 2018-04-21:
Oops, that is e2fsprogs that needs to be fixed.  Looking at e2fsprogs release notes
http://e2fsprogs.sourceforge.net/e2fsprogs-release.html#1.44.1
I see
E2fsprogs 1.43.8 (January 1, 2018)
...
Fix compatibility problems with glibc 2.27 casued by a collision with copy_file_range().

Buildroot is using 1.43.1, see buildroot/package/e2fsprogs/e2fsprogs.*.
There may be a buildroot patch for that somewhere.

michaeljclark, 2018-04-21:
Ya. I've set up an Ubuntu 16.04 LTS amd64 VM and am trying to build again now...

I updated my Debian back to Buster and now I'm getting a new error building libtirpc:

```
$ make qemu
make -C /home/mclark/src/sifive/freedom-u-sdk/buildroot RISCV=/home/mclark/src/sifive/freedom-u-sdk/toolchain PATH=/home/mclark/src/sifive/freedom-u-sdk/toolchain/bin:/home/mclark/bin:/home/mclark/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/opt/riscv/toolchain/bin:/opt/riscv/qemu/bin:/opt/riscv/musl-riscv-toolchain-7.2.0-7/bin O=/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs
make[1]: Entering directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
>>> libtirpc 1.0.1 Configuring
>>> libtirpc 1.0.1 Autoreconfiguring
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
Unescaped left brace in regex is illegal here in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/automake line 3936.
autoreconf: /home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/host/usr/bin/automake failed with exit status: 255
package/pkg-generic.mk:185: recipe for target '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/libtirpc-1.0.1/.stamp_configured' failed
make[1]: *** [/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/build/libtirpc-1.0.1/.stamp_configured] Error 1
make[1]: Leaving directory '/home/mclark/src/sifive/freedom-u-sdk/buildroot'
Makefile:90: recipe for target '/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar' failed
make: *** [/home/mclark/src/sifive/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar] Error 2
```

michaeljclark, 2018-04-21:
It's a warning that has become an error in perl 5.26 

- https://unix.stackexchange.com/questions/238539/automake-error-unescaped-left-brace-in-regex-is-deprecated

```
$ perl --version

This is perl 5, version 26, subversion 1 (v5.26.1) built for x86_64-linux-gnu-thread-multi
(with 62 registered patches, see perl -V for more detail)

Copyright 1987-2017, Larry Wall

Perl may be copied only under the terms of either the Artistic License or the
GNU General Public License, which may be found in the Perl 5 source kit.

Complete documentation for Perl, including FAQ lists, should be found on
this system using "man perl" or "perldoc perl".  If you have access to the
Internet, point your browser at http://www.perl.org/, the Perl Home Page.
```

michaeljclark, 2018-04-21:
Appears to be broken on Ubuntu 18.04 LTS also

- https://github.com/raspberrypi/noobs/issues/470

michaeljclark, 2018-04-21:
Ah. I followed the advice on that bug, by manually editing automake and it now works. It seems there must be a new automake dependency that is needed by buildroot to work on latest perl (5.26).

```
$ vi work/buildroot_initramfs/host/usr/bin/automake
goto line :3936 and change
$text =~ s/\${([^ \t=:+{}]+)}/substitute_ac_subst_variables_worker ($1)/ge;
to
$text =~ s/\$\{([^ \t=:+\{\}]+)\}/substitute_ac_subst_variables_worker ($1)/ge;
```

jim-wilson, 2018-04-21:
I ran into the automake problem once building on my laptop.  I guess I forgot to submit a patch for it before deleting the build tree.  Editing the automake script isn't a good solution as it can be overwritten on a rebuild.  A better fix is to drop a small patch file into
    buildroot/package/automake
There is already a 0001-noman.patch file there.  Add a 0002-whatever.patch file that fixes automake and it will be used on the next build.

michaeljclark, 2018-04-21:
Okay I have fixed both issues. I have apull is against the sifive buildroot repo. It is parented on the commit: https://github.com/sifive/buildroot/commit/dfc15319 which is where the freedom-u-sdk submodule is pointing. We probably need to check whether these issues have been fixed in upstream buildroot.

See: https://github.com/sifive/buildroot/pull/5

- Bump e2fsprogs to 1.44.1 to fix issue with glibc 2.27
    
    glibc 2.27 introduces a copy_file_range function that
    clashes with a static function in e2fsprogs 1.43.1
    causing a compile error. It This has been fixed in
    e2fsprogs 1.43.8. Bump to latest e2fsprogs 1.44.1
    
    - http://e2fsprogs.sourceforge.net/e2fsprogs-release.html#1.43.8

- Bump automake to 1.16.1 to fix issue with perl 5.26
    
    Fixes "Unescaped left brace in regex is deprecated, passed
    through in regex" error message when using perl 5.26
    
    - https://lists.gnu.org/archive/html/bug-automake/2017-06/msg00003.html

michaeljclark, 2018-04-21:
This is on a glib-2.27 system with perl-5.26. Two separate issues 😀 

```
$make qemu
[snip]
/home/mclark/src/sifive/freedom-u-sdk/work/riscv-qemu/prefix/bin/qemu-system-riscv64 -nographic -machine virt -kernel /home/mclark/src/sifive/freedom-u-sdk/work/riscv-pk/bbl \
	-drive file=/home/mclark/src/sifive/freedom-u-sdk/work/rootfs.bin,format=raw,id=hd0 -device virtio-blk-device,drive=hd0 \
	-netdev user,id=net0 -device virtio-net-device,netdev=net0
bbl loader

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Coreplex
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00044-g2b0aa1de45f6 (mclark@minty) (gcc version 7.2.0 (GCC)) #1 SMP Sat Apr 21 16:18:58 NZST 2018
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (9457664 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000]   Normal   [mem 0x0000000088000000-0x0000087fffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] software IO TLB [mem 0x83e3d000-0x87e3d000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 14 pages/cpu @        (ptrval) s28632 r0 d28712 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 31815
[    0.000000] Kernel command line: earlyprintk
[    0.000000] Dentry cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 4, 65536 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 46628K/129024K available (3073K kernel code, 214K rwdata, 842K rodata, 9399K init, 780K bss, 82396K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU event tracing is enabled.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,plic0,c000000: mapped 10 interrupts to 1/2 handlers
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000145] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.003237] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.005254] pid_max: default: 32768 minimum: 301
[    0.006899] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.008167] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.026616] Hierarchical SRCU implementation.
[... 42 lines cut ...]
[    1.045980] ehci-pci: EHCI PCI platform driver
[    1.046548] usbcore: registered new interface driver usb-storage
[    1.047886] usbcore: registered new interface driver usbhid
[    1.048099] usbhid: USB HID core driver
[    1.048776] NET: Registered protocol family 17
[    1.050801] Loading compiled-in X.509 certificates
[    1.065808] Freeing unused kernel memory: 9396K
[    1.066064] This architecture does not have kernel memory protection.
Starting logging: OK
Starting mdev...
sort: /sys/devices/platform/Fixed: No such file or directory
modprobe: can't change directory to '/lib/modules': No such file or directory
Initializing random number generator... done.
Starting network...
udhcpc (v1.24.2) started
Sending discover...
Sending select for 10.0.2.15...
Lease of 10.0.2.15 obtained, lease time 86400
deleting routers
adding dns 10.0.2.3
Starting dropbear sshd: OK

Welcome to Buildroot
buildroot login: 
```

michaeljclark, 2018-04-21:
This has been bothering me for some time because I was not able to test riscv-qemu in freedom-u-sdk on my build machine which is Debian 10 (Buster):

```
$ gcc --version
gcc (Debian 7.3.0-16) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
and

```
$ /lib/x86_64-linux-gnu/libc.so.6 --version
GNU C Library (Debian GLIBC 2.27-3) stable release version 2.27.
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 7.3.0.
libc ABIs: UNIQUE IFUNC
For bug reporting instructions, please see:
<http://www.debian.org/Bugs/>.
```

michaeljclark, 2018-04-21:
- automake-1.15 is from 2015-01-05 (the version in SiFive's buildroot)
- fix to automake for perl was made in 2016-03-31
  - https://git.savannah.gnu.org/cgit/automake.git/commit/?id=13f00eb4493c217269b76614759e452d8302955e
- automake-1.15.1.tar.xz is from 2017-06-19

So from the timeline it looks like automake 1.15.1 might have the perl fix.

atishp04, 2018-05-05:
I am facing same issue on ubuntu 18.04.
After applying the fixes @michaeljclark suggested I am getting this error.

Makefile.legacy:9: *** "You have legacy configuration in your .config! Please check your configuration.".  Stop.

jim-wilson, 2018-05-06:
I just did a build on Ubuntu 18.04.  The only extra step I did was to cd into buildroot, and checkout the sifive branch, to get the fixes that Michael Clark checked in, along with some other misc fixes.

If you have the necessary patches, maybe your build failed because of corruption in the buildroot build tree.  I'd suggest deleting the buildroot build tree that failed to try building it again from scratch.  If you tried manually patching buildroot, try checking out the sifive branch instead.

atishp04, 2018-05-07:
That worked. @jim-wilson : Thanks!!

closed 2022-11-15 by tprrt