sifiveinc/freedom-u-sdk #56: failed to boot linux

xubaqian, 2018-05-15:
freedom commit id cd9a525a662b9eaf27c14533159d723efcf4c184
freedom-u-sdk commit id 62ae153449574c6d9a9f83510e735ef632a3ef8b
vivado 2016.4
Nothing went wrong during compilation. But when I boot linux, error message occurred

```
INIT
CMD0
CMD8
ACMD41
CMD58
CMD16
CMD18
LOADING  
BOOT
bbl loader

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Coreplex
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00044-g2b0aa1d (xqz@cas) (gcc version 7.2.0 (GCC)) #3 SMP Tue May 15 15:34:10 CST 2018
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (9436160 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000]   Normal   [mem 0x00000000c0000000-0x00000bffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] software IO TLB [mem 0xbb1fb000-0xbf1fb000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 14 pages/cpu @        (ptrval) s28632 r0 d28712 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258055
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 950092K/1046528K available (3110K kernel code, 215K rwdata, 856K rodata, 9379K init, 780K bss, 96436K reserved, )
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  RCU event tracing is enabled.
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,1: 64 local interrupts mapped
[... 61 lines cut ...]
[   15.854437] [<        (ptrval)>] do_one_initcall+0x6c/0xf8
[   15.859921] [<        (ptrval)>] kernel_init_freeable+0x126/0x1be
[   15.866077] [<        (ptrval)>] kernel_init+0xe/0xf0
[   15.871079] [<        (ptrval)>] ret_from_syscall+0xa/0xe
[   15.876871] ---[ end trace 28977a436e0715df ]---
[   15.884523] NET: Registered protocol family 16
[   15.889207] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000008
[   15.897452] Oops [#1]
[   15.899770] CPU: 3 PID: 1 Comm: swapper/3 Tainted: G        W        4.15.0-00044-g2b0aa1d #3
[   15.908256] sepc: ffffffe000b97290 ra : ffffffe000b97bfc sp : ffffffe03a679be0
[   15.915470]  gp : ffffffe000d3ea80 tp : ffffffe03a670000 t0 : ffffffe03a6460c0
[   15.922690]  t1 : 0000000200000022 t2 : ffffffe03a679c70 s0 : ffffffe03a679c00
[   15.929905]  s1 : ffffffe03a679c28 a0 : 0000000000000010 a1 : 0000000000000002
[   15.937126]  a2 : 0000000000000000 a3 : ffffffe03a679c28 a4 : 0000000000000000
[   15.944354]  a5 : 0000000000000000 a6 : ffffffe000deb770 a7 : 000000003a661000
[   15.951568]  s2 : 0000000000000002 s3 : 0000000000000000 s4 : 0000000000000000
[   15.958800]  s5 : ffffffe000d33430 s6 : ffffffe000d40050 s7 : 000000000000001f
[   15.966024]  s8 : ffffffe000c75270 s9 : 00000000000000d8 s10: ffffffe000d3eb68
[   15.973238]  s11: 0000000000000000 t3 : 000000000001e3c8 t4 : 000000000001e3c8
[   15.980446]  t5 : 0000000000000000 t6 : ffffffe03a646095
[   15.985774] sstatus: 0000000200000120 sbadaddr: 0000000000000008 scause: 000000000000000d
[   15.994327] ---[ end trace 28977a436e0715e0 ]---
[   15.998715] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[   15.998715] 
```

terpstra, 2018-05-15:
Right. This is probably because the current kernel does not support the DTS output by the current freedom repository. We need to fix this.

terpstra, 2018-05-15:
The bug is actually in the freedom repository, which we are also currently working on, so we will fix it there in the next week.

xubaqian, 2018-05-21:
@terpstra Thanks, sir. I found the newest freedom [commits](https://github.com/sifive/freedom/tree/6e0d7df069e262a16a43eab270d159acbd890ad5) on May 16, 2018 still didn't work. Would you please remind me when you fix this bug?  Thanks again!

terpstra, 2018-05-21:
Yeah, sorry for my overly optimistic time estimate. It seems this is probably not going to happen immediately. If you want to diagnose it yourself, try to find out which function the kernel is crashing in from the sepc register. We would like to ideally convert our existing boot process from the HiFive Unleashed boards to also work on the FPGA, but I'm not sure when someone will take up this task.

xubaqian, 2018-05-22:
@terpstra OK. So is there any earlier commit available? Even in 2017.

xubaqian, 2018-05-23:
@terpstra Well, I have found [one](https://github.com/sifive/freedom/pull/40) and it works. Thanks

closed 2018-05-23 by xubaqian