sifiveinc/freedom-u-sdk #61: "make all" Failure

atishp04, 2018-06-25:
Hi,
I am getting following compilation error  after a git update.
Any suggestions?
         ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/atish/freedom-u-sdk/toolchain/sysroot/usr/include/features.h:447:0,
                 from /home/atish/freedom-u-sdk/toolchain/sysroot/usr/include/bits/libc-header-start.h:33,
                 from /home/atish/freedom-u-sdk/toolchain/sysroot/usr/include/stdint.h:26,
                 from /home/atish/freedom-u-sdk/toolchain/lib/gcc/riscv64-unknown-linux-gnu/7.2.0/include/stdint.h:9,
                 from /home/atish/freedom-u-sdk/riscv-pk/softfloat/f32_to_i64.c:3:
/home/atish/freedom-u-sdk/toolchain/sysroot/usr/include/gnu/stubs.h:17:11: fatal error: gnu/stubs-lp64.h: No such file or directory
 # include <gnu/stubs-lp64.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:320: recipe for target 'f32_to_i64_r_minMag.o' failed
make[1]: *** [f32_to_i64_r_minMag.o] Error 1
Makefile:320: recipe for target 'f32_sqrt.o' failed
make[1]: *** [f32_sqrt.o] Error 1
Makefile:320: recipe for target 'f32_to_i64.o' failed
make[1]: *** [f32_to_i64.o] Error 1
In file included from /home/atish/freedom-u-sdk/toolchain/sysroot/usr/include/features.h:447:0,
                 from /home/atish/freedom-u-sdk/toolchain/sysroot/usr/include/bits/libc-header-start.h:33,
                 from /home/atish/freedom-u-sdk/toolchain/sysroot/usr/include/stdint.h:26,
                 from /home/atish/freedom-u-sdk/toolchain/lib/gcc/riscv64-unknown-linux-gnu/7.2.0/include/stdint.h:9,
                 from /home/atish/freedom-u-sdk/riscv-pk/softfloat/f32_to_i32.c:3:
/home/atish/freedom-u-sdk/toolchain/sysroot/usr/include/gnu/stubs.h:17:11: fatal error: gnu/stubs-lp64.h: No such file or directory

jim-wilson, 2018-06-25:
The error indicates some kind of glibc build or install problem.  You might need a make clean after a git update if some submodule version changed.  If you changed arch/abi configure options you could run into a problem like this if you don't do a make clean first.  You get could an error like this if a build was interrupted due to a power failure, disk full error, etc, ended up in an inconsistent state, and you tried to continue without a make clean.

If the error repeats after a make clean then there is something more seriously wrong.  I just tried a build on the master branch, and it worked fine on an Ubuntu 16.04 machine.

atishp04, 2018-06-25:
@jim-wilson @emfive : Thanks.
I did a clean build after git update. I am still getting some other error.

/usr/bin/install -m 0644 support/misc/target-dir-warning.txt /home/atish/freedom-u-sdk/work/buildroot_initramfs/target/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM
>>> toolchain-external undefined Extracting
>>> toolchain-external undefined Patching
>>> toolchain-external undefined Configuring
RPC support available in C library, please enable BR2_TOOLCHAIN_EXTERNAL_INET_RPC
package/pkg-generic.mk:185: recipe for target '/home/atish/freedom-u-sdk/work/buildroot_initramfs/build/toolchain-external-undefined/.stamp_configured' failed
make[2]: *** [/home/atish/freedom-u-sdk/work/buildroot_initramfs/build/toolchain-external-undefined/.stamp_configured] Error 1
Makefile:36: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/home/atish/freedom-u-sdk/buildroot'
Makefile:90: recipe for target '/home/atish/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar' failed
make: *** [/home/atish/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar] Error 2

I did a git submodule update --init --recursive inside toolchain.
Do I need to do that for entire freedom-u-sdk ?

jim-wilson, 2018-06-25:
If you run "git submodule update --init --recursive" at the top level, then you get mutually consistent updated versions of all subpackages.

If you run "git submodule update --init --recursive" inside the riscv-gnu-toolchain dir, then you might get a newer toolchain that isn't compatible with the other as yet un-updated parts of the freedom-u-sdk repository.  Though I'm not sure what exactly will happen in this case, since I don't know if this can work if you haven't updated riscv-gnu-toolchain itself at the top level.  I don't think this will do anything unless you have tried checking out a different branch inside the riscv-gnu-toolchain dir, and that is a risky thing to try.

If you want to mix and match different versions of stuff, then you will have to figure out how to make that work yourself.  But just checking out the master branch and building it should work fine.

The make command you are running may also make a difference.  Are you using "make", "make sim", "make qemu", or maybe something else?

The host system you are using for builds might matter.  We primarily use Ubuntu 16.04 here.

atishp04, 2018-06-25:
Thanks @jim-wilson. I think I corrupted my workspace. A new git clone worked fine.

atishp04, 2018-06-27:
I found out what was the issue with old workspace. I checked out master branch in riscv-pk.
But freedom-u-sdk points to commit 5a0e3e5.

Just wanted to update it for future reference.

Shouldn't freedom-u-sdk point to the latest code in riscv-pk ?

jim-wilson, 2018-06-27:
freedom-u-sdk points at versions of things that are known to build and work together on the host systems and target boards we use here at SiFive.  it would be nice if various components were updated, but that takes time and effort, and there are a great many things that still need to be done.  It is likely that various parts will be updated in the future when time is available to do that work, and when we have a need to upgrade something.  But I'm not aware of any need for a risck-pk update at this time.

atishp04, 2018-06-27:
ok. That makes sense. Thanks again @jim-wilson.
Closing this issue.

closed 2018-06-27 by atishp04