sifiveinc/freedom-u-sdk #62: linux failed to bringup on vc707 board

shawnofthedead, 2018-06-29:
Hello,

I try to use tag: hifiveu-1.0-2018-03-20 (commit id: 0f03e51) to bring up linux on vc707 board.
My freedom sources are up to date (commit id: 397c395). 

However, my terminal hangs as you can see in my terminal log below. 
It seems that in both spi-sifive and serial-sifive, they require clock inputs from rocket-chip generated device tree but the clocks are not there. Has anyone encountered this issue before? How can I resolve this issue?
 
Here is terminal log I got:
INIT
CMD0
CMD8
ACMD41
CMD58
CMD16
CMD18
LOADING  
BOOT
bbl loader

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Coreplex
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00044-g2b0aa1d (shawn@shawn-VirtualBox) (gcc version 7.2.0 (GCC)) #1 SMP Thu Jun 28 18:13:44 PDT 2018
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (9436160 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000]   Normal   [mem 0x00000000c0000000-0x00000bffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] software IO TLB [mem 0xbb1fd000-0xbf1fd000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 14 pages/cpu @        (ptrval) s28632 r0 d28712 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258055
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 950368K/1046528K available (3055K kernel code, 212K rwdata, 832K rodata, 9371K init, 775K bss, 96160K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  RCU event tracing is enabled.
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,plic0,c000000: mapped 7 interrupts to 1/2 handlers
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[    0.000127] sched_clock: 64 bits at 1000kHz, resolution 1000ns, wraps every 2199023255500ns
[    0.009031] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.019100] pid_max: default: 32768 minimum: 301
[    0.025908] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.032878] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.062473] Hierarchical SRCU implementation.
[    0.077576] smp: Bringing up secondary CPUs ...
[    0.081578] smp: Brought up 1 node, 1 CPU
[    0.093534] devtmpfs: initialized
[    0.115748] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.125610] futex hash table entries: 256 (order: 2, 16384 bytes)
[    0.139039] random: get_random_u32 called from bucket_table_alloc+0xe6/0x276 with crng_init=0
[    0.150190] NET: Registered protocol family 16
[    0.265926] vgaarb: loaded
[    0.273426] SCSI subsystem initialized
[    0.284123] usbcore: registered new interface driver usbfs
[    0.289795] usbcore: registered new interface driver hub
[    0.295873] usbcore: registered new device driver usb
[    0.301893] pps_core: LinuxPPS API ver. 1 registered
[    0.306672] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.315866] PTP clock support registered
[    0.330469] clocksource: Switched to clocksource riscv_clocksource
[    0.360712] NET: Registered protocol family 2
[    0.375614] TCP established hash table entries: 8192 (order: 4, 65536 bytes)
[    0.386077] TCP bind hash table entries: 8192 (order: 5, 131072 bytes)
[    0.397154] TCP: Hash tables configured (established 8192 bind 8192)
[    0.407362] UDP hash table entries: 512 (order: 2, 16384 bytes)
[    0.413955] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes)
[    0.423733] NET: Registered protocol family 1
[    1.883937] Unpacking initramfs...
[    3.462752] Initialise system trusted keyrings
[    3.468762] workingset: timestamp_bits=62 max_order=18 bucket_order=0
[    3.615513] random: fast init done
[    3.678553] Key type asymmetric registered
[    3.682829] Asymmetric key parser 'x509' registered
[    3.687341] io scheduler noop registered
[    3.695994] io scheduler cfq registered (default)
[    3.700701] io scheduler mq-deadline registered
[    3.704670] io scheduler kyber registered
[    3.724190] sifive_gpio 64002000.gpio: SiFive GPIO chip registered 4 GPIOs
[    3.743100] xilinx-pcie 2000000000.pci: PCIe Link is UP
[    3.747976] OF: PCI: host bridge /soc/pci@2000000000 ranges:
[    3.753931] OF: PCI:   No bus range found for /soc/pci@2000000000, using [bus 00-ff]
[    3.761705] OF: PCI:   MEM 0x40000000..0x5fffffff -> 0x40000000
[    3.769263] xilinx-pcie 2000000000.pci: PCI host bridge to bus 0000:00
[    3.775744] pci_bus 0000:00: root bus resource [bus 00-ff]
[    3.781126] pci_bus 0000:00: root bus resource [mem 0x40000000-0x5fffffff]
[    3.788637] pci 0000:00:00.0: enabling Extended Tags
[    3.796827] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    3.813347] pci 0000:00:00.0: BAR 8: assigned [mem 0x40000000-0x400fffff]
[    3.820067] pci 0000:00:00.0: BAR 7: no space for [io  size 0x1000]
[    3.825918] pci 0000:00:00.0: BAR 7: failed to assign [io  size 0x1000]
[    3.832972] pci 0000:01:00.0: BAR 1: assigned [mem 0x40000000-0x4007ffff]
[    3.839414] pci 0000:01:00.0: BAR 6: assigned [mem 0x40080000-0x400bffff pref]
[    3.846911] pci 0000:01:00.0: BAR 0: assigned [mem 0x400c0000-0x400dffff]
[    3.853696] pci 0000:01:00.0: BAR 3: assigned [mem 0x400e0000-0x400e3fff]
[    3.860453] pci 0000:01:00.0: BAR 2: no space for [io  size 0x0020]
[    3.866369] pci 0000:01:00.0: BAR 2: failed to assign [io  size 0x0020]
[    3.873293] pci 0000:00:00.0: PCI bridge to [bus 01]
[    3.877996] pci 0000:00:00.0:   bridge window [mem 0x40000000-0x400fffff]
[    4.944327] sifive-serial 64000000.serial: unable to find controller clock
[    4.951349] sifive-serial: probe of 64000000.serial failed with error -2
[    4.972274] sifive_spi 64001000.spi: Unable to find bus clock
[    4.977800] sifive_spi: probe of 64001000.spi failed with error -2
[    4.987075] libphy: Fixed MDIO Bus: probed
[    4.995099] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    5.001463] ehci-pci: EHCI PCI platform driver
[    5.007743] usbcore: registered new interface driver usb-storage
[    5.022082] usbcore: registered new interface driver usbhid
[    5.027072] usbhid: USB HID core driver
[    5.036411] NET: Registered protocol family 17
[    5.049572] bootconsole [early0] uses init memory and must be disabled even before the real one is ready
[    5.058887] bootconsole [early0] disabled

emfive, 2018-06-29:
okay. that's funky.

i've used that same build, but with a kernel and FSBL from around that time frame (3/30) and i didn't see anything like this.

if you can tell me which FSBL & Kernel you're using, I can give it a try when I'm back in the office on Monday.

i *did* get some weird errors w/ an older version of freedom and a newer version of the FSBL, but they didn't look like this and i haven't had time to track them down yet.

maybe the solution to both are problems is to remove any local changes to freedom & then find a stable, newer version of freedom / fsbl / kernel and then loop back around to make changes to the bitstream.

(actually... i was just assuming you were modifying the bistream in the vc707. it's probably a much easier fix if you're not.)

closed 2018-07-06 by terpstra

terpstra, 2018-07-07:
This is still not completely fixed.  When the final version of https://github.com/sifive/freedom/pull/76 is merged, freedom-u-sdk and freedom should work together again.

shawnofthedead, 2018-07-09:
Thanks for letting me know. One of the problems I encounter using the newest freedom project is in sifive.c, "of_alias_get_id" returns me "missing aliases entry" .
For now, I work around it by changing the error check to "if (id < 0) id = 0;".
Just want to inform you this issue if it hasn't been addressed in sifive/freedom#76.

terpstra, 2018-07-09:
It should have been. The driver loads for me at this point and I can manually start getty on the console, but it does not switch the console to the UART for some reason.

terpstra, 2018-07-09:
You probably did not run 'git submodule update --recursive' in freedom after checking out the 'uart-driver-loads' branch. Anyway, I hope to figure out the last remaining issue today.

shawnofthedead, 2018-07-10:
I try out 'uart-driver-loads' branch, but my terminal still hangs after 'bootconsole [early0] disabled'.
I refer to Issue #49, Issue #44, it seems maybe the linux def_config doesn't support the new sifive-serial.
If I run 'make qemu', my simulation is able to bring up linux terminal just fine.  
In comparison, this is "qemu" UART initialization,
Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
10000000.uart: ttyS0 at MMIO 0x10000000 (irq = 13, base_baud = 230400) is a 16550A
This is my board console log,
[    5.278460] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    5.305996] 64000000.serial: ttySI0 at MMIO 0x64000000 (irq = 1, base_baud = 0) is a sifive-serial

terpstra, 2018-07-10:
That is the expected output at this time. I have isolated the problem that prevents the console from switching to the UART. It's a kernel bug that only manifests when there is 1x UART instead of 2x.

terpstra, 2018-07-11:
freedom/master + freedom-u-sdk/master should now have a working console again.

That was a lot of work for a small bug report!

PR #64 (2018-07-10): linux: bump for fixed UART console
Fixes #62.