sifiveinc/freedom-u-sdk #65: fopen returns errno 38

shawnofthedead, 2018-07-11:
Hello,
I use the newest freedom-u-sdk, commit id: 97b2ca1, to run a binary contains fopen function. However, it returns errno 38, which is 'not implemented'. This exact same issue had been addressed in Issue #86 in riscv-linux sub-directory, but the problem seems to manifest again with the newest commit.   

My freedom project commit id is 89059e7.
Thank you

jim-wilson, 2018-07-11:
How exactly did you compile this binary?  If you used a riscv*-elf embedded compiler, then you linked with newlib, which as an embedded target library does not support fopen or other file operations.  So this is the expected result.  If you want a working fopen, you have to use a linux compiler to compile linux applications.  A linux compiler will link against glibc which does support fopen.

freedom-u-sdk uses a riscv*-elf compiler to build the kernel, but that works because the kernel doesn't use any C libraries.  Even so, this is arguably wrong, and should not be done, as the elf compiler supports a slightly different ABI than the Linux compiler and this may eventually lead to problems.

shawnofthedead, 2018-07-11:
Correct me if I am wrong. I used 'riscv-unknown-elf-gcc' to compile this binary. I believe this compile does link with riscv-glibc, because I am able to compile math library with it. 

It might be a problem that my riscv-gcc compiler is out of date. I will rebuild a new one from riscv-tools and try again.

jim-wilson, 2018-07-11:
No, riscv*-unknown-elf-gcc uses newlib, this is a compiler for embedded systems, and it is only an accident that the binaries run on riscv-linux at all.  Newlib does contain a math library in addition to a C library.

I would suggest using riscv-gnu-toolchain instead of riscv-tools, as riscv-tools is poorly maintained and difficult to build.  You can get a linux compiler from riscv-gnu-toolchain by using --enable-linux when configuring it.

freedom-u-sdk should build a linux compiler though.  I have one in my build tree.

shawnofthedead, 2018-07-12:
Thank you. 
I use 'riscv*-unknown-linux-gnu-gcc' in my freedom-u-sdk, and it works just fine.

closed 2018-07-12 by shawnofthedead