sifiveinc/freedom-u-sdk #66: linux compilation failure with latest riscv-all branch

atishp04, 2018-07-26:
I checked out the latest riscv-all branch from riscv-linux and started getting following compilation error. Here is my commit log which matches [https://github.com/riscv/riscv-linux/commits/riscv-all]
``
54d5353c automerging branch "kernel.org-palmer-linux/wip-xilinx_of" into "riscv-all"
2665c958 automerging branch "kernel.org-palmer-linux/wip-xilinx_32bit_dma" into "riscv-all"
5d5ffd76 automerging branch "kernel.org-palmer-linux/wip-timer" into "riscv-all"
15460ab8 automerging branch "kernel.org-palmer-linux/wip-sifive_serial" into "riscv-all"
12cb6d82 automerging branch "kernel.org-palmer-linux/wip-seccomp" into "riscv-all"
5deeb826 automerging branch "kernel.org-palmer-linux/wip-prci_clock" into "riscv-all"
e6a6cf2a automerging branch "kernel.org-palmer-linux/wip-of_order" into "riscv-all"
58f43107 automerging branch "kernel.org-palmer-linux/wip-is25wp256d" into "riscv-all"
e67cd5b7 automerging branch "kernel.org-palmer-linux/wip-irq-3-plic" into "riscv-all"
e249a1b8 automerging branch "kernel.org-palmer-linux/wip-gemgxl_clock_driver" into "riscv-all"
706f9467 automerging branch "kernel.org-palmer-linux/wip-futex_cmpxchg" into "riscv-all"
76cf2cee automerging branch "kernel.org-palmer-linux/wip-fespi" into "riscv-all"
7eacc17a automerging branch "kernel.org-palmer-linux/wip-dt_enable" into "riscv-all"
30f4e8e5 automerging branch "kernel.org-palmer-linux/wip-dma" into "riscv-all"
fa6f7ded automerging branch "kernel.org-palmer-linux/wip-cleanup_generic_irq" into "riscv-all"
113dbaa9 automerging branch "kernel.org-palmer-linux/wip-32bit_dma" into "riscv-all"
005e05ee automerging branch "kernel.org-palmer-linux/review-intc" into "riscv-all"
230999e8 automerging branch "kernel.org-palmer-linux/review-early_printk" into "riscv-all"
a5746143 automerging branch "kernel.org-palmer-linux/review-dt-timebase" into "riscv-all"
1aae169d automerging branch "kernel.org-palmer-linux/fix-ptrace" into "riscv-all"
9f400bab automerging branch "kernel.org-palmer-linux/fix-lustig" into "riscv-all"
9d2c75e4 automerging branch "kernel.org-palmer-linux/fix-intc" into "riscv-all"
013dbd9f automerging branch "kernel.org-palmer-linux/fix-device-init" into "riscv-all"
d40dc71d automerging branch "kernel.org-palmer-linux/fix-addsub32" into "riscv-all"
463ab03d automerging branch "github.com-riscv-riscv_linux/wip-perf_events" into "riscv-all"
d72e90f3 Linux 4.18-rc6
``

Here is the build error:
``
gcc: error: unrecognized argument in option ‘-mabi=lp64’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
  CC      scripts/mod/devicetable-offsets.s
gcc: error: unrecognized command line option ‘-mno-save-restore’
/home/atish/workspace/freedom-u-sdk/linux/scripts/Makefile.build:317: recipe for target 'scripts/mod/empty.o' failed
make[4]: *** [scripts/mod/empty.o] Error 1
make[4]: *** Waiting for unfinished jobs....
gcc: error: unrecognized argument in option ‘-mabi=lp64’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gcc: error: unrecognized command line option ‘-mno-save-restore’
``

However, if I checkout riscv-all based on 4.17, it works fine. I have not modified any other repo in between. Here is the commit history that builds fine.

``
f37675f7 automerging branch "kernel.org-palmer-linux/wip-xilinx_of" into "riscv-all"
8538d0f4 automerging branch "kernel.org-palmer-linux/wip-timer" into "riscv-all"
4b520023 automerging branch "kernel.org-palmer-linux/wip-sifive_serial" into "riscv-all"
c41f4a3c automerging branch "kernel.org-palmer-linux/wip-seccomp" into "riscv-all"
ed547257 automerging branch "kernel.org-palmer-linux/wip-prci_clock" into "riscv-all"
e6413491 automerging branch "kernel.org-palmer-linux/wip-of_order" into "riscv-all"
ef97dca8 automerging branch "kernel.org-palmer-linux/wip-is25wp256d" into "riscv-all"
328762a3 automerging branch "kernel.org-palmer-linux/wip-irq-3-plic" into "riscv-all"
438f7875 automerging branch "kernel.org-palmer-linux/wip-irq-2-hlic" into "riscv-all"
50ed6521 automerging branch "kernel.org-palmer-linux/wip-gemgxl_clock_driver" into "riscv-all"
2bc4cbac automerging branch "kernel.org-palmer-linux/wip-futex_cmpxchg" into "riscv-all"
2b42e640 automerging branch "kernel.org-palmer-linux/wip-fespi" into "riscv-all"
aa059c0b automerging branch "kernel.org-palmer-linux/wip-early_printk" into "riscv-all"
7f9a09bd automerging branch "kernel.org-palmer-linux/wip-dt_enable" into "riscv-all"
d2e978e1 automerging branch "kernel.org-palmer-linux/wip-dt-timebase" into "riscv-all"
841e320f automerging branch "kernel.org-palmer-linux/wip-dma" into "riscv-all"
2b1d264b automerging branch "kernel.org-palmer-linux/wip-cleanup_generic_irq" into "riscv-all"
a6dbb425 automerging branch "kernel.org-palmer-linux/review-console_defconfig" into "riscv-all"
250d493d automerging branch "kernel.org-palmer-linux/next-swiotlb" into "riscv-all"
8015ec09 automerging branch "kernel.org-palmer-linux/next-perf" into "riscv-all"
3e45350c automerging branch "kernel.org-palmer-linux/fix-sparse" into "riscv-all"
a7c82dbd automerging branch "kernel.org-palmer-linux/fix-sifive_maintainers" into "riscv-all"
078f862c automerging branch "kernel.org-palmer-linux/fix-r_riscv_32" into "riscv-all"
3ec20502 automerging branch "kernel.org-palmer-linux/fix-module_alloc" into "riscv-all"
b4ee212a automerging branch "kernel.org-palmer-linux/fix-memaccess_fixup" into "riscv-all"
4d0ca61c automerging branch "kernel.org-palmer-linux/fix-mcount_ftrace" into "riscv-all"
8c0af6f6 automerging branch "kernel.org-palmer-linux/fix-checkflags" into "riscv-all"
0ff797fb automerging branch "github.com-riscv-riscv_linux/wip-perf_events" into "riscv-all"
86406d51 riscv: split the declaration of __copy_user
a93858bc RISC-V: Handle R_RISCV_32 in modules
2f09f36b riscv/ftrace: Export _mcount when DYNAMIC_FTRACE isn't set
9bf97390 riscv: no __user for probe_kernel_address()
2861ae30 riscv: use NULL instead of a plain 0
3b15f447 riscv: add riscv-specific predefines to CHECKFLAGS
9c521764 MAINTAINERS: Update Albert's email, he's back at Berkeley
3ed45d7f MAINTAINERS: Add myself as a maintainer for SiFive's drivers
25d26dfc RISC-V: Load modules within relative jump range of the kernel text.
12eee80b riscv: Fix the bug in memory access fixup code
29dcea88 Linux 4.17
``

michaeljclark, 2018-07-26:
It looks like it's trying to use the host compiler from your gcc error
messages.

Perhaps the compiler prefix has been removed from Kconfig and you now need
to use CROSS_COMPILE

e.g. edit the Makefile and add CROSS_COMPILE (iirc, that is the correct
environment variable)

make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu

atishp04, 2018-07-26:
Thanks @michaeljclark.

It happened because recently CONFIG_CROSS_COMPILE is removed from root Kconfig. 
commit f1089c92 
kbuild: remove CONFIG_CROSS_COMPILE support

I will add it to arch & send a PR.

closed 2022-11-15 by tprrt