sifiveinc/freedom-u-sdk #68: No console output after bootconsole disables

suedadam, 2018-08-07:
I'm having a similar issue to #44 . 
I'm trying to run it with Spike. 

Here's the console log:

```
ssr-risc-v@chisel:~/TestEnv/freedom-u-sdk> ./work/riscv-isa-sim/spike -p4 ./work/riscv-pk/bbl
bbl loader

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Coreplex
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00048-gfe92d7905c6e (ssr-risc-v@chisel) (gcc version 7.2.0 (GCC)) #3 SMP Tue Aug 7 14:15:13 PDT 2018
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (9436160 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000fffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] software IO TLB [mem 0xfa3fd000-0xfe3fd000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 14 pages/cpu @        (ptrval) s28632 r0 d28712 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516615
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 1982868K/2095104K available (3071K kernel code, 217K rwdata, 841K rodata, 9375K init, 777K bss, 112236K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU event tracing is enabled.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,1: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,2: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,3: 64 local interrupts mapped
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000010] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.000995] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.002200] pid_max: default: 32768 minimum: 301
[    0.002805] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.003590] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.005355] Hierarchical SRCU implementation.
[    0.007430] smp: Bringing up secondary CPUs ...
[    0.008570] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.009880] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.011640] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.012950] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.014690] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.016000] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.017145] smp: Brought up 1 node, 4 CPUs
[    0.018075] devtmpfs: initialized
[    0.019020] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.020160] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.021110] random: get_random_u32 called from bucket_table_alloc+0xe6/0x276 with crng_init=0
[    0.022160] NET: Registered protocol family 16
[    0.025460] vgaarb: loaded
[    0.025900] SCSI subsystem initialized
[    0.026635] usbcore: registered new interface driver usbfs
[    0.027285] usbcore: registered new interface driver hub
[    0.028050] usbcore: registered new device driver usb
[    0.028665] pps_core: LinuxPPS API ver. 1 registered
[    0.029245] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.030310] PTP clock support registered
[    0.031200] clocksource: Switched to clocksource riscv_clocksource
[    0.033115] NET: Registered protocol family 2
[    0.033905] TCP established hash table entries: 16384 (order: 5, 131072 bytes)
[    0.034945] TCP bind hash table entries: 16384 (order: 6, 262144 bytes)
[    0.035910] TCP: Hash tables configured (established 16384 bind 16384)
[    0.036705] UDP hash table entries: 1024 (order: 3, 32768 bytes)
[    0.037440] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)
[    0.038275] NET: Registered protocol family 1
[    0.061995] Unpacking initramfs...
[    0.089100] Initialise system trusted keyrings
[    0.089725] workingset: timestamp_bits=62 max_order=19 bucket_order=0
[    0.094880] random: fast init done
[    0.108685] jitterentropy: Initialization failed with host not compliant with requirements: 2
[    0.109675] Key type asymmetric registered
[    0.110155] Asymmetric key parser 'x509' registered
[    0.110725] io scheduler noop registered
[    0.111400] io scheduler cfq registered (default)
[    0.111950] io scheduler mq-deadline registered
[    0.112480] io scheduler kyber registered
[    0.171355] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    0.173040] libphy: Fixed MDIO Bus: probed
[    0.173590] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.174350] ehci-pci: EHCI PCI platform driver
[    0.174910] usbcore: registered new interface driver usb-storage
[    0.175745] usbcore: registered new interface driver usbhid
[    0.176395] usbhid: USB HID core driver
[    0.177010] NET: Registered protocol family 17
[    0.177990] bootconsole [early0] uses init memory and must be disabled even before the real one is ready
[    0.179090] bootconsole [early0] disabled

```

jim-wilson, 2018-08-07:
The default serial driver is the one that works best on the Hifive Unleashed hardware, but it unfortunately does not work on spike.  To get the old serial driver, which does work on spike, enable CONFIG_HVC_RISCV_SBI in conf/linux_defconfig.

suedadam, 2018-08-07:
That worked! Thanks 💯

closed 2018-08-07 by suedadam

yonghaoXuGit, 2018-10-14:
I encountered the same problem, but it can't help after i set CONFIG_HVC_RISCV_SBI=y.

jim-wilson, 2018-10-14:
@yonghaoXuGit Are you running on spike?  Did you rebuild the kernel after setting CONFIG_HVC_RISCV_SBI?  Did you verify that the kernel .config file in the build tree has CONFIG_HVC_RISCV_SBI set?