sifiveinc/freedom-u-sdk #69: unable to copy text file from a sifive riscv-linux root file system to X86 platform

billasurendra, 2018-08-08:
I am executing a one OpenCV executable in sifive riscv-linux and it is generating output as a text file. Generated text file I can see by using ls command but how I want to copy text file from riscv-linux.Can any one Please Help me regarding this issue.

Thanks in Advance

brucehoult, 2018-08-08:
You don't say, but I'm assuming you're using a HiFive Unleashed.

If the file is not too big you can "cat" it to the terminal and copy&paste in your terminal.

Are you connecting only using USB console, or do you have ethernet? If so then there are lots of ways.

You could also pop the SD card into your PC and mount it and read the test file directly.

How did you get the OpenCV executable onto the board?

billasurendra, 2018-08-08:
Thanks for the reply,I am using Sifive riscv-linux port. My output file containing approximately 50,000 lines,by using "cat" entire file information we can't copy.Please give me any other solution because in sifive riscv-linux port  "vi" editor support we don't have more options.

Thanks in Advance

jim-wilson, 2018-08-08:
You need to be clearer about what exactly you are doing.  Are you using "make qemu" to boot a kernel?  Are you using "make sim" to boot a kernel?  Are you copying bbl+kernel onto an SDcard and booting on a HiFive Unleashed?  The answers are different.

Also, Bruce made a good point.  If you have a way to copy OpenCV code into the root file system, then you should have a copy to copy out a data file.

Keep in mind that freedom-u-sdk is mostly for demonstration purposes.  If you want to do real work, you should use a real linux system, such as one of the debian, fedora, opensuse, or openembdded ports.  I've been using Fedora.  
    https://fedoraproject.org/wiki/Architectures/RISC-V
    https://fedorapeople.org/groups/risc-v/disk-images/
They have images and instructions for how to boot on qemu.  This has openssh support, which means you can use ssh to log in, and scp to copy file in and out.  If you are using freedom-u-sdk "make qemu", then you don't have networking support, so can't use ssh or scp.

billasurendra, 2018-08-20:
Thanks for the reply..my problem solved.

closed 2018-08-20 by terpstra