sifiveinc/freedom-u-sdk #70: I am trying to modify makefile for changing ISA ,which is there in freedom-u-sdk

billasurendra, 2018-08-20:
I want to build a sifive port for supporting **ISA-rv64ima**.Here I am modifying makefile, instead of "ISA ?= rv64imafdc" i am planing to change "ISA ?= rv64ima" and here question is what to give instead of "ABI ?= lp64d". Please give me solution to my problem.

Thanks in advance

brucehoult, 2018-08-20:
There is no need to modify the makefile. The whole point of ?= is that this line in the makefile does nothing if you've already defined that variable on the command line.

If you don't have (or don't want to use) a floating point register file, then you'll want "lp64" ABI. You can also use "lp64" ABI with an rv64imafdc ISA which will use floating point instructions within functions, but pass floating point function arguments and return values in the integer registers.

billasurendra, 2018-08-20:
finally if I want to use sifive riscv-linux port with ISA=rv64ima option what is the best way ?

Thanks in Advance

billasurendra, 2018-08-21:
> You can specify makefile variables when running make, e.g.
> make ISA=rv64ima ABI=lp64
>
> You can also set environment variables before running make, e.g.
> export ISA=rv64ima
> export ABI=lp64
> make
> But if you set environment variables, don't forget that you did set them,
> or you may accidentally break other builds you do later in the same shell.
>
> Note however that we only test the rv64gc support. If you build for a
> different target, you may run into build problems, and you will likely have
> to fix them yourself.

As you given reply I have followed.but when I am booting it's struck here

freedom-u-sdk$ make qemu
/home/billa/Pictures/ima/freedom-u-sdk/work/riscv-qemu/prefix/bin/qemu-system-riscv64
-nographic -machine virt -kernel
/home/billa/Pictures/ima/freedom-u-sdk/work/riscv-pk/bbl \
    -drive
file=/home/billa/Pictures/ima/freedom-u-sdk/work/rootfs.bin,format=raw,id=hd0
-device virtio-blk-device,drive=hd0 \
    -netdev user,id=net0 -device virtio-net-device,netdev=net0
bbl loader
/home/billa/Pictures/ima/freedom-u-sdk/riscv-pk/machine/minit.c:74:
assertion failed: !(read_csr(misa) & fd_mask)
Power off

Please give me solution to my problem.

Thanks in advance

jim-wilson, 2018-08-21:
Like I said, you will have to debug this yourself and fix this yourself, as we only support rv64gc.  The kernel may not have support for machines with no FPU.  You might have to add that yourself.

closed 2018-08-21 by jim-wilson

palmer-dabbelt, 2018-08-21:
The kernel is currently broken for non-FPU configurations.  There's active discussion of a patch set on the mailing list: http://lists.infradead.org/pipermail/linux-riscv/2018-August/000924.html