sifiveinc/freedom-u-sdk #72: Python build finished, but the necessary bits to build these modules were not found

SpaceboyRoss01, 2018-08-24:
This happened when I tried `make qemu`.
```
Python build finished, but the necessary bits to build these modules were not found:
_bsddb             _sqlite3           _tkinter        
bsddb185           bz2                dbm             
dl                 gdbm               imageop         
readline           sunaudiodev                        
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

Failed to build these modules:
_hashlib           _ssl                               

Makefile:518: recipe for target 'sharedmods' failed
make[2]: *** [sharedmods] Error 1
make[2]: Leaving directory '/home/spaceboyross/TheBlueSystem/freedom-u-sdk/work/buildroot_rootfs/build/host-python-2.7.11'
package/pkg-generic.mk:195: recipe for target '/home/spaceboyross/TheBlueSystem/freedom-u-sdk/work/buildroot_rootfs/build/host-python-2.7.11/.stamp_built' failed
make[1]: *** [/home/spaceboyross/TheBlueSystem/freedom-u-sdk/work/buildroot_rootfs/build/host-python-2.7.11/.stamp_built] Error 2
make[1]: Leaving directory '/home/spaceboyross/TheBlueSystem/freedom-u-sdk/buildroot'
Makefile:105: recipe for target '/home/spaceboyross/TheBlueSystem/freedom-u-sdk/work/buildroot_rootfs/images/rootfs.ext4' failed
make: *** [/home/spaceboyross/TheBlueSystem/freedom-u-sdk/work/buildroot_rootfs/images/rootfs.ext4] Error 2
```

jim-wilson, 2018-08-28:
I don't see any host-python-2.7.11 info in my build logs.  Did you change anything?  buildroot isn't our package, so we can't guarantee that all buildroot features will work.  We can only support what we build.  If you have questions about buildroot, you probably need to contact the buildroot developers.

SpaceboyRoss01, 2018-08-30:
It's the host part of the python package. I only enabled X11 but got this error doing so.

jim-wilson, 2018-08-30:
If you want X11, you should not use buildroot.  Use debian, ubuntu, gentoo, openembedded, or one of the other real OS ports instead of buildroot.
https://fedoraproject.org/wiki/Architectures/RISC-V
https://wiki.debian.org/RISC-V

Or if you really do want buildroot for this, then you will have to figure it out yourself, or ask for help from buildroot developers.

closed 2018-08-30 by jim-wilson