sifiveinc/freedom-u-sdk #74: buildroot compilation fails with glibc-2.28

acarno, 2018-09-14:
I'm building the freedom-u-sdk on Arch Linux, which ships with glibc-2.28. Compilation breaks while building buildroot:

```
/usr/bin/gcc  -I.   -I/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o fflush.o fflush.c
/usr/bin/gcc  -I.   -I/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o fpurge.o fpurge.c
/usr/bin/gcc  -I.   -I/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o freadahead.o freadahead.c
/usr/bin/gcc  -I.   -I/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o fseek.o fseek.c
/usr/bin/gcc  -I.   -I/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/host/usr/include  -O2 -I/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/host/usr/include -c -o fseeko.o fseeko.c
freadahead.c: In function 'freadahead':
freadahead.c:91:3: error: #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
  #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
   ^~~~~
make[5]: *** [Makefile:1842: freadahead.o] Error 1
make[5]: *** Waiting for unfinished jobs....
fseeko.c: In function 'rpl_fseeko':
fseeko.c:109:4: error: #error "Please port gnulib fseeko.c to your platform! Look at the code in fseeko.c, then report this to bug-gnulib."
   #error "Please port gnulib fseeko.c to your platform! Look at the code in fseeko.c, then report this to bug-gnulib."
    ^~~~~
make[5]: *** [Makefile:1842: fseeko.o] Error 1
make[5]: Leaving directory '/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/build/host-m4-1.4.17/lib'
make[4]: *** [Makefile:1602: all] Error 2
make[4]: Leaving directory '/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/build/host-m4-1.4.17/lib'
make[3]: *** [Makefile:1506: all-recursive] Error 1
make[3]: Leaving directory '/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/build/host-m4-1.4.17'
make[2]: *** [Makefile:1461: all] Error 2
make[2]: Leaving directory '/home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/build/host-m4-1.4.17'
make[1]: *** [package/pkg-generic.mk:198: /home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/build/host-m4-1.4.17/.stamp_built] Error 2
make[1]: Leaving directory '/home/anthony/riscv/freedom-u-sdk/buildroot'
make: *** [Makefile:91: /home/anthony/riscv/freedom-u-sdk/work/buildroot_initramfs/images/rootfs.tar] Error 2
```

This is a known problem for buildroot; it stems from having to build m4, which in turn has an old version of gnulib. I believe the buildroot folks patched this in a more recent version -- see [here](https://git.busybox.net/buildroot/commit/?id=c48f8a64626c60bd1b46804b7cf1a699ff53cdf3).

It also affects bison, as described [here](https://git.busybox.net/buildroot/commit/?id=373ac58b82d2430b745e90c7c8b39e319949882a).

The easiest solution problem involves upgrading the buildroot version shipped with the SDK; once I've got the current version working (by applying the patches), I'll try to spend some time seeing how tricky upgrading buildroot would be.

ArmandArthur, 2020-02-02:
Hello, i have a similar problem 👍 
`$ make
C:/Program Files/Git/mingw64/bin/make.exe  all-recursive
make[1]: Entering directory 'C:/Program Files/httpd-2.4.41/srclib/m4'
Making all in .
/usr/bin/sh: line 20: C:/Program: No such file or directory
make[1]: *** [Makefile:1506: all-recursive] Error 1
make[1]: Leaving directory 'C:/Program Files/httpd-2.4.41/srclib/m4'
make: *** [Makefile:1461: all] Error 2
`
What did i do wrong? or it's a bug?

jim-wilson, 2020-02-02:
This does not look related to the earlier problem.  Please create new issues for new problems.

You are building on a windows machine, and windows likes to put spaces in directory names and file names. But Unix based programs don't like that because space is used as a delimiter in shell and make and lots of other places.  It is best to create a directory at top level to hold unix based programs like mingw64, and don't use spaces for any dir name or file name.

ArmandArthur, 2022-11-15:
Old lol