sifiveinc/freedom-u-sdk #75: CONFIG_CMDLINE arguments not being passed to spike anymore?

sbates130272, 2018-10-01:
Hi All

A clean 'make -j 32 sim' of freedom-u-sdk at 1dfb584c07e4ca47adb0cd706eb1d571ac2d9ba3 does not seem to pass CONFIG_CMDLINE arguments into spike anymore?

sbates@hades:~/freedom-u-sdk-2$ make -j 32 sim
/home/users/sbates/freedom-u-sdk-2/work/riscv-isa-sim/prefix/bin/spike --isa=rv64imafdc -p4 /home/users/sbates/freedom-u-sdk-2/work/riscv-pk/bbl
bbl loader
<stuff>
</stuff>
[    0.000000] Kernel command line: 
<stuff>
</stuff>

Changing things on CONFIG_CMDLINE in conf/linux_defconfig seems to cause a recompile of the bbl but no change to that line in the console and no impact on the booting kernel.

Stephen

PS The CONFIG_CMDLINE changes DO appear in the console in a QEMU boot and DO impact that kernel in that emulator.

palmer-dabbelt, 2018-10-01:
I think this is actually a generic device-tree bug in Linux where if you have no "/chosen" node in the device tree then it will ignore `CONFIG_CMDLINE`.  Here's someone else who just found the bug: http://lists.infradead.org/pipermail/linux-riscv/2018-September/001620.html

sbates130272, 2018-10-03:
Palmer

OK thanks. So the link to the current patch proposed to address this issue (which is a Linux kernel issue) is here:

http://lists.infradead.org/pipermail/linux-riscv/2018-October/001635.html

So this is not a freedom-u-sdk issue per se.

Stephen

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt