sifiveinc/freedom-u-sdk #77: Compilation issue with

amaier17, 2018-11-17:


jim-wilson, 2018-11-17:
This is a versioning issue.  A warning in an older version of perl changed to an error in a more recent version of perl.  We use Ubuntu 16.04 here, and this code works OK on this version of Ubuntu.

There is a buildroot branch mjc/fix-glibc-2.27-perl-5.26 which has a fix for this issue and a glibc versioning issue.  I think it is Ubuntu 18.04 that this branch is known to work on.  You can try checking out this branch of buildroot, and then don't run git submodule at the top level again or it will give you the default buildroot back.

amaier17, 2018-11-20:
Thanks Jim! Looks like I was able to install it after fixing the versioning issue though I've reverted back to 16.04 for now.

closed 2018-11-20 by amaier17