sifiveinc/freedom-u-sdk #79: error in linux booting for riscv

priyanka611, 2018-11-26:
i am executing  spike bbl vmlinux this command but got error like   bootconsole [early0] disabled

give me solution for these.

root@priyanka-Veriton-M6650G:/home/priyanka/RISCV1/riscv-tools/riscv-pk/build# spike bbl vmlinux
OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
Linux version 4.19.0-29675-g9d93e87 (priyanka@priyanka-Veriton-M6650G) (gcc version 7.2.0 (GCC)) #5 SMP Mon Nov 19 11:24:15 IST 2018
bootconsole [early0] enabled
initrd not found or empty - disabling initrd
Zone ranges:
  DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
  Normal   [mem 0x0000000100000000-0x00000fffffffffff]
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x0000000080200000-0x00000000ffffffff]
Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
software IO TLB: mapped [mem 0xfa3fe000-0xfe3fe000] (64MB)
elf_hwcap is 0x112d
percpu: Embedded 17 pages/cpu @(____ptrval____) s29912 r8192 d31528 u69632
Built 1 zonelists, mobility grouping on.  Total pages: 516615
Kernel command line: 
Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
Sorting __ex_table...
Memory: 1971260K/2095104K available (5453K kernel code, 329K rwdata, 1745K rodata, 17709K init, 807K bss, 123844K reserved, 0K cma-reserved)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
rcu: Hierarchical RCU implementation.
rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
Console: colour dummy device 80x25
console [tty0] enabled
bootconsole [early0] disabled

jim-wilson, 2018-11-26:
The default serial driver is the one that works best on the hardware, but it unfortunately does not work on spike.  So to use spike you need to enable the old serial driver.

```
diff --git a/conf/linux_defconfig b/conf/linux_defconfig
index cd87340..87b480f 100644
--- a/conf/linux_defconfig
+++ b/conf/linux_defconfig
@@ -53,7 +53,7 @@ CONFIG_SERIAL_8250_CONSOLE=y
 CONFIG_SERIAL_OF_PLATFORM=y
 CONFIG_SERIAL_SIFIVE=y
 CONFIG_SERIAL_SIFIVE_CONSOLE=y
-# CONFIG_HVC_RISCV_SBI is not set
+CONFIG_HVC_RISCV_SBI=y
 CONFIG_VIRTIO_CONSOLE=y
 # CONFIG_HW_RANDOM is not set
 CONFIG_I2C=y
```

priyanka611, 2018-11-27:
i enabled it using above patch but still i am stucking in booting

jim-wilson, 2018-11-27:
Does "make sim" work?  I think running the kernel directly without an initram won't do much interesting.  make sim will create a buildroot initram with an init that the kernel can run.

closed 2022-11-15 by tprrt