sifiveinc/freedom-u-sdk #83: Git submodule update error: riscv-qemu

aurabindo, 2019-01-09:
Upon doing `git submodule update --recursive --init --depth=1`, I get the below error:

```
Cloning into '/media/aj/Buffer/git-clones/freedom-u-sdk/riscv-qemu'...
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: no such remote ref dfc15319532373fb8cea6b67edfd7f16c366d188
Fetched in submodule path 'buildroot', but it did not contain dfc15319532373fb8cea6b67edfd7f16c366d188. Direct fetching of that commit failed.
```

brucehoult, 2019-01-09:
That commit definitely exists, and is over a year old:

https://github.com/sifive/buildroot/tree/dfc15319532373fb8cea6b67edfd7f16c366d188
palmer-dabbelt Add tyrquake, an open source quake 1 implementation
Latest commit dfc1531  on 14 Dec 2017

aurabindo, 2019-01-09:
@brucehoult Its likely `--depth=1` is the culprit

jim-wilson, 2019-01-09:
freedom-u-sdk is mostly using one year old versions of everything.  I think --depth is limiting git to the top of tree for submodules, but none of the submodules are using top of tree, and hence this can't work.  Maybe this is a git bug, where --depth should be relative to the commit that was specified, but isn't.

I found partial confirmation of this here:
https://stackoverflow.com/questions/2144406/how-to-make-shallow-git-submodules
which mentions that this can fail exactly the way I guessed in one of the comments.  Look for "reference is not a tree".  I didn't bother to read through the whole thing, maybe there is a solution in there somewhere.  And maybe the solution depends on git versions, but we don't control the git support at github so we can't do anything about that.

brucehoult, 2019-01-10:
Ah yes. It seems git in general allows you do specify --depth from any branch, but not from a commit. That's maybe a little inconsistent as a branch is nothing more than a name for a commit.

This is regardless of whether it's being used as a submodule or not.

You're not going to gain much from --depth anyway. In the case of buildroot the entire repo is 81 MB and a checkout of a single version is 48 MB (that commit) or 56 MB (head). I think you'd save about 70 MB of disk if you could do a --depth 1 clone of just the revision.

closed 2022-11-15 by tprrt