sifiveinc/freedom-u-sdk #87: linux cannot mount after booting for qemu and vc707

yanyh15, 2019-02-13:
We have encountered the following error and it seems the linux cannot mount rootfs after being booted. we tested on both qemu and vc707, it report the same error. Any idea what we should look into or a solution?

Yonghong

````

yanyh@lynx:~/chips/freedom-u-sdk$ make qemu 
/home/yanyh/chips/freedom-u-sdk/work/riscv-qemu/prefix/bin/qemu-system-riscv64 -nographic -machine virt -kernel /home/yanyh/chips/freedom-u-sdk/work/riscv-pk/bbl \
	-drive file=/home/yanyh/chips/freedom-u-sdk/work/rootfs.bin,format=raw,id=hd0 -device virtio-blk-device,drive=hd0 \
	-netdev user,id=net0 -device virtio-net-device,netdev=net0
bbl loader

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Core IP
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-00048-gfe92d79 (yanyh@lynx) (gcc version 7.2.0 (GCC)) #2 SMP Thu Jan 10 12:30:03 EST 2019
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x        (ptrval) (9436160 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000]   Normal   [mem 0x0000000088000000-0x0000087fffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] software IO TLB [mem 0x83e3d000-0x87e3d000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 14 pages/cpu @        (ptrval) s28632 r0 d28712 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 31815
[    0.000000] Kernel command line: earlyprintk
[    0.000000] Dentry cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 4, 65536 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 46652K/129024K available (3073K kernel code, 213K rwdata, 839K rodata, 9379K init, 780K bss, 82372K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU event tracing is enabled.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] riscv,plic0,c000000: mapped 10 interrupts to 1/2 handlers
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000103] sched_clock: 64 bits at 10MHz, resolution 100ns, wraps every 4398046511100ns
[    0.001740] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.002142] pid_max: default: 32768 minimum: 301
[    0.002811] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.003027] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.016783] Hierarchical SRCU implementation.
[    0.020545] smp: Bringing up secondary CPUs ...
[    0.020745] smp: Brought up 1 node, 1 CPU
[    0.042788] devtmpfs: initialized
[    0.047258] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.047691] futex hash table entries: 256 (order: 2, 16384 bytes)
[    0.050967] random: get_random_u32 called from bucket_table_alloc+0xe6/0x276 with crng_init=0
[    0.051683] NET: Registered protocol family 16
[    0.064976] vgaarb: loaded
[    0.065551] SCSI subsystem initialized
[    0.066606] usbcore: registered new interface driver usbfs
[    0.066972] usbcore: registered new interface driver hub
[    0.067276] usbcore: registered new device driver usb
[    0.067601] pps_core: LinuxPPS API ver. 1 registered
[    0.067764] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.068085] PTP clock support registered
[    0.071999] clocksource: Switched to clocksource riscv_clocksource
[    0.075229] NET: Registered protocol family 2
[    0.077216] TCP established hash table entries: 1024 (order: 1, 8192 bytes)
[    0.077543] TCP bind hash table entries: 1024 (order: 2, 16384 bytes)
[    0.077799] TCP: Hash tables configured (established 1024 bind 1024)
[    0.078945] UDP hash table entries: 256 (order: 1, 8192 bytes)
[    0.079240] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes)
[    0.080099] NET: Registered protocol family 1
[    0.313281] Unpacking initramfs...
[    0.575713] Initialise system trusted keyrings
[    0.576690] workingset: timestamp_bits=62 max_order=14 bucket_order=0
[    0.586491] random: fast init done
[    0.599689] Key type asymmetric registered
[    0.599874] Asymmetric key parser 'x509' registered
[    0.600179] io scheduler noop registered
[    0.600670] io scheduler cfq registered (default)
[    0.600882] io scheduler mq-deadline registered
[    0.601053] io scheduler kyber registered
[    0.696460] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    0.700080] console [ttyS0] disabled
[    0.700744] 10000000.uart: ttyS0 at MMIO 0x10000000 (irq = 1, base_baud = 230400) is a 16550A
[    0.701653] console [ttyS0] enabled
[    0.701653] console [ttyS0] enabled
[    0.702255] bootconsole [early0] disabled
[    0.702255] bootconsole [early0] disabled
[    0.721062] libphy: Fixed MDIO Bus: probed
[    0.723744] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.723935] ehci-pci: EHCI PCI platform driver
[    0.724335] usbcore: registered new interface driver usb-storage
[    0.725429] usbcore: registered new interface driver usbhid
[    0.725582] usbhid: USB HID core driver
[    0.726104] NET: Registered protocol family 17
[    0.727560] Loading compiled-in X.509 certificates
[    0.738375] Freeing unused kernel memory: 9376K
[    0.738541] This architecture does not have kernel memory protection.
mount: you must be root
mount: you must be root
mkdir: can't create directory '/dev/pts': Permission denied
mkdir: can't create directory '/dev/shm': Permission denied
mount: you must be root
hostname: sethostname: Operation not permitted
Starting logging: OK
Starting mdev...
/etc/init.d/S10mdev: line 21: can't create /proc/sys/kernel/hotplug: nonexistent directory
mdev: /sys/class: No such file or directory
modprobe: can't change directory to '/lib/modules': No such file or directory
Starting network...
ip: RTNETLINK answers: Operation not permitted
ip: SIOCSIFFLAGS: Operation not permitted
sed: /proc/mounts: No such file or directory
Waiting for interface eth0 to appear^C........
....... timeout!
run-parts: /etc/network/if-pre-up.d/wait_iface: exit status 1
/etc/init.d/S50dropbear: line 59: can't create /dev/null: Permission denied
No persistent location to store SSH host keys. New keys will be
generated at each boot. Are you sure this is what you want to do?
Starting dropbear sshd: FAIL
can't open /dev/console: Permission denied
can't open /dev/console: Permission denied
can't open /dev/console: Permission denied
can't open /dev/console: Permission denied
can't open /dev/console: Permission denied
can't open /dev/console: Permission denied
can't open /dev/console: Permission denied
can't open /dev/console: Permission denied
can't open /dev/console: Permission denied

````

jim-wilson, 2019-02-13:
It works for us on Ubuntu 16.04.  What system are you building on?  Maybe the host compiler is buggy, and is miscompiling qemu for instance.

This should also work on Ubuntu 18.04.  We have patches for it, they just aren't on mainline yet.

yanyh15, 2019-02-14:
Thank you. I am not sure what the problem was. I did a fresh clone and re-build, it works. It is on 16.04. Thank you and I will close this issue.

closed 2019-02-14 by yanyh15