sifiveinc/freedom-u-sdk #90: pci switch and fpga heat issues

tmagik, 2019-03-20:
It appears that a Microsemi evaluation board requires more airflow/cooling that is provided by ambient airflow circulation, otherwise the pci switch chip and fpga heat up and cause occasional bit errors. This tends to manifest with one bit flipped in a shared library all the way up to ext4 filesystem corruption, which tends to be recoverable with fsck.

A desktop fan with relatively low flow rates seems provide sufficient cooling for stable operation, and keeps both chips cool to the touch.

tmagik, 2019-04-11:
When there is ext4 filesystem corruption, this can be anything from a complete failure to boot, to having the root filesystem mounted read-only, which is not immediately obvious as the system will likely still boot and run X-windows.

jim-wilson, 2019-04-11:
If your filesystem is mounted read-only, then check the uEnv.txt file to make sure it has rw instead of ro in the boot arguments.

closed 2022-11-15 by tprrt