sifiveinc/freedom-u-sdk #93: Is it OK if I start adding the section of Booting on vc707 in README?

mcd500, 2019-03-23:
This is not an issue, just proposal.
Is it OK if I start adding the section of Booting on vc707 in README?

mcd500, 2019-04-13:
I started the instructions here, and would like to have a comment.
https://github.com/mcd500/freedom-u-sdk/tree/linux_u500vc707devkit_config

Then I would like to revise any before making pull request, after when I have time to bump them up to latest in master which moved to u-boot.

It requires the changes at https://github.com/sifive/freedom/issues/116

These are simplified instructions:
## Cloning sources
```
git clone https://github.com/mcd500/freedom-u-sdk.git
cd freedom-u-sdk
git submodule update --init --recursive --force
git checkout -b linux_u500vc707devkit_config remotes/origin/linux_u500vc707devkit_config
cd buildroot
git checkout -b sifive remotes/origin/sifive
cd ..
```
## Building for vc707 dev board with PCIe HMC module
```
make -j4 BOARD=vc707devkit
```
## Building for vc707 dev board without PCIe HMC module
```
make -j4 BOARD=vc707devkit_nopci
```
When the build is finished, write the image to the sd card.
```
make DISK=/dev/whereisSDcard vc707-sd-write
```

tmagik, 2019-04-15:
That looks good so far, please try it against the latest master with U-boot and see how it goes.

I am also working on an update to https://github.com/sifive/freedom , with the goal of putting U-boot into the FPGA ROM, so that we can just load the exact same work/image.fit file we create for the HiFive Unleashed.

mcd500, 2019-04-20:
Hi @tmagik, I re-read your comment and I think we are working on the same goal. Primary of my goal was to run keystone on vc707 for our purpose.
The first step was to customize the image of HiFive Unleashed to run it on vc707 and we were able to have it running now, 
https://github.com/sifive/freedom-u-sdk/pull/91

Next step was to unify the sd boot image. For unifying the sd boot image, we started from changing the io addressees of vc707 to be same with Unleashed and this seems to be working. 
https://github.com/sifive/freedom/issues/119

Next is to revise the FPGA ROM to read partitions of SD card as what Unleashed does. Currently the bootrom of vc707 does not read any partition of sd card, Just read as if it is a raw device, and we are working on porting zsbl to vc707 which is at the link bellow.
https://github.com/sifive/freedom-u540-c000-bootloader/tree/master/zsbl

I wanted to try the latest master with u-boot but keystone still uses bbl and just did not have time yet.

Can we work together of putting U-boot into the FPGA ROM?

tmagik, 2019-04-20:
> sifive/freedom#119

Oh I think you definitely want to look at my U-boot development sandbox then..

https://github.com/tmagik/freedom-u-sdk/tree/dev/u-boot

I have U-boot loading BBL (which could then load keystone) with the old HiFive U-boot, and I have a freedom build that adds a ‘uboot’ target to freedom/bootrom (which I need to push out)

mcd500, 2019-04-20:
Your u-boot tree is great!. Do you mind accepting my pull requests discussed on the https://github.com/sifive/freedom/issues/116 ?
This should not conflict with anything. https://github.com/sifive/freedom/pull/117

mcd500, 2019-04-20:
The sgdisk in the Makefile in
https://github.com/sifive/HiFive_U-Boot/tree/ca05d269cea0bcdc23d62660b1f8a78bb29fc622
might require the customized sgdisk uploaded here?
https://github.com/sifive/gptfdisk

mcd500, 2019-04-23:
I tried to try the latest master with U-boot today.
The some of the commits of submodule did not seem to be pushed to github yet.
This is the log.
```
$ git submodule status
 ca05d269cea0bcdc23d62660b1f8a78bb29fc622 HiFive_U-Boot (remotes/origin/regression)
 c4ddfe7a5fd964274c99033bd87df3dc7534d196 buildroot (2016.05-529-gc4ddfe7a5f)
+998246b5d00db61bb6449aa81a1df145e5aadf6b linux (v4.17-29681-g998246b5d00d)
 f683e01542acf60e50774d061bcb396b628e3e67 riscv-fesvr (remotes/origin/debug_v013-10-gf683e01)
 092b464c06d2eba580b95a3ac7013449cb2687e4 riscv-isa-sim (remotes/origin/dtm_reset_error-33-g092b464)
 92434c4f697564929f2ceb724bc82d3b6bc58879 riscv-pk (archive-1.9.1-126-g92434c4)
+ad8989a3b5359eb1afa57d7ffedeea6a3c797d9c riscv-qemu (v2.12.0-36-gad8989a3b5)
```

I could not clone the submodules `linux` and `riscv-qemu`with the correct commits and it failed the build.
Do you mind pushing them?

jim-wilson, 2019-04-23:
This can happen when the git repo is switched for a submodule.  On March 7, riscv-qemu was switched from riscv/riscv-qemu to sifive/riscv-qemu.  Similarly, the linux submodule was switched from riscv/riscv-linux to sifive/riscv-linux Dec 29.  I don't know how to fix this with git commands.  I just rm -rf and clone a new tree.  Maybe using git submodule deinit would work?  And then you would have to init them again.  And of course if you have any local changes, be sure to save diffs before you do anything to your git tree.

mcd500, 2019-04-24:
Hi @jim-wilson, thank you for the valuable information.
I edited config files manually and was able to clone the correct commit id.
```
vi .git/config 
vi .git/modules/linux/config 
vi .git/modules/riscv-qemu/config
git submodule update --init --recursive --force
git submodule status 
 ca05d269cea0bcdc23d62660b1f8a78bb29fc622 HiFive_U-Boot (remotes/origin/regression)
 4eddd28f9991cc9bf0d061c069cd057d8bbcffb0 buildroot (2017.08-7449-g4eddd28f99)
 20eeb6522e3302c5f6e435c0bdba40ff57ffa41a linux (v4.17-29693-g20eeb6522e33)
 f683e01542acf60e50774d061bcb396b628e3e67 riscv-fesvr (remotes/origin/debug_v013-10-gf683e01)
 092b464c06d2eba580b95a3ac7013449cb2687e4 riscv-isa-sim (remotes/origin/dtm_reset_error-33-g092b464)
 92434c4f697564929f2ceb724bc82d3b6bc58879 riscv-pk (archive-1.9.1-126-g92434c4)
 16aad2989d1f0067dc88f37207d5c685b7716fa5 riscv-qemu (riscv-for-master-3.1-rc2-1449-g16aad2989d)
```

Thanks again,