sifiveinc/freedom-u-sdk #94: env variable RISCV is used for two different toolchain

mcd500, 2019-03-23:
This is not an issue, just my thought after I was tried building bitstream of rocket core for FPGA vc707 and building sd boot image for the vc707.

The env variable RISCV is used for two different toolchains.
When building rocket-chip for Xilinx VC707 with the bellow link
https://github.com/sifive/freedom
it requires to use `riscv64-unknown-elf-gcc`
and then when I build the sd boot image with freedum-u-sdk
it requires to use `gcc-riscv64-linux-gnu-gcc`,
and I was using `unset RISCV` just in case when going back and forward between
building bitstream of rocket core on FPGA vc707 and building sd boot image.

Would it be good to use different env variable name for `riscv64-unknown-elf-gcc` and `gcc-riscv64-linux-gnu-gcc`, such as RISCV for former and RISCV_LINUX for latter?

tmagik, 2019-03-23:
It’s been on my list of to-do items to replace SD boot with u-boot in the freedom repo builds, and on native RiscV systems, it will generally be a lot easier to use a riscv64-linux-gnu compiler rather than having to cross-compile with an elf gcc.

Otherwise I’m thinking we should do something like https://github.com/tmagik/freedom/commit/37ae7a067e6e30b80455ac6e1cc9d0d6313bc86f or let the CROSS_COMPILE variable specify the gcc to use.

mcd500, 2019-03-25:
I personally prefer using CROSS_COMPILE variable to specifying toochain similar to `make CROSS_COMPILE=riscv64-unknown-linux-gnu` when cross compiling and `make` for native build than using RISCV variable.
Moving the u-boot for the SD boot would be nice.

mcd500, 2019-04-13:
Hi @tmagik, I started to push my changes to my repo.
https://github.com/mcd500/freedom-u-sdk/tree/linux_u500vc707devkit_config
https://github.com/mcd500/freedom/tree/vc707-nopci

once my direction of changes of instructions and Makefile on the previous bbl are being OK, then I will bump them up on the u-boot in master for the pull request.
I am not English native person, and always appreciate help for my sentences in readme.

closed 2022-11-15 by tprrt