sifiveinc/freedom-u-sdk #96: Secure Boot

snathicktechno, 2019-04-01:
i am working with secure boot for High-Five unleashed board.
enabled verified boot options in Hi-Five u-boot then built the u-boot
created key pair
appended cryptographic material to board .dts file and generated dtb using dtc tool
obtained the binary images of vmlinux and dtb
created image tree source file  with vmlinux and dtb and generated fit image using mkimage tool
signed the kernel
put the public key in u-boot image 
then i again rebuild the u-boot, u-boot bin is generated
then using freedom u-sdk, in work directory i am placing u-boot.bin in work directory 
then partioning the sd card and placing u-boot.bin but there is also bbl.bin
then using tftpboot to load images to memory but hash verification not getting during boot
booting up without hash verification
kindly help me

tmagik, 2019-04-09:
Can you paste the serial console boot output, and any patches/changes you made to freedom-u-sdk?

snathicktechno, 2019-04-09:
Environment size: 182/1020 bytes
RISC-V # 
baudrate=115200
bootdelay=3
bootfile=bbl.bin
ethact=gmac0
ethaddr=70:b3:d5:92:f0:00
fdtcontroladdr=fffde200
ip_dyn=yes
serverip=10.60.132.52
stderr=serial
stdin=serial
stdout=serial

Environment size: 182/1020 bytes
RISC-V # iminfo

## Checking Image at 80000000 ...
Unknown image format!
RISC-V # 
RISC-V # 
RISC-V # setenv ipaddr 192.168.27.2
RISC-V # setenv serverip 192.168.27.1
RISC-V # 
  base bdinfo boot bootd bootefi bootelf bootm bootp bootvx cmp coninfo cp
  crc32 dhcp echo editenv env erase ...
RISC-V # tftpboot 0x80000000 fitimage
gmac0: PHY present at 0
gmac0: Starting autonegotiation...
gmac0: Autonegotiation complete
gmac0: link up, 1000Mbps full-duplex (lpa: 0x6800)
Using gmac0 device
TFTP from server 192.168.27.1; our IP address is 192.168.27.2
Filename 'fitimage'.
Load address: 0x80000000
Loading: *
TFTP error: 'File not found' (1)
Not retrying...
RISC-V # tftpboot 0x80000000 fitImage
gmac0: PHY present at 0
gmac0: Starting autonegotiation...
gmac0: Autonegotiation complete
gmac0: link up, 1000Mbps full-duplex (lpa: 0x2800)
Using gmac0 device
TFTP from server 192.168.27.1; our IP address is 192.168.27.2
Filename 'fitImage'.
Load address: 0x80000000
Loading: #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         ##############################
         499 KiB/s
done
Bytes transferred = 11134698 (a9e6ea hex)
RISC-V # bootm 0x80000000 - 0x80200000
## Loading kernel from FIT Image at 80000000 ...
   Using 'config@1' configuration
   Verifying Hash Integrity ... OK
   Trying 'linux_kernel@1' kernel subimage
     Description:  vmlinux
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x800000d4
     Data Size:    11116696 Bytes = 10.6 MiB
     Architecture: RISC-V
     OS:           Linux
     Load Address: 0x80200000
     Entry Point:  0x80200000
     Hash algo:    sha256
     Hash value:   c3548a7b357a003279c7a07d56955530812da7324183218f477f5f52f08edcd1
   Verifying Hash Integrity ... sha256+ OK
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree

Thank you for the response

tmagik, 2019-04-09:
The first problem I see is the FIT image is attempting to load the kernel to 0x80200000, and this is going to overwrite the FIT image, leading to data corruption.

For this reason I load the FIT images to 0xa0000000.

Second I don’t know what version of u-boot you are using, or what patches you’ve applied. Can you try the one at https://github.com/sifive/HiFive_U-Boot/ with the ‘HiFive-U540_regression_defconfig’ as the .config template? (you can run 'make HiFive-U540_regression_defconfig’)