sifiveinc/freedom-u-sdk #101: Support for RTEMS

boehmerst, 2019-04-12:
Are there any plans to support other kernels e.g. RTEMS which support SMP?

jim-wilson, 2019-04-12:
freedom-u-sdk only supports linux.  freedom-u-sdk is what we used to build the initial image for the Freedom Unleashed board.  It isn't meant to be the best place to get an OS image, or any OS for that matter.  It is mainly a nice tutorial for how to build a linux system from scratch.

There is an RTEMS port, and I believe we have customers using it.  The best place to find rtems info is probably rtems.org.

sachinfive, 2019-04-15:
There is RTEMS port available internally but not tested on SMP yet. It will be upstream soon. Eventually it will be available for unleashed board with SMP support.

boehmerst, 2019-04-16:
Thanks a lot for the update. I am aware of the RTEMS port. However, there is little info about how to get it up and running. Especially as it requires a boot loader which does basic HW initializing. Therefore, I think it would fit nicely to the unleashed board as a nice tutorial similar to the Linux motivation which jim-wilson mentioned.

closed 2019-04-16 by boehmerst