sifiveinc/freedom-u-sdk #104: Which branch can bringup in Unleadshed?The master branch cann't work?

mintow, 2019-05-13:
I tried  the current master branch in Unleashed, but the serial port doesn't output anything.

So, I went back to "remotes/origin/regression/kernel4.15", and I met some other problems.
After I enlarged the size of /dev/sdb3 and built the bbl manually,  Linux stuck in the position：
```
SiFive FSBL:       2018-03-20
HiFive-U serial #: 0000028b
bbl loader
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.15.0-g79a9752848da (duanmt@duanmt-u18) (gcc version 8.3.0 (GCC)) #6 SMP Sun May 12 14:57:33 CST 2019
[    0.000000] bootconsole [early0] enabled
[    0.000000] initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x000027ffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x000000027fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x000000027fffffff]
[    0.000000] software IO TLB [mem 0xfbfff000-0xfffff000] (64MB) mapped at [        (ptrval)-        (ptrval)]
[    0.000000] CPU with hartid=0 has a non-okay status of "masked"
[    0.000000] elf_hwcap is 0x1105
[    0.000000] percpu: Embedded 15 pages/cpu @        (ptrval) s28888 r0 d32552 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2067975
[    0.000000] Kernel command line: earlyprintk
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 8184888K/8386560K available (5302K kernel code, 326K rwdata, 1768K rodata, 184K init, 836K bss, 201672K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=5, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  RCU event tracing is enabled.
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=5.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=5
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] CPU with hartid=0 has a non-okay status of "masked"
[    0.000000] riscv,cpu_intc,1: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,2: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,3: 64 local interrupts mapped
[    0.000000] riscv,cpu_intc,4: 64 local interrupts mapped
[    0.000000] ------------[ cut here ]------------
[    0.000000] WARNING: CPU: 1 PID: 0 at /home/duanmt/workspace/github/freedom-u-sdk/linux/drivers/irqchip/irq-riscv-intc.c:167 riscv_irq_enable+0x90/0x92
[    0.000000] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.15.0-g79a9752848da #6
[    0.000000] Call Trace:
[    0.000000] [<        (ptrval)>] walk_stackframe+0x0/0xa2
[    0.000000] [<        (ptrval)>] show_stack+0x26/0x34
[    0.000000] [<        (ptrval)>] dump_stack+0x5e/0x7c
[    0.000000] [<        (ptrval)>] __warn+0x94/0xd2
[    0.000000] [<        (ptrval)>] warn_slowpath_null+0x2c/0x3e
[    0.000000] [<        (ptrval)>] riscv_irq_enable+0x8c/0x92
[    0.000000] [<        (ptrval)>] irq_enable+0x28/0x5e
[    0.000000] [<        (ptrval)>] __irq_startup+0x5a/0x88
[    0.000000] [<        (ptrval)>] irq_startup+0x48/0xe6
[    0.000000] [<        (ptrval)>] irq_activate_and_startup+0x1c/0x28
[    0.000000] [<        (ptrval)>] __irq_do_set_handler+0x6e/0x132
[    0.000000] [<        (ptrval)>] irq_set_chained_handler_and_data+0x2c/0x4a
[    0.000000] [<        (ptrval)>] plic_init+0x216/0x34a
[    0.000000] [<        (ptrval)>] of_irq_init+0x132/0x232
[    0.000000] [<        (ptrval)>] irqchip_init+0x10/0x1c
[    0.000000] [<        (ptrval)>] init_IRQ+0x8/0x14
[    0.000000] [<        (ptrval)>] start_kernel+0x1f6/0x330
[    0.000000] [<        (ptrval)>] _sinittext+0x4e/0x56
[    0.000000] random: get_random_bytes called from print_oops_end_marker+0x1e/0x3e with crng_init=0
[    0.000000] ---[ end trace 66814e3a8c80ec10 ]---
[    0.000000] riscv,plic0,c000000: mapped 53 interrupts to 4/9 handlers
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[    0.000006] sched_clock: 64 bits at 1000kHz, resolution 1000ns, wraps every 2199023255500ns
[    0.008480] Console: colour dummy device 80x25
[    0.012765] console [tty0] enabled
[    0.016148] bootconsole [early0] disabled
```
I will appreciate for any advice.

jim-wilson, 2019-05-13:
The master branch should work.  Did you change the dip switches as mentioned in the docs?  The master branch and the legacy branch require different dip switch positions.  On the master branch, you boot from u-boot instead of flash, and should see messages from u-boot.

If you don't want the master branch, then try the legacy v1_0 branch, and be sure the dip switches are in the correct position for the legacy branch.  The legacy branch gives you a pretty limited buildroot system that can't do much though.  The master branch boots into a debian rootfs.

mintow, 2019-05-13:
@jim-wilson  Thank you very much! 
My dip switches did wrong configuration. MSEL2 should be 0.
The master branch dose work now! Thanks.

closed 2019-05-13 by jim-wilson