sifiveinc/freedom-u-sdk #106: Latest support for VC707 ?

weihuang, 2019-05-14:
Does the current master branch (commit 2adfbe8d383df9305fdec08633c17d248d769431) work with VC707 board? I see a branch named "[linux_u500vc707devkit_config](https://github.com/sifive/freedom-u-sdk/tree/linux_u500vc707devkit_config)", should I use that instead of the current master branch?

I tried both above but neither boots Linux (after "Loading", it stucks at "Boot") like this:
> INIT
> CMD0
> CMD8
> ACMD41
> CMD58
> CMD16
> CMD18
> LOADING
> BOOT

I reverted back to commit [2d533f13f2373b657885834b812fd02253a5f1db](https://github.com/sifive/freedom-u-sdk/commit/2d533f13f2373b657885834b812fd02253a5f1db) in 2016, then VC707 can boot Linux, so I am wondering whether the support for VC707 is still alive.

mcd500, 2019-05-19:
Might reading this thread useful.
https://github.com/sifive/freedom/issues/116
In the middle of thread, this is the procedure it works for me.
```
git clone https://github.com/mcd500/freedom-u-sdk.git
cd freedom-u-sdk
git checkout -b linux_u500vc707devkit_config remotes/origin/linux_u500vc707devkit_config
git submodule update --init --recursive --force
cd buildroot
git checkout -b sifive remotes/origin/sifive
cd ..
make -j4 BOARD=vc707devkit_nopci
```
This will write the image to SD card.
```
make DISK=/dev/whereisSDcard vc707-sd-write
```
And the readme 
https://github.com/mcd500/freedom-u-sdk/tree/linux_u500vc707devkit_config
for the difference between
```
make -j4 BOARD=vc707devkit_nopci
```
```
make -j4 BOARD=vc707devkit
```

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt