sifiveinc/freedom-u-sdk #107: Kernel doesn't mount root automatically after boot.

carlosedp, 2019-05-17:
I've build the complete SDK and flashed into an SD card. After booting, the Kernel doesn't mount `/dev/mmcblk0s2` as root and I go into buildroot. I've flashed both `format-boot-loader` and `format-demo-image`. Also tried with my own Fedora rootfs.

There is also an error that can't change to /lib/modules that might be causing this:

```
...
[    2.120000] mmc0: new SDXC card on SPI
[    2.130000] mmcblk0: mmc0:0000 SE64G 59.5 GiB 
Starting syslogd: [    2.160000]  mmcblk0: p1 p2 p3 p4
OK
Starting klogd: OK
Starting mdev...
modprobe: can't change directory to '/lib/modules': No such file or directory
Initializing random number generator... [    3.090000] random: dd: uninitialized urandom read (512 bytes read)
done.
Starting network: [    3.160000] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
...
```

Full boot logs below:
<details>

```
## Starting application at 0x80000000 ...
bbl loader

                SIFIVE, INC.

         5555555555555555555555555
        5555                   5555
       5555                     5555
      5555                       5555
     5555       5555555555555555555555
    5555       555555555555555555555555
   5555                             5555
  5555                               5555
 5555                                 5555
5555555555555555555555555555          55555
 55555           555555555           55555
   55555           55555           55555
     55555           5           55555
       55555                   55555
         55555               55555
           55555           55555
             55555       55555
               55555   55555
                 555555555
                   55555
                     5

           SiFive RISC-V Core IP
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.19.0-sifive-1+ (carlosedp@ubuntu-risvc) (gcc version 8.3.0 (Buildroot 2019.02-07449-g4eddd28)) #8 SMP Thu May 16 11:01:12 -03 2019
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x(____ptrval____) (16777216 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x000027ffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x000000027fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x000000027fffffff]
[    0.000000] On node 0 totalpages: 2096640
[    0.000000]   DMA32 zone: 7161 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 523776 pages, LIFO batch:63
[    0.000000]   Normal zone: 21504 pages used for memmap
[    0.000000]   Normal zone: 1572864 pages, LIFO batch:63
[    0.000000] software IO TLB: mapped [mem 0xfbfff000-0xfffff000] (64MB)
[    0.000000] CPU with hartid=0 has a non-okay status of "masked"
[... 260 lines cut ...]
OK
Starting klogd: OK
Starting mdev...
modprobe: can't change directory to '/lib/modules': No such file or directory
Initializing random number generator... [    3.090000] random: dd: uninitialized urandom read (512 bytes read)
done.
Starting network: [    3.160000] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
udhcpc: started, v1.29.3
udhcpc: sending discover
udhcpc: sending discover
[    7.360000] macb 10090000.ethernet eth0: link up (1000/Full)
[    7.360000] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
udhcpc: sending discover
udhcpc: sending select for 192.168.15.156
udhcpc: lease of 192.168.15.156 obtained, lease time 43200
deleting routers
adding dns 192.168.15.2
OK
Starting dropbear sshd: [    9.770000] random: dropbear: uninitialized urandom read (32 bytes read)
OK

Welcome to Buildroot
```

</details>

jim-wilson, 2019-05-17:
Mount /dev/mmcblk0p1 and check the uEnv.txt file.  The bootargs line should mention /dev/mmcblk0p2.  Also, there are two setupchosen lines.  You want the second one, the one that does not use setupird, as that loads the initramfs.  This is the part that is probably wrong.  After fixing uEnv.txt, you can reboot to get into the right OS.

Or alternatively, you can edit the conf/uEnv.txt file in the source tree and rebuild the image for the SD card.  But nothing is broken other than the uEnv.txt file which you can edit on the target.

The uEnv.txt file supports both booting into the initramfs and booting into a rootfs in the second partition, and we can have only one uEnv.txt file, so it will always be wrong by default for one of the two choices.  It would be nice if we had better docs in the README.md file that explained exactly how to switch between the two choices though.  There are comments in the uEnv.txt file, but it isn't obvious that one should look there.

carlosedp, 2019-05-17:
Thanks a lot Jim! Swapping the `setupchosen` line did the trick. I really looked around for an answer before opening the issue.

I plan on writing an article on basic setup and etc. I will try to submit a PR on the Readme as well pointing this out.

closed 2019-05-17 by carlosedp