sifiveinc/freedom-u-sdk #110: Gcc Command not found in buildroot

RakeshExaleap, 2019-05-23:
Hi everyone,

    I cloned Freedom-u-SDK and built-in x86 and it's created all the images based on riscv64. Now i want to test qemu-simulator by the command make qemu, Then it gone to the qemu-simulator, i'm trying to run hello world program over there using gcc compiler but it's showing the following Error.
 # gcc --v
-sh: gcc: not found
So how to enable gcc-compiler inside the Freedom-u-sdk and also how to add packages.

Thanks

jim-wilson, 2019-05-23:
buildroot is a build it from scratch system.  There is no package server, and hence no way to add packages.  You can change the buildroot configuration if you want more packages to be built.  See the buildroot documentation.

But if you just want a real OS and don't need to build it yourself, then you are probably better off using one of the linux distros, like Fedora, Debian, or OpenSuSe.  I use Fedora myself.
https://fedoraproject.org/wiki/Architectures/RISC-V/Installing

RakeshExaleap, 2019-05-24:
Hi sir,

   I gone through that mail, you did in Fedora but we are trying to do  in Ubuntu-16.04 and next version of ubuntu. Can you please suggest me a steps to do it in ubuntu. How to enable the gcc in Freedom-u-SDK(buildroot).

Also send me the Details of Fedora version, Arch.

Thanks

https://fedoraproject.org/wiki/Architectures/RISC-V/Installing

tmagik, 2019-05-24:
you can try the ‘make format-demo-image’ target which will download a Debian image that you can install compilers with.

You’ll need to edit the uEnv.txt file on the first partition of the SDcard to boot from Debian instead of the buildroot initramfs.

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt