sifiveinc/freedom-u-sdk #111: Iptables fails to load both as module or built-in

carlosedp, 2019-06-06:
I added IPtables to the current kernel, built and loaded it into the Unleashed. When I try to load the module `sudo modprobe iptables` or run `sudo iptables -L`, I get this dump:

```
[   75.270000] nf_tables: target ffffffe000036618 can not be addressed by the 32-bit offset from PC = 0000000048491271
[   75.380000] nf_tables: target ffffffe000036618 can not be addressed by the 32-bit offset from PC = 00000000cef2682b
[   75.490000] nf_tables: target ffffffe000036618 can not be addressed by the 32-bit offset from PC = 00000000bbab8d61
[   75.630000] nf_tables: target ffffffe000036618 can not be addressed by the 32-bit offset from PC = 0000000054c58b75
[   75.780000] nf_tables: target ffffffe000036618 can not be addressed by the 32-bit offset from PC = 00000000c9df4dac
```

This is my full dmesg:

<details>

```
➜ dmesg
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.19.0-sifive-1+ (carlosedp@ubuntu-risvc) (gcc version 8.3.0 (Buildroot 2019.02-07449-g4eddd28)) #5 SMP Tue Jun 4 14:17:29 -03 2019
[    0.000000] bootconsole [early0] enabled
[    0.000000] initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x000027ffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x000000027fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x000000027fffffff]
[    0.000000] On node 0 totalpages: 2096640
[    0.000000]   DMA32 zone: 8184 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 523776 pages, LIFO batch:63
[    0.000000]   Normal zone: 24576 pages used for memmap
[    0.000000]   Normal zone: 1572864 pages, LIFO batch:63
[    0.000000] software IO TLB: mapped [mem 0xfbfff000-0xfffff000] (64MB)
[    0.000000] CPU with hartid=0 has a non-okay status of "masked"
[    0.000000] CPU with hartid=0 has a non-okay status of "masked"
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 17 pages/cpu @(____ptrval____) s30232 r8192 d31208 u69632
[    0.000000] pcpu-alloc: s30232 r8192 d31208 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2063880
[    0.000000] Kernel command line: console=tty0 console=ttySIF0 root=/dev/mmcblk0p2 rootwait
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 8165888K/8386560K available (6927K kernel code, 377K rwdata, 2639K rodata, 217K init, 836K bss, 220672K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] plic: mapped 53 interrupts to 4 (out of 9) handlers.
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[    0.000000] Console: colour dummy device 80x25
[    0.000000] console [tty0] enabled
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.010000] pid_max: default: 32768 minimum: 301
[    0.010000] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.020000] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.020000] rcu: Hierarchical SRCU implementation.
[    0.030000] smp: Bringing up secondary CPUs ...
[    0.030000] smp: Brought up 1 node, 4 CPUs
[    0.040000] devtmpfs: initialized
[    0.040000] random: get_random_u32 called from bucket_table_alloc+0x76/0x182 with crng_init=0
[    0.050000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.050000] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.060000] NET: Registered protocol family 16
[    0.060000] sifive-u54-prci 10000000.prci: Registered U54 core clocks
[    0.070000] sifive-gemgxl-mgmt 100a0000.cadence-gemgxl-mgmt: Registered clock switch 'cadence-gemgxl-mgmt'
[... 131 lines cut ...]
[   59.050000] sepc: ffffffe0006f7042 ra : ffffffd0089716a6 sp : ffffffe1f3e9d920
[   59.050000]  gp : ffffffe0009ec680 tp : ffffffe1f2c23a80 t0 : ffffffd008979734
[   59.060000]  t1 : ffffffe0006f7042 t2 : 0000000000000080 s0 : ffffffe1f3e9d940
[   59.070000]  s1 : 0000000000000000 a0 : 0000000000000038 a1 : 0000000000000001
[   59.080000]  a2 : 0000000000000000 a3 : 0000000000000001 a4 : ffffffe1f2c24e00
[   59.080000]  a5 : ffffffe1f2c24e48 a6 : ffffffe1f51ba000 a7 : 0000000000000002
[   59.090000]  s2 : 0000000000000038 s3 : ffffffe1f6b3ea40 s4 : ffffffe1ed725d00
[   59.100000]  s5 : ffffffe1edfee000 s6 : 0000000000000000 s7 : ffffffe1f2c2409d
[   59.100000]  s8 : ffffffd009675f68 s9 : 0000003fffe417e0 s10: ffffffe1f3e9dc50
[   59.110000]  s11: ffffffd009a5be18 t3 : 000000000000000f t4 : 000000000000023d
[   59.120000]  t5 : ffffffe1f4e3ca00 t6 : ffffffe1f51bb288
[   59.120000] sstatus: 0000000200000120 sbadaddr: 0000000000000038 scause: 000000000000000d
[   59.130000] ---[ end trace e7e0f8e06256ee07 ]---
[   75.270000] nf_tables: target ffffffe000036618 can not be addressed by the 32-bit offset from PC = 0000000048491271
[   75.380000] nf_tables: target ffffffe000036618 can not be addressed by the 32-bit offset from PC = 00000000cef2682b
[   75.490000] nf_tables: target ffffffe000036618 can not be addressed by the 32-bit offset from PC = 00000000bbab8d61
[   75.630000] nf_tables: target ffffffe000036618 can not be addressed by the 32-bit offset from PC = 0000000054c58b75
[   75.780000] nf_tables: target ffffffe000036618 can not be addressed by the 32-bit offset from PC = 00000000c9df4dac
```

</details>

Any special config I need to get Iptables on the Unleashed?

My Kernel config is on https://gist.github.com/carlosedp/cb673f3201809a4f3986febd9723eeff

carlosedp, 2019-06-06:
After building the Kernel with all iptables features built-in (not modules), I've been able to use iptables.

jim-wilson, 2019-06-06:
freedom-u-sdk is not actively maintained.  We just use it for demos.  You might have better luck asking questions someplace where more people are contributing, like one of the riscv IRC chat rooms, or the linux-riscv mailing list.

FYI Fedora has modules working.  This is with a 5.1 kernel running on qemu.
[root@fedora-riscv ~]# lsmod
Module                  Size  Used by
xt_CHECKSUM             6726  1
ipt_MASQUERADE          6662  3
tun                   587733  1
ebtable_nat             5540  0
ip6table_security       6118  0
iptable_security        6025  0
ip_set                389683  0
nfnetlink              72509  1 ip_set
ebtable_filter          5560  0
ebtables              175695  2 ebtable_nat,ebtable_filter
[root@fedora-riscv ~]#

tmagik, 2019-06-06:
The ‘master’ branch of freedom-u-sdk has module support.

carlosedp, 2019-06-06:
@tmagik I've used the master branch, applied my config but modules didn't work for iptables. It only worked when I built with iptables support built-in.

Also I had no success building a 5.1 Kernel for the unleashed. It boots on Qemu but not in the board. Any tips on what might be missing?

jim-wilson, 2019-06-06:
Not all drivers are upstream yet.  You can find a list of patches in the open-embedded/yocto port.
    https://github.com/riscv/meta-riscv/tree/master/recipes-kernel/linux/files/freedom-u540
I think that is the list for 4.19, but the 5.1 list should be similar.

tmagik, 2019-06-06:
5.1 needs a new device tree format, which makes it hard to upgrade from 4.19 to 5.x without also changing the bootloader.

This is on my list of stuff to look at, but it’s a ways down on the list right now.

carlosedp, 2019-06-06:
What's the most up-to-date Kernel with patches that boot on Unleashed? I'm using the same scheme as the SDK, with U-boot and four partitions.

Can't I build Palmer's tree from https://git.kernel.org/pub/scm/linux/kernel/git/palmer/riscv-linux.git/ and apply those patches Jim linked?

atishp04, 2019-06-06:
> What's the most up-to-date Kernel with patches that boot on Unleashed? I'm
> using the same scheme as the SDK, with U-boot and four partitions.
>
> Can't I build Palmer's tree from
> https://git.kernel.org/pub/scm/linux/kernel/git/palmer/riscv-linux.git/
> and apply those patches Jim linked?
>
> Nope. The patches Jim linked has old drivers. It doesn't have the latest
drivers.
However, the latest drivers uses a new DT which is not upstreamed.
Hopefully, it will be upstreamed during next release.

Here is my 5.2-rc1 tree that works.
https://github.com/atishp04/linux/tree/5.2-rc1_unleashed

FYI: It uses old drivers. So make sure you have following items enabled in
your config if you are planning to use this tree.

CONFIG_CLK_U54_PRCI=y
CONFIG_CLK_GEMGXL_MGMT=y

Sorry, it's bit confusing at this moment. Hopefully, all this mess will be
sorted out soon.

carlosedp, 2019-06-06:
Thanks @atishp04 ! I'll try to build it right now. Do it support modules as well?
Do you have a config for it that works?

atishp04, 2019-06-06:
> Thanks @atishp04 <https://github.com/atishp04> ! I'll try to build it
> right now. Do it support modules as well?
> Do you have a config for it that works?
I never tried modules. So I doubt  if my config works.
But modules should work on 5.2-rc1 kernel. Otherwise, we have a bigger
issue :) :)

tmagik, 2019-06-06:
> On Jun 6, 2019, at 3:02 PM, Atish Patra <notifications@github.com> wrote:
> 
> On Thu, Jun 6, 2019 at 2:53 PM Carlos Eduardo <notifications@github.com>
> wrote:
> 
> > What's the most up-to-date Kernel with patches that boot on Unleashed? I'm
> > using the same scheme as the SDK, with U-boot and four partitions.
> >
> > Can't I build Palmer's tree from
> > https://git.kernel.org/pub/scm/linux/kernel/git/palmer/riscv-linux.git/
> > and apply those patches Jim linked?
> >
> > Nope. The patches Jim linked has old drivers. It doesn't have the latest
> drivers.
> However, the latest drivers uses a new DT which is not upstreamed.
> Hopefully, it will be upstreamed during next release.
> 
> Here is my 5.2-rc1 tree that works.
> https://github.com/atishp04/linux/tree/5.2-rc1_unleashed
> 
> FYI: It uses old drivers. So make sure you have following items enabled in
> your config if you are planning to use this tree.
> 
> CONFIG_CLK_U54_PRCI=y
> CONFIG_CLK_GEMGXL_MGMT=y
> 
> Sorry, it's bit confusing at this moment. Hopefully, all this mess will be
> sorted out soon.

I just added a ‘make test’ target that loads the legacy HiFive U-boot to memory,
runs it, and then loads a FIT image via tftp. The idea here is to use this as the 
starting point for some CI tests so it’s a little easier to clean up the multiple 
different branches and patches we have floating around for different versions.

https://github.com/sifive/freedom-u-sdk/pull/112

I have a ‘make test_s’ target as well that loads the ‘new’ U-boot as an OpenSBI
payload, however it needs a little more work, and since the new u-boot does not
get the ethernet MAC address from OTP, it doesn’t quite work in the test environment
I currently have setup.

carlosedp, 2019-06-07:
I built @atishp04 tree both with my own config (and those two config flags) and using the defconfig (also with the additional configs) but neither boot on the Unleashed.

I'm using the same script as the SDK with four partitions and u-boot. Is there anything I'm missing?

atishp04, 2019-06-07:
> I built @atishp04 <https://github.com/atishp04> tree both with my own
> config (and those two config flags) and using the defconfig (also with the
> additional configs) but neither boot on the Unleashed.
>
> I'm using the same script as the SDK with four partitions and u-boot. Is
> there anything I'm missing?
>
> I have not used freedom-u-sdk in ages. I think it changed significantly
since last time I used.
I usually use OpenSBI + U-Boot + Linux workflow directly. Here are some
documentation.

https://github.com/riscv/opensbi/blob/master/docs/platform/sifive_fu540.md#building-sifive-fu540-platform

One advantage of U-Boot is that, you can load your kernel from network
instead of compiling it in.

Troy's latest changes to freedom-u-sdk also looks promising but haven't
tested them out.

carlosedp, 2019-06-07:
I've successfully build the @atishp04 tree and packed it like the linked doc from OpenSBI. Thanks!

Any idea on why Fedora loads and runs iptables from modules and Debian doesn't?

Here is an output of it on Debian:

```
root@HiFiveU:~# lsmod
Module                  Size  Used by
nf_conntrack_ftp       36730  0
nf_conntrack         1042683  1 nf_conntrack_ftp
nf_defrag_ipv6         84191  1 nf_conntrack
nf_defrag_ipv4         12240  1 nf_conntrack
libcrc32c               4251  1 nf_conntrack
nfnetlink              53852  0
sunrpc               2464917  1
ip_tables             114536  0
autofs4               436098  0
root@HiFiveU:~# 
root@HiFiveU:~# iptables -L
[ 1003.861197] nf_tables: target ffffffe000039518 can not be addressed by the 32-bit offset from PC = 000000000f1e0b6c
[ 1003.969482] nf_tables: target ffffffe000039518 can not be addressed by the 32-bit offset from PC = 000000000f1e0b6c
[ 1004.079396] nf_tables: target ffffffe000039518 can not be addressed by the 32-bit offset from PC = 000000000f1e0b6c
[ 1004.199333] nf_tables: target ffffffe000039518 can not be addressed by the 32-bit offset from PC = 000000000f1e0b6c
[ 1004.309460] nf_tables: target ffffffe000039518 can not be addressed by the 32-bit offset from PC = 000000000f1e0b6c
iptables: Operation not supported.
```

On Fedora using the same Kernel and modules, it works!

atishp04, 2019-06-07:
> I've successfully build the @atishp04 <https://github.com/atishp04> tree
> and packed it like the linked doc from OpenSBI. Thanks!
>
> Awesome.

> Any idea on why Fedora loads and runs iptables from modules and Debian
> doesn't?
>
> Sorry. No idea about that. Probably, IRC chatrooms can help.

irc.oftc.net / irc.debian.org (https://www.oftc.net/)

   - #debian-riscv

FreeNode IRC: #fedora-riscv

> Here is an output of it on Debian:
>
> root@HiFiveU:~# lsmod
> Module                  Size  Used by
> nf_conntrack_ftp       36730  0
> nf_conntrack         1042683  1 nf_conntrack_ftp [...]

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt