sifiveinc/freedom-u-sdk #115: qemu e31 demo project build failing

fsaud, 2019-07-18:
I am trying to build the qemu-sifive-e31-hello template, but it is failing.

Following is the build log

`23:06:35 **** Build of configuration debug for project qemu-sifive-e31-hello ****
make all CONFIGURATION=debug 
cd /c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/build/debug/ && \
	CFLAGS="-march=rv32imac -mabi=ilp32 -mcmodel=medlow -ffunction-sections -fdata-sections -I/c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/install/include --specs=nano.specs -O0 -g" \
	/c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/freedom-metal/configure \
	--host=riscv64-unknown-elf \
	--prefix=/c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/install \
	--libdir=/c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/install/lib/debug \
	--disable-maintainer-mode \
	--with-preconfigured \
	--with-machine-name= \
	--with-machine-header=/c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/metal.h \
                --with-machine-inline=/c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/metal-inline.h \
	--with-platform-header=/c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/metal-platform.h \
	--with-machine-ldscript=/c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/metal.default.lds \
	--with-builtin-libgloss
checking for a BSD-compatible install... /c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/freedom-metal/install-sh -c
checking whether build environment is sane... yes
checking for riscv64-unknown-elf-strip... riscv64-unknown-elf-strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... no
checking whether to enable maintainer-specific portions of Makefiles... no
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... no
configure: error: in `/c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/build/debug':
configure: error: C compiler cannot create executables
See `config.log' for more details
make: *** [Makefile:270: /c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/build/debug/Makefile] Error 77

23:07:03 Build Finished (took 27s.983ms)
Attached is the config.log
`
[e31log.txt](https://github.com/sifive/freedom-u-sdk/files/3407853/e31log.txt)

Kindly resolve the issue as it is, as far as I understand, related to tools

jim-wilson, 2019-07-18:
FYI This is a freedom-e-sdk problem not a freedom-u-sdk problem.  They are maintained by different people and do different things.

The error appears to be in the file /c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/build/debug/config.log which you did not attach so there isn't enough info here to tell what went wrong.

fsaud, 2019-07-18:
[config.log](https://github.com/sifive/freedom-u-sdk/files/3407910/config.log)

fsaud, 2019-07-18:
Thanks Jim,
I am just starting with riscv and setting the things up, trying to execute the demos but unfortunately it is not building. I would be very much thankful to you if you help in this matter

jim-wilson, 2019-07-18:
The error appears to be:
configure:2921: riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -mcmodel=medlow -ffunction-sections -fdata-sections -I/c/Users/fsaud/wsFreedomStudio/qemu-sifive-e31-hello/bsp/install/include --specs=nano.specs -O0 -g   conftest.c  >&5
riscv64-unknown-elf-gcc.exe: error: CreateProcess: No such file or directory

I think CreateProcess is a windows system call.  Maybe it can't find the cc1 binary?  Or is it the gcc binary that it can't find?  The error messge isn't very helpful.  freedom-u-sdk unfortunately is for our Unix support.  I don't have access to a windows machine, and don't know much about how the windows support works.  It might be a PATH problem.  What happens if you run riscv64-unknown-elf-gcc.exe directly?  Does that work?  If not, then you probably need to modify your PATH to add the toolchain install directory.

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt