sifiveinc/freedom-u-sdk #116: Building Linux tools with buildroot

nmeum, 2019-07-20:
Buildroot has a Linux tools package `BR2_PACKAGE_LINUX_TOOLS` for building various utilities contained in the [Linux tree](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/tools?h=v5.2). To enable this packages it seems to be required to build the Linux kernel itself with buildroot. Unfortunately, freedom-u-sdk currently builds the kernel independently of buildroot and only builds the userland with buildroot. Is there any specific reason why the kernel is not build with buildroot?

I encountered this issue because I wanted to enable [perf](https://perf.wiki.kernel.org) (which is part of `BR2_PACKAGE_LINUX_TOOLS`) to make some performance benchmarks.

jim-wilson, 2019-07-21:
Note that freedom-u-sdk is not actively maintained.  We use it for building demos to run on a hifive unleashed board, and we do the minimum amount of work necessary to keep this working.

I would guess a historical issue here.  We probably had the linux kernel working first, and then we added buildroot to get a root file system.  This would have been done a few years ago, and we never changed it because we never had any need to change it.

At the time we did this, buidlroot did not have RISC-V support, and we had to hack it to get it working.  Since then, RISC-V support has been contributed upstream to buidlroot.  You could try using buildroot directly instead of freedom-u-sdk.  I've never tried this myself.

In general, if you want to do anything interesting, you are probably better off using a real linux distro rather than freedom-u-sdk.  Fedora, debian, opensuse, open-embedded all have RISC-V support, and all have more features than freedom-u-sdk.  open-embedded can give you an image that boots directly on a freedom unleashed board.  For the others, if you edit the u-boot uEnv.txt file on the first partition, and install the linux distro rootfs on the second partition, you can use freedom-u-sdk to boot directly into any one of these real linux distros.  I've done this.  freedom-u-sdk comes with a debian rootfs that you can use for this.  "make format-demo-image" copies a debian rootfs to the sdcard.

tprrt, 2022-11-15:
This issue is no longer relevant.