sifiveinc/freedom-u-sdk #117: Makefile:123: *** multiple target patterns.  Stop.

tromedo, 2019-07-21:
Hey Guys,

I just executed "make sim" to run linux in spike and I got the error: Makefile:123: *** multiple target patterns.  Stop.

How do I fix this? Sorry noob here.

jim-wilson, 2019-07-21:
make sim isn't supported at the moment.  spike lacks the driver support necessary to run u-boot.  make qemu still works.  If you really need make sim, then check out the legacy v1_0 branch, though this will give you old versions of everything.

I don't know how you got the multiple target patterns error though.  123 is a line number in the makefile.  Did you accidentally modify it perhaps?

tromedo, 2019-07-21:
> make sim isn't supported at the moment. spike lacks the driver support necessary to run u-boot. make qemu still works. If you really need make sim, then check out the legacy v1_0 branch, though this [...]

Yeah, I am using version 1_0 branch only.

I didn't change the makefile. But I disabled virtual console in Linux folder using make menuconfig. Because Linux boot stopped at "Bootconsole[early0] disabled". Even though the line "CONFIG_HVC_RISCV_SBI=y" is there.

jim-wilson, 2019-07-23:
It works for me on Ubuntu 16.04.  I don't need any change other than turning on CONFIG_HVC_RISCV_SBI.

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt