sifiveinc/freedom-u-sdk #118: Does not let me reprogram expansion board bitstream

ryanmendels, 2019-07-23:
Hi,
I'm trying to flash the expansion board bitstream, the board is powered on and the flash pro 5 is connected to the vjtag header yet the software says vjtag not connected or shorted. It let me flash it once but I erased the flash and now it does not let me flash it anymore. Can someone tell me if there is a button to click or a jumper to manipulate?

Thanks!

JimSughrue, 2019-07-24:
Hi,
The only jumper that has a direct effect on Polar Fire FPGA JTAG is J22 and it should be open - no jumper present.  J22 is within 1/4 inch of the FPGA device but be careful because its near header J21 which should have jumper installed.

The 10pin programming ribbon cable is polarized and keyed on each end.  Correct mechanical connections on each end would be something to check.

Sometimes ribbon cables aren't cut cleanly and have frayed conductors or pop out.  You could inspect cable ends.

If there are any PCIe cards, HiFive Unleashed boards, or other removable hardware installed you could try removing them and retry programming.

ryanmendels, 2019-07-24:
Hello Jim,
The board is still not flashing with a new ribbon cable and a new flash pro 5. It only allows flashing after it’s been offline a few hours and it still fails. The board revision is VERA KIT BOARD REV-2.

Thank you,
Ryan.

JimSughrue, 2019-07-24:
Hi Ryan,
The expansion board runs hot and benefits from some airflow - especially on heatsinked PCIe switch device.  FPGA you're trying to program is near the PCIe switch.

You'd asked about jumpers previously, there are also LED's.  I'm attaching a pic of working board with jumpers and LED's.  You might compare this to your board for a mismatch.  I believe some LED's would apply to PCIe which may not be a factor in the issue you've described.

[image]

ryanmendels, 2019-07-24:
Hello Jim,
Thank you for helping out. I have attached the picture of my board. Unfortunately it is not showing the orange led (only after few hours disconnected) which seems to be why it does not allow flashing.
[image]

JimSughrue, 2019-07-24:
Hi Ryan,

The yellow/orange LED2 and LED2 are debug outputs from the FPGA.  Since you've erased your FPGA they are likely undriven and probably another symptom instead of the root cause of the FPGA programming issue.

From the pic it looks like the inductor north of the power switch is askew.  Is it soldered down good?

FYA, schematics for Expansion board and other info can be found here: https://www.microsemi.com/hifive-unleashed-expansion-board

ryanmendels, 2019-07-24:
Hello Jim,
Yes, the inductor seem to be titled but it looks like it is soldered down good. It does not move if I touch it a little and it is soldered on the contact points.

I programmed the board once before and it was fine, why it does not let me reprogram it anymore? Also when the board was working, the entire pcie bus (pcie gpu, pcie usb ports) would not appear all the time. It also had problems detecting nvme/sata at boot.  Do you think the board is defective?

Thanks,
Ryan.

ryanmendels, 2019-07-26:
Hi there,
I forgot to ask, but I have found a SPI header on the expansion board. Is there a way to flash it by using the flashpro 5 and the SPI header? I have cables that I can use to hook up the spi pins to the flashpro 5. I'm asking because it's not flashing still via the JTAG header.

Thank,
Ryan.

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt