sifiveinc/freedom-u-sdk #121: Stuck at Sifive logo during boot while trying to add gcc to target

Kues-Ba-Skito, 2019-08-19:
**Summary:** I'm trying to make freedom-u-sdk work on vc707 with a gcc for target, but stuck on sifive logo during boot.

**Detailed background information:** I've successfully booted linux on vc707 without PCIe using [freedom #116](https://github.com/sifive/freedom/issues/116). I want to be able to create programs on this linux, in other words I want a compiler on the target system. However, freedom-u-sdk uses buildroot, which does not really entertain the idea of having a compiler on target. The solutions I was able to come up with are:
1) Install an image of an actual linux distribution on the second partition of sdcard, then chroot into it.
2) Add the gcc through buildroot (as if it is a package).

As far as I understood, vc707 is unable to use partitions of the sdcard, so the 1st option is off the table. For the 2nd option, I found this buildroot https://bellard.org/tinyemu/buildroot.html which has a gcc-target2 package (a gcc for target linux), and the buildroot version is really close to the one [I used](https://github.com/mcd500/freedom-u-sdk/tree/linux_u500vc707devkit_config) . I simply replace the existing buildroot folder with the one on the link, and have it work without problem after adding a missing package patch. Since I've confirmed that this new buildroot can be used to create a bbl for vc707, I use:
`
make buildroot_initramfs-menuconfig 
`
and select the _gcc-target2_ package to be installed on target. So here I faced a chain of problems mainly because (I suppose) this package required different buildroot configurations compared to freedom-u-sdk. I didn't manage to get through every error, so I came up with a different solution:
1) make the downloaded buildroot by itself. (Generates an image file rootfs.tar)
2) Locate the steps buildroot is used within the freedom-u-sdk makefile. (It appeared it is used to generate the image rootfs.tar)
3) Run the freedom-u-sdk makefile step by step. Replace the image under _work/buildroot_initramfs/images_ with the image generated in step 1.
4) Run the remaining steps after the generation of image in makefile. (bbl.bin is generated)

I load the generated binary file into sdcard but it gets stuck after the sifive logo during the boot and nothing happens. It appears I simply cannot assume everyting would work properly after copying an image from outside, or so I thought; BUT then I tried this: I opened _rootfs.tar_ under images folder, and manually deleted gcc-related files under _./usr/lib*_. (inside the tar). I repeat all the makefile steps after the generation of tar (but not before). The acquired bbl.bin file successfully boots on vc707, and everything is working properly!
Note that configuration files are all generated before image in makefile, so they are the same in both cases.

**My question is:** It appears to me that the existence of gcc in target image single-handedly prevents bootloader from working, or being generated properly (because it is the only difference between the working one and problematic one). Could that actually be the case for a reason, or there might be something else getting me stuck on sifive logo?

tmagik, 2019-08-19:
What I do if I want a compiler on a vc707 target is NFS-root, however the currently supported mechanism for that requires an overpriced FMC to PCI adapter card, which is not going to work for you.

I'll try to ask @erikdanie if we can get you some chisel code that might allow use of the on-board Ethernet, as well as some patches that update the ROM bootloader code to support partitions, which would allow using the same SDcard image with debian we support on the HiFive Unleashed.

My guess is your problem is the ROM boot code will only load the first 16MB or so of the kernel+ramdisk, so you would get the logo, but the kernel would likely fail to boot.

Kues-Ba-Skito, 2019-08-20:
It seems like the size of the bbl.bin file is the problem as you mentioned. I have tried gradually increasing the size of the image rootfs.tar by adding dummy text files, and the resulting binary file wouldn't boot after something like 25-30 MB. The gcc-target2 package I'm trying to add already takes that much space by itself so it doesn't look possible to add it this way.

> I'll try to ask @erikdanie if we can get you some chisel code that might allow use of the on-board Ethernet, as well as some patches that update the ROM bootloader code to support partitions, which would allow using the same SDcard image with debian we support on the HiFive Unleashed.

Having even one of these options would be really great, as we would have a solid way of adding programs to the vc707 without PCIe adapter!

ErsinCK, 2019-09-29:
I am also interested in this for my research. I am actively looking into it. I could use any tips on how to enable the on-board Ethernet or how to modify the bootloader.

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt